ardera/flutter-pi #1: Can't compile libflutter_engine.so

volkrass, 2019-06-13:
I know is not related directly to you repository, but might be you have an advice for me. 
I'm followed the steps from flutter from scratch article. I get nearly all my files in engine/src folder compiled. Build process stuck AT ../flutter/shell/platform/glfw/flutter_glfw.key_event_handler with ../../third_party/glfw/include/GLFW/glfw3.h:171:12: fatal error: 'GL/gl.h' file not found

In my /opt/vc/include folder copied form my Raspberry PI 3 there are only header files in EGL/GLES/GLES2. I don't have a /opt/vc/include/GL
Is there an option to use EGL or GLES instead? Or do I missing some packages on my Raspberry PI? 
Thanks in advance.

ardera, 2019-06-13:
(I also replied to you on [medium](https://medium.com/@theardera/yep-remember-i-got-that-error-too-3dcfe8c33a9c))

You need to enable the Raspberry Pi GL support by doing `sudo raspi-config` navigate to Advanced, GL Driver and select _Fake KMS_ (or _Full KMS_). The header files should then be in `/usr/include` and the OpenGL lib in `/usr/lib/`.

Then, as usual, rsync the raspberry pi root to your host-machine's cross-compilation sysroot.

For the embedder & engine to run, you need to disable the raspberry pi's GL support again (simply switch to _Legacy_ in `raspi-config`) otherwise you'll get a "service already in use" message.

volkrass, 2019-06-13:
Thank you for your support. Sorry for doubling my question, but I read too late that you have created this repo. Therefore I thought it would be better to discuss technical details here.

volkrass, 2019-06-13:
Looks like I',m very close to get my own libflutter_engine.so compiled. What I don't get for the compilation process is the libGL file. All other other dependencies are solved (like GTK3 and so on) Installed gtk3-dev and libglib2.0-dev on my Raspi (rsync'd it) and configured like you mention OpenGL with Full KMS via raspi-config.  Do you have any hint for me? Thanks in advance.

volkrass, 2019-06-13:
Looks like I have fixed my problem. I installed following packages gtk3-dev, libglib2.0-dev and libgl1-mesa-devon on my Rasp PI to solved my linker issues.  Thanks for your support.

closed 2019-06-13 by volkrass

ardera, 2019-06-13:
nice! no problem