ardera/flutter-pi #9: GPIO acces

jogiji, 2019-10-23:
Can we have GPIO access of some sort with this implementation ?? Just asking

ardera, 2019-10-29:
Yeah, that's a good idea.

akilinomendez, 2019-11-27:
Alternatively you can create a backend with flask or django and call with dio in flutter to execute gpio commands using python.

Is tested and working

ardera, 2019-11-27:
> Alternatively you can create a backend with flask or django and call with dio in flutter to execute gpio commands using python.
> 
> Is tested and working

That's kinda overkill but yeah, sounds like that should be working.

I found a package that does exactly what you want: https://pub.dev/packages/rpi_gpio

Only works for Raspberry Pi though, while flutter-pi (despite its name) works on much more embedded devices.

mortenboye, 2020-01-23:
Here’s another vote for GPIO access or maybe just a wiki page showing how to write and use plugins with flutter-pi (unless dart:ffi gets working soon)

ardera, 2020-01-24:
> Here’s another vote for GPIO access (unless dart:ffi gets working soon)

Yeah actually I started work on GPIO access using `dart:ffi` and `libgpiod` a while ago, because it's probably faster than platform channels. Had basic functionality going & was ready to test, then I found out `dart:ffi` isn't yet supported on ARM hardfp platforms.

mortenboye, 2020-01-24:
> Yeah actually I started work on GPIO access using `dart:ffi` and `libgpiod` a while ago, because it's probably faster than platform channels. Had basic functionality going & was ready to test, then [...]

So, how did you solve this challenge? Write a flutter plugin for GPIO?

ardera, 2020-01-24:
> So, how did you solve this challenge? Write a flutter plugin for GPIO?

I didn't (yet).

ardera, 2020-02-26:
GPIO (more precisely, `libgpiod`) support is now integrated into `flutter-pi`.
The [dart interface](https://github.com/ardera/flutter_gpiod) still needs some work. Once it's done, I'll publish it on [pub.dev](https://pub.dev).

ardera, 2020-03-23:
Just [published it.](https://pub.dev/packages/flutter_gpiod)

closed 2020-03-23 by ardera

mortenboye, 2020-03-31:
dart:ffi seems to be working with hardfp by now, according to https://github.com/dart-lang/sdk/commit/a71d60c1cf9ff350d4f216b62aa44835182d98f5

@ardera Any chance you could share your FFI based plugin for reference?

ardera, 2020-03-31:
> dart:ffi seems to be working with hardfp by now, according to [dart-lang/sdk@a71d60c](https://github.com/dart-lang/sdk/commit/a71d60c1cf9ff350d4f216b62aa44835182d98f5)

Nice! Keep in mind though that this feature is only available to flutter once the next stable version of the dart SDK (2.8) is released. I'll consider migrating the `flutter_gpiod` package to `dart:ffi` once that happens. That would mean a lot of the ugly async stuff, mutexes, etc in `flutter_gpiod` can go away.

> @ardera Any chance you could share your FFI based plugin for reference?

Sure. The code is _reaally_ ugly, very far from finished and doesn't look like the code of the `flutter_gpiod` package at all, though. I was just kinda experimenting.

[flutter_gpiod_ffi.zip](https://github.com/ardera/flutter-pi/files/4409463/flutter_gpiod_ffi.zip)

hhk7734, 2020-04-01:
Test environment
* OS: Linux kernel 5.4, ubuntu 20.04 minimal
* HW: s905x3 board

Thank you :)
flutter_gpiod works well.