ardera/flutter-pi #12: SunFounder RasPad Touch not work

akilinomendez, 2019-11-11:
The raspberry open de app and apparently all work but Touchscreen dont capture  click in any /dev/input/event.

Any idea.

Thank you.

ardera, 2019-11-14:
I hope I understood your question correctly. You mean that flutter-pi won't register taps from your SunFounder RaspPad and you tried all /dev/input/event files?

Could you please run `sudo evtest` (install evtest using `sudo apt install evtest`), select your the input device that corresponds to your touchscreen (the friendly names of the input devices are printed out), tap a few times on the RaspPad while evtest is running, and send me the complete output of evtest?

akilinomendez, 2019-11-14:
Thanks for your answer,

It really only records touches on some sites of the application. I have played with width and ratio because the pixels are 1280 x 800 and 10.10 '.

It seems like only one zone is clickable while the application occupies the entire screen.

ardera, 2019-11-14:
> I have played with width and ratio because the pixels are 1280 x 800 and 10.10 '.

What _exactly_ did you change? I assume by width you mean the physical width (and height) of the display, and by ratio the device pixel ratio used by the flutter engine, which is stored in the `width_mm`, `height_mm` and `pixel_ratio` variables in `flutter-pi.c`?
(You actually don't need to manually calculate pixel_ratio, it's calculated automatically using width_mm and height_mm, or if these are not given, using the physical size data supplied by the screen)

> It seems like only one zone is clickable while the application occupies the entire screen.

And what is this zone? Is it, for example, a rectangle, or a quarter of the screen?

Again, an evtest log (including the part where you select the input device) would really help me figure out what's going on. Maybe you could do some taps that are in the zone that flutter-pi recognizes, and some in the zone where it doesn't, and then send me log.

akilinomendez, 2019-11-15:
I reset  to default flutter-pi.

OK, i created one app  in flutter to get position on tap, only work in left-top small square in screen with this result:

flutter: tapped!
flutter: tap down 90.0, 748.0
flutter: tap up 90.0, 748.0
flutter: tapped!
flutter: tap down 38.0, 763.0
flutter: tap down 12.0, 752.0
flutter: tap down 30.0, 687.0
flutter: tap down 68.0, 645.0
flutter: tap down 68.0, 603.0
flutter: tap down 176.0, 607.0
flutter: tap down 202.0, 526.0
flutter: tap down 90.0, 702.0
flutter: tap down 86.0, 771.0
flutter: tap down 81.0, 672.0
flutter: tap down 107.0, 790.0
flutter: tap down 211.0, 775.0

And evtest tap in similary zone:

Event: time 1573817756.307149, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 688
Event: time 1573817756.307149, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 715
Event: time 1573817756.307149, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 1218
Event: time 1573817756.307149, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1573817756.307149, type 3 (EV_ABS), code 0 (ABS_X), value 715
Event: time 1573817756.307149, type 3 (EV_ABS), code 1 (ABS_Y), value 1218
Event: time 1573817756.307149, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 0
Event: time 1573817756.307149, -------------- SYN_REPORT ------------
Event: time 1573817756.314118, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 8000
Event: time 1573817756.314118, -------------- SYN_REPORT ------------
Event: time 1573817756.320117, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 16000
Event: time 1573817756.320117, -------------- SYN_REPORT ------------
Event: time 1573817756.326115, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 24000
Event: time 1573817756.326115, -------------- SYN_REPORT ------------
Event: time 1573817756.333117, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 32000
Event: time 1573817756.333117, -------------- SYN_REPORT ------------
Event: time 1573817756.339114, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 40000
Event: time 1573817756.339114, -------------- SYN_REPORT ------------
Event: time 1573817756.345117, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 48000
Event: time 1573817756.345117, -------------- SYN_REPORT ------------
Event: time 1573817756.353119, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 56000
Event: time 1573817756.353119, -------------- SYN_REPORT ------------
Event: time 1573817756.358118, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 64000
Event: time 1573817756.358118, -------------- SYN_REPORT ------------
Event: time 1573817756.364121, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 72000
Event: time 1573817756.364121, -------------- SYN_REPORT ------------
Event: time 1573817756.372121, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 80000
Event: time 1573817756.372121, -------------- SYN_REPORT ------------
Event: time 1573817756.378120, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 88000
Event: time 1573817756.378120, -------------- SYN_REPORT ------------
Event: time 1573817756.383121, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 96000
Event: time 1573817756.383121, -------------- SYN_REPORT ------------
Event: time 1573817756.391119, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 104000
Event: time 1573817756.391119, -------------- SYN_REPORT ------------
Event: time 1573817756.397123, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 112000
Event: time 1573817756.397123, -------------- SYN_REPORT ------------
Event: time 1573817756.402125, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 120000
Event: time 1573817756.402125, -------------- SYN_REPORT ------------
Event: time 1573817756.410125, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 128000
Event: time 1573817756.410125, -------------- SYN_REPORT ------------
Event: time 1573817756.416125, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 136000
Event: time 1573817756.416125, -------------- SYN_REPORT ------------
Event: time 1573817756.422132, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 144000
Event: time 1573817756.422132, -------------- SYN_REPORT ------------
Event: time 1573817756.429128, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 152000
Event: time 1573817756.429128, -------------- SYN_REPORT ------------
Event: time 1573817756.435130, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 160000
[... 125 lines cut ...]
Event: time 1573817720.820798, -------------- SYN_REPORT ------------
Event: time 1573817720.826798, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 176000
Event: time 1573817720.826798, -------------- SYN_REPORT ------------
Event: time 1573817720.831800, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 184000
Event: time 1573817720.831800, -------------- SYN_REPORT ------------
Event: time 1573817720.839804, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 192000
Event: time 1573817720.839804, -------------- SYN_REPORT ------------
Event: time 1573817720.845802, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 200000
Event: time 1573817720.845802, -------------- SYN_REPORT ------------
Event: time 1573817720.851804, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 208000
Event: time 1573817720.851804, -------------- SYN_REPORT ------------
Event: time 1573817720.858802, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 216000
Event: time 1573817720.858802, -------------- SYN_REPORT ------------
Event: time 1573817720.864806, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 224000
Event: time 1573817720.864806, -------------- SYN_REPORT ------------
[... cut ...]
Event: time 1573817720.870811, type 3 (EV_ABS), code 1 (ABS_Y), value 9599
Event: time 1573817720.870811, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 232000
Event: time 1573817720.870811, -------------- SYN_REPORT ------------
Event: time 1573817720.875802, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1573817720.875802, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1573817720.875802, type 4 (EV_MSC), code 5 (MSC_TIMESTAMP), value 240000
Event: time 1573817720.875802, -------------- SYN_REPORT ------------

ardera, 2019-11-15:
Ah, I know what's going on. The X and Y values reported by your touchscreen are not aligned to the pixel grid, they're between 0 and 16384 (as it seems) instead of between 0 and 1280 (for example).

That's perfectly legal by the spec, just didn't think it'd be a problem in practice. Should be an easy fix, I just need to query min and max X and Y values using some ioctl's.

ardera, 2019-11-15:
#15 
Could you please test if it works with the newest version?
if it does, I will close this issue.

akilinomendez, 2019-11-15:
thanks, as soon as I can try it

akilinomendez, 2019-11-18:
It works perfectly thanks, I will continue trying a greeting.

closed 2019-11-18 by akilinomendez