ardera/flutter-pi #13: drmModeGetResources failed: Invalid argument

dluksza, 2019-11-13:
Hi,

I'm trying to get `flutter-pi` running on _Raspbian Buster Lite_ image using engine files from your other github repository. I've modified `flutter_gallery` app, build assets on my Linux desktop, then transferred them to raspberry-pi. Display with touchscreen is connected to HDMI0 port. I've also installed lib packages you have listed in the README.md. But still can't run the flutter app because of DRM related issue.

Here is console output from `flutter-pi`:
```
root@raspberrypi:~# /home/pi/flutter-pi/out/flutter-pi -t /dev/input/event0 /home/pi/flutter_assets/
Using touchscreen input from /dev/input/event0
Asset bundle path: /home/pi/flutter_assets/
engine_argv[0] = /home/pi/flutter-pi/out/flutter-pi
initializing display...
Opening DRM device...
Getting DRM resources...
drmModeGetResources failed: Invalid argument
```
Here are the installed packages version:
- libdrm-dev 2.4.99-1~bpo10~1
- libegl-mesa0 19.2.0~rc1-1~bpo10+1~rpt2
- libgbm-dev 19.2.0~rc1-1~bpo10+1~rpt2
- libgl1-mesa-dev 19.2.0~rc1-1~bpo10+1~rpt2
- libgles2-mesa-dev 19.2.0~rc1-1~bpo10+1~rpt2

Any hint what could cause `drmModeGetResources failed`?

ardera, 2019-11-14:
did you activate the v3d OpenGL driver in raspi-config?
i.e.
```sudo raspi-config --> Advanced --> GL Driver --> Fake-KMS```

If you have activated the v3d driver, or it already was activated, and the error still ocurrs, could you please tell me what files there are inside `/dev/dri/`? flutter-pi by default uses `/dev/dri/card0` as its DRM device.

There's actually not much that is leading up to _drmModeGetResources_, flutter-pi just opens `/dev/dri/card0` and then calls _drmModeGetResources_ on it.

dluksza, 2019-11-15:
Yes, I did enabled `Fake-KMS` as a GL driver.

I do see `/dev/dri/card0` (card1 and renderD128). What is `v3d` driver, I've just installed `gldriver-test` and `xinv3d` packages but they didn't fixed this issue. Any other hints?

ardera, 2019-11-15:
The v3d driver is the open source OpenGL driver for the VideoCore 4 made by Eric Anholt, that's the GL driver you're activating when choosing full-kms or fake-kms. If it's set to legacy, the broadcom one is used.

Ah, so you're using a Pi 4! I actually don't have a Pi 4, I'm testing on a Pi 3A+. In that case, the fix is simple. `/dev/dri/card0` is not the device flutter-pi should use on the Pi 4. it's `/dev/dri/card1`. For now, you could simply look for the line inside flutter-pi.c where it says

```C
. . . = open("/dev/dri/card0", O_RDWR);
```

and change the path to `/dev/dri/card1`.

I really thought the drm device would be the same on the 4, but I was wrong as it seems. I'm currently working on an update where flutter-pi will automatically choose it's devices, seems like I need to add capability for automatically selecting the correct DRM device too.

dluksza, 2019-11-15:
Yes, I do have Pi 4. Changing card to `card1` did helped!! Thanks!

Now I'm having different problem with flutter binary format :| Will open another issue for it.

closed 2019-11-15 by dluksza