ardera/flutter-pi #16: Raspberry 4 seems to not be supported

dluksza, 2019-11-15:
Don't know what I'm missing right now. After modifying source code to use `card1`, downgrading my local flutter to 1.9.1 I still can't see the demo app.

It seems that app is running, `flutter-pi` doesn't exit it just stops, but nothing is visible on the monitor. I've checked both HDHMI outputs, also activated `fake_KMS`, for now I run out of ideas what could be wrong.

Also seems that flutter engine arguments are not respected, as you can see in the output I'm setting `observatory-host`, but later on flutter still prints `127.0.0.1` as observatory host.

```
Using touchscreen input from /dev/input/event0
Asset bundle path: /home/pi/flutter-raspberry/flutter_assets/
engine_argv[0] = /home/pi/flutter-pi/out/flutter-pi
engine_argv[1] = --observatory-host
engine_argv[2] = 192.168.0.16
initializing display...
Opening DRM device...
Getting DRM resources...
Finding a connected connector from 1 available connectors...
  connectors[0]: connected? yes, type: 0x0B (HDMI-A), 340mm x 190mm
Choosing DRM mode from 26 available modes...
  modes[0]: name: "1366x768", 1366x768p, 60Hz, type: 72, flags: 5
the chosen DRM mode is preferred by DRM. (DRM_MODE_TYPE_PREFERRED)
Display properties:
  1366 x 768, 60Hz
  340mm x 190mm
  pixel_ratio = 1.057276
Finding DRM encoder...
Creating GBM device
Querying EGL client extensions...
Getting EGL display for GBM device...
Initializing EGL...
Querying EGL display extensions...
Using display 18351384 with EGL version 1.4
===================================
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless EGL_EXT_platform_device"
  display extensions: "EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_WL_bind_wayland_display "
===================================
Binding OpenGL ES API...
Choosing EGL config...
Finding EGL configs with appropriate attributes...
Creating EGL context...
Creating EGL window surface...
===================================
OpenGL ES 2.x information:
  version: "OpenGL ES 3.0 Mesa 19.2.0-rc1"
  shading language version: "OpenGL ES GLSL ES 3.00"
  vendor: "Broadcom"
  renderer: "V3D 4.2"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_texture_border_clamp GL_KHR_context_flush_control GL_OES_draw_elements_base_vertex GL_OES_texture_border_clamp GL_OES_texture_stencil8 GL_EXT_float_blend GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_MESA_shader_integer_functions GL_KHR_parallel_shader_compile GL_EXT_texture_query_lod "
===================================
Swapping buffers...
Locking front buffer...
getting new framebuffer for BO...
Setting CRTC...
Clearing current context...
finished display setup!
Initializing Input devices...
Initializing Application...
Initializing Plugin Registry...
Initialized Services plugin.
[elm327plugin] elm_open: process doesn't have access to serial device "/dev/rfcomm0": No such file or directory
[elm327plugin] ELM327Plugin_init: ELM327 communication was not initialized successfully. elm327plugin won't supply any OBDII data. error code: No such file or directory
[elm327plugin] running pid query queue processor
[ERROR:flutter/shell/platform/embedder/embedder_surface_gl.cc(107)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
flutter: Observatory listening on http://127.0.0.1:35297/gARv1zh8gSg=/
[ERROR:flutter/shell/platform/embedder/embedder_surface_gl.cc(107)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
flutter engine successfully started up.
Running IO thread...
Sending kAdd 0 to Flutter Engine
Running message loop...
Sending kAdd 1 to Flutter Engine
Sending kAdd 2 to Flutter Engine
Sending kAdd 3 to Flutter Engine
Sending kAdd 4 to Flutter Engine
Sending kAdd 5 to Flutter Engine
Sending kAdd 6 to Flutter Engine
Sending kAdd 7 to Flutter Engine
Sending kAdd 8 to Flutter Engine
Sending kAdd 9 to Flutter Engine
```

ardera, 2019-11-15:
Hmm, that's weird. In your previous issue you said you're running Raspbian Buster Lite, so I assume you're not running a desktop environment, is that correct?

The thing is, I don't have a Pi 4 to test so it's just luck if something works on the 4 out of the box. Maybe you could try running `kmscube` (probably need to install it), with the `--device=/dev/dri/card1` argument, or enable Full-KMS and see if that works.

> Also seems that flutter engine arguments are not respected, as you can see in the output I'm setting `observatory-host`, but later on flutter still prints `127.0.0.1` as observatory host.

Yeah, I know about that one. Seems to be a bug in the engine, I don't know what's causing that.

dluksza, 2019-11-16:
> Hmm, that's weird. In your previous issue you said you're running Raspbian Buster Lite, so I assume you're not running a desktop environment, is that correct?

Yes, I'm not running X.

> The thing is, I don't have a Pi 4 to test so it's just luck if something works on the 4 out of the box. 

No worries, I'm not blaming you nor I'm upset that it doesn't work. Lets try to figure why it isn't and make `flutter-pi` compatible with Pi 4. I'm open for helping you making that happen ;)

> Maybe you could try running kmscube (probably need to install it), with the --device=/dev/dri/card1 argument, or enable Full-KMS and see if that works.

I've installed `kmscube`, and it worked without any other packages or configuration. Just out of the box it renders a spinning cube with gradients ;) Here is the output, if it helps with anything:

```
Using display 0x1aff308 with EGL version 1.4
===================================
EGL information:
  version: "1.4"
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless EGL_EXT_platform_device"
  display extensions: "EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_WL_bind_wayland_display "
===================================
OpenGL ES 2.x information:
  version: "OpenGL ES 3.0 Mesa 19.2.0-rc1"
  shading language version: "OpenGL ES GLSL ES 3.00"
  vendor: "Broadcom"
  renderer: "V3D 4.2"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_texture_border_clamp GL_KHR_context_flush_control GL_OES_draw_elements_base_vertex GL_OES_texture_border_clamp GL_OES_texture_stencil8 GL_EXT_float_blend GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_MESA_shader_integer_functions GL_KHR_parallel_shader_compile GL_EXT_texture_query_lod "
===================================
```
How can I help to make `flutter-pi` working on Pi 4?

ardera, 2019-11-16:
> I'm open for helping you making that happen ;)

Thanks!

> How can I help to make `flutter-pi` working on Pi 4?

There are many things we could try.
I'm pretty sure the problem is somewhere in modeset and GL ES context creation, (the `init_display` procedure). There are some environment variables you can set in order to make mesa print some debugging info (see [here](https://www.mesa3d.org/envvars.html)), i.e. `LIBGL_DEBUG=verbose`, `MESA_DEBUG=1` and `EGL_LOG_LEVEL=debug`. You can try setting these and see if they output some errors. Also, you can try setting `MESA_GLES_VERSION_OVERRIDE=2.0`, to see if maybe the higher GLES version on the pi 4 is the cause.

In the meantime I'll try making `init_display` more verbose. I will create a new branch `devel-pi4` for this.

I also had to apply some hacks to `glGetString(GL_EXTENSIONS)`, because it originally listed some extensions that were not supported by the v3d driver for VideoCore 4. I'm not sure whether that could be the cause though. The flutter engine should print some error messages if it doesn't find some gl procedures that it needs, also that'd mean setting `MESA_GLES_VERSION_OVERRIDE=2.0` would fix that.

dluksza, 2019-11-16:
Here is the output with all the debug flags set.

Kinda interesting is that DRI configuration seems to be not found, also libEGL complains about some _optional_ missing extensions and not supported formats. Other than that output looks completly normal for me. I'll also switch to `devel-pi4` and wait for your updates ;)

```
Using touchscreen input from /dev/input/event0
Asset bundle path: /home/pi/flutter-raspberry/flutter_assets/
engine_argv[0] = /home/pi/flutter-pi/out/flutter-pi
initializing display...
Opening DRM device...
Getting DRM resources...
Finding a connected connector from 1 available connectors...
  connectors[0]: connected? yes, type: 0x0B (HDMI-A), 340mm x 190mm
Choosing DRM mode from 26 available modes...
  modes[0]: name: "1366x768", 1366x768p, 60Hz, type: 72, flags: 5
the chosen DRM mode is preferred by DRM. (DRM_MODE_TYPE_PREFERRED)
Display properties:
  1366 x 768, 60Hz
  340mm x 190mm
  pixel_ratio = 1.057276
Finding DRM encoder...
Creating GBM device
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /root/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /root/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /root/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /root/.drirc: No such file or directory.
drmOpenDevice: node name is /dev/dri/renderD128
drmOpenDevice: open result is 5, (OK)
drmOpenDevice: node name is /dev/dri/renderD129
drmOpenDevice: node name is /dev/dri/renderD130
drmOpenDevice: node name is /dev/dri/renderD131
drmOpenDevice: node name is /dev/dri/renderD132
drmOpenDevice: node name is /dev/dri/renderD133
drmOpenDevice: node name is /dev/dri/renderD134
drmOpenDevice: node name is /dev/dri/renderD135
drmOpenDevice: node name is /dev/dri/renderD136
drmOpenDevice: node name is /dev/dri/renderD137
drmOpenDevice: node name is /dev/dri/renderD138
drmOpenDevice: node name is /dev/dri/renderD139
drmOpenDevice: node name is /dev/dri/renderD140
drmOpenDevice: node name is /dev/dri/renderD141
drmOpenDevice: node name is /dev/dri/renderD142
drmOpenDevice: node name is /dev/dri/renderD143
drmOpenDevice: node name is /dev/dri/renderD128
drmOpenDevice: open result is 5, (OK)
drmOpenDevice: node name is /dev/dri/renderD129
drmOpenDevice: node name is /dev/dri/renderD130
drmOpenDevice: node name is /dev/dri/renderD131
drmOpenDevice: node name is /dev/dri/renderD132
drmOpenDevice: node name is /dev/dri/renderD133
drmOpenDevice: node name is /dev/dri/renderD134
drmOpenDevice: node name is /dev/dri/renderD135
drmOpenDevice: node name is /dev/dri/renderD136
drmOpenDevice: node name is /dev/dri/renderD137
drmOpenDevice: node name is /dev/dri/renderD138
drmOpenDevice: node name is /dev/dri/renderD139
drmOpenDevice: node name is /dev/dri/renderD140
drmOpenDevice: node name is /dev/dri/renderD141
drmOpenDevice: node name is /dev/dri/renderD142
drmOpenDevice: node name is /dev/dri/renderD143
drmOpenDevice: node name is /dev/dri/renderD128
drmOpenDevice: open result is 5, (OK)
drmOpenDevice: node name is /dev/dri/renderD129
drmOpenDevice: node name is /dev/dri/renderD130
drmOpenDevice: node name is /dev/dri/renderD131
drmOpenDevice: node name is /dev/dri/renderD132
[... 105 lines cut ...]
finished display setup!
Initializing Input devices...
Initializing Application...
Initializing Plugin Registry...
Initialized Services plugin.
[elm327plugin] elm_open: process doesn't have access to serial device "/dev/rfcomm0": No such file or directory
[elm327plugin] ELM327Plugin_init: ELM327 communication was not initialized successfully. elm327plugin won't supply any OBDII data. error code: No such file or directory
[elm327plugin] running pid query queue processor
[ERROR:flutter/shell/platform/embedder/embedder_surface_gl.cc(107)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
[ERROR:flutter/shell/platform/embedder/embedder_surface_gl.cc(107)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
flutter engine successfully started up.
Running IO thread...
Running message loop...
Sending kAdd 0 to Flutter Engine
Sending kAdd 1 to Flutter Engine
Sending kAdd 2 to Flutter Engine
Sending kAdd 3 to Flutter Engine
Sending kAdd 4 to Flutter Engine
Sending kAdd 5 to Flutter Engine
Sending kAdd 6 to Flutter Engine
Sending kAdd 7 to Flutter Engine
Sending kAdd 8 to Flutter Engine
Sending kAdd 9 to Flutter Engine
flutter: Observatory listening on http://127.0.0.1:35087/eqsI5kUJQTc=/
```

ardera, 2019-11-16:
Yeah, it looks pretty much the same for me, except for the lines that start with `drmOpenDevice`. These are not there for me.

I just now pushed a new commit to devel-pi4 that contains some more verbose modesetting. The DRM device is now chosen automatically, so you now longer need to patch the `/dev/dri/card0` to `/dev/dri/card1`. Also, I tried to change some things to be more close to the kmscube code, maybe that helps.

Could you try and run that? You no longer need to specify the GL ES version override environment variable, the appropriate `glGetString` is now chosen depending on the underlying graphics chip (vc4 or vc6). If you see some error messages from the flutter engine or skia, basically saying the GL context initialization failed or something like that, that means the vc6 needs a hacked `glGetString` too.

dluksza, 2019-11-16:
Still no luck :| I don't see any errors, also checked `dmesg` but nothing caught my attention. I even recompile flutter bundle again from scratch. Still nothing shows on the screen.

```
Using touchscreen input from /dev/input/event0
Asset bundle path: /home/pi/flutter-raspberry/flutter_assets/
engine_argv[0] = /home/pi/flutter-pi/out/flutter-pi
initializing display...
Finding a suitable DRM device, since none is given...
looking for a suitable DRM device from 2 available DRM devices...
  devices[0]: 
    available nodes: DRM_NODE_PRIMARY, 
    nodes[DRM_NODE_PRIMARY] = "/dev/dri/card1"
    bustype: DRM_BUS_PLATFORM
    businfo.fullname: /soc/gpu
    opening DRM device candidate at "/dev/dri/card1"...
    getting resources of DRM device candidate at "/dev/dri/card1"...
    flutter-pi chose "/dev/dri/card1" as its DRM device.
  devices[1]: 
    available nodes: DRM_NODE_PRIMARY, DRM_NODE_RENDER
    nodes[DRM_NODE_PRIMARY] = "/dev/dri/card0"
    nodes[DRM_NODE_RENDER] = "/dev/dri/renderD128"
    bustype: DRM_BUS_PLATFORM
    businfo.fullname: /v3dbus/v3d@7ec04000
Finding a connected connector from 1 available connectors...
  connectors[0]: connected? yes, type: 0x0B (HDMI-A), 340mm x 190mm
Choosing DRM mode from 26 available modes...
  modes[0]: name: "1366x768", 1366x768p, 60Hz, type: 72, flags: 5
    this mode is preferred by DRM. (DRM_MODE_TYPE_PREFERRED)
  modes[1]: name: "1920x1080", 1920x1080p, 60Hz, type: 64, flags: 5
  modes[2]: name: "1920x1080", 1920x1080p, 60Hz, type: 64, flags: 5
  modes[3]: name: "1920x1080", 1920x1080p, 50Hz, type: 64, flags: 5
  modes[4]: name: "1280x800", 1280x800p, 60Hz, type: 64, flags: 9
  modes[5]: name: "1280x720", 1280x720p, 60Hz, type: 64, flags: 5
  modes[6]: name: "1280x720", 1280x720p, 60Hz, type: 64, flags: 5
  modes[7]: name: "1280x720", 1280x720p, 50Hz, type: 64, flags: 5
  modes[8]: name: "1024x768", 1024x768p, 75Hz, type: 64, flags: 5
  modes[9]: name: "1024x768", 1024x768p, 70Hz, type: 64, flags: 10
  modes[10]: name: "1024x768", 1024x768p, 66Hz, type: 0, flags: 6
  modes[11]: name: "1024x768", 1024x768p, 60Hz, type: 64, flags: 10
  modes[12]: name: "832x624", 832x624p, 75Hz, type: 64, flags: 10
  modes[13]: name: "800x600", 800x600p, 75Hz, type: 64, flags: 5
  modes[14]: name: "800x600", 800x600p, 72Hz, type: 64, flags: 5
  modes[15]: name: "800x600", 800x600p, 60Hz, type: 64, flags: 5
  modes[16]: name: "800x600", 800x600p, 56Hz, type: 64, flags: 5
  modes[17]: name: "720x576", 720x576p, 50Hz, type: 64, flags: 10
  modes[18]: name: "720x480", 720x480p, 60Hz, type: 64, flags: 10
  modes[19]: name: "720x480", 720x480p, 60Hz, type: 64, flags: 10
  modes[20]: name: "640x480", 640x480p, 75Hz, type: 64, flags: 10
  modes[21]: name: "640x480", 640x480p, 73Hz, type: 64, flags: 10
  modes[22]: name: "640x480", 640x480p, 67Hz, type: 64, flags: 10
  modes[23]: name: "640x480", 640x480p, 60Hz, type: 64, flags: 10
  modes[24]: name: "640x480", 640x480p, 60Hz, type: 64, flags: 10
  modes[25]: name: "720x400", 720x400p, 70Hz, type: 64, flags: 6
Display properties:
  1366 x 768, 60Hz
  340mm x 190mm
  pixel_ratio = 1.057276
Finding DRM encoder...
Creating GBM device
Querying EGL client extensions...
Getting EGL display for GBM device...
Initializing EGL...
Querying EGL display extensions...
Using display 19212960 with EGL version 1.4
===================================
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless EGL_EXT_platform_device"
  display extensions: "EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_WL_bind_wayland_display "
[... 19 lines cut ...]
finished display setup!
Initializing Input devices...
Initializing Application...
Initializing Plugin Registry...
Initialized Services plugin.
[elm327plugin] elm_open: process doesn't have access to serial device "/dev/rfcomm0": No such file or directory
[elm327plugin] ELM327Plugin_init: ELM327 communication was not initialized successfully. elm327plugin won't supply any OBDII data. error code: No such file or directory
[elm327plugin] running pid query queue processor
[ERROR:flutter/shell/platform/embedder/embedder_surface_gl.cc(107)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
[ERROR:flutter/shell/platform/embedder/embedder_surface_gl.cc(107)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
flutter engine successfully started up.
Running IO thread...
Sending kAdd 0 to Flutter Engine
Sending kAdd 1 to Flutter Engine
Sending kAdd 2 to Flutter Engine
Sending kAdd 3 to Flutter Engine
Sending kAdd 4 to Flutter Engine
Sending kAdd 5 to Flutter Engine
Running message loop...
Sending kAdd 6 to Flutter Engine
Sending kAdd 7 to Flutter Engine
Sending kAdd 8 to Flutter Engine
Sending kAdd 9 to Flutter Engine
flutter: Observatory listening on http://127.0.0.1:43251/atB1HJLPjUE=/
```

ardera, 2019-11-17:
Pushed another small update. flutter-pi should now be doing _exactly_ the same things that kmscube is doing to initialize the display. If this doesn't work, maybe the problem isn't in display initialization after all.

By the way, the native resolution of your display is 1366x768, right? At least that's the one flutter-pi chooses (because it has the `DRM_MODE_TYPE_PREFERRED` type). I'm kinda confused because in the modes listing, it says your display also supports 1920x1080 60Hz.

May I ask what App you are trying to run using flutter-pi? Just the example `flutter create` app, or something else?

dluksza, 2019-11-17:
> By the way, the native resolution of your display is 1366x768, right?

Yes, the native resolution should be 1366 x 768

> May I ask what App you are trying to run using flutter-pi? 

I'm running `flutter create`, the _counter app_ with target platform set to Fuchsia. It's build on my desktop using downgraded version of flutter, I usually work on the `beta` channel.

Still it doesn't work. The screen not even flickers, it stays black all the time :( Here is the output.

```
root@raspberrypi:~# /home/pi/flutter-pi/out/flutter-pi -t /dev/input/event0 /home/pi/flutter-raspberry/flutter_assets/
Using touchscreen input from /dev/input/event0
Asset bundle path: /home/pi/flutter-raspberry/flutter_assets/
engine_argv[0] = /home/pi/flutter-pi/out/flutter-pi
initializing display...
Finding a suitable DRM device, since none is given...
looking for a suitable DRM device from 2 available DRM devices...
  devices[0]: 
    available nodes: DRM_NODE_PRIMARY, 
    nodes[DRM_NODE_PRIMARY] = "/dev/dri/card1"
    bustype: DRM_BUS_PLATFORM
    businfo.fullname: /soc/gpu
    opening DRM device candidate at "/dev/dri/card1"...
    getting resources of DRM device candidate at "/dev/dri/card1"...
    flutter-pi chose "/dev/dri/card1" as its DRM device.
  devices[1]: 
    available nodes: DRM_NODE_PRIMARY, DRM_NODE_RENDER
    nodes[DRM_NODE_PRIMARY] = "/dev/dri/card0"
    nodes[DRM_NODE_RENDER] = "/dev/dri/renderD128"
    bustype: DRM_BUS_PLATFORM
    businfo.fullname: /v3dbus/v3d@7ec04000
Finding a connected connector from 1 available connectors...
  connectors[0]: connected? yes, type: 0x0B (HDMI-A), 340mm x 190mm
Choosing DRM mode from 26 available modes...
  modes[0]: name: "1366x768", 1366x768p, 60Hz, type: 72, flags: 5
    this mode is preferred by DRM. (DRM_MODE_TYPE_PREFERRED)
  modes[1]: name: "1920x1080", 1920x1080p, 60Hz, type: 64, flags: 5
  modes[2]: name: "1920x1080", 1920x1080p, 60Hz, type: 64, flags: 5
  modes[3]: name: "1920x1080", 1920x1080p, 50Hz, type: 64, flags: 5
  modes[4]: name: "1280x800", 1280x800p, 60Hz, type: 64, flags: 9
  modes[5]: name: "1280x720", 1280x720p, 60Hz, type: 64, flags: 5
  modes[6]: name: "1280x720", 1280x720p, 60Hz, type: 64, flags: 5
  modes[7]: name: "1280x720", 1280x720p, 50Hz, type: 64, flags: 5
  modes[8]: name: "1024x768", 1024x768p, 75Hz, type: 64, flags: 5
  modes[9]: name: "1024x768", 1024x768p, 70Hz, type: 64, flags: 10
  modes[10]: name: "1024x768", 1024x768p, 66Hz, type: 0, flags: 6
  modes[11]: name: "1024x768", 1024x768p, 60Hz, type: 64, flags: 10
  modes[12]: name: "832x624", 832x624p, 75Hz, type: 64, flags: 10
  modes[13]: name: "800x600", 800x600p, 75Hz, type: 64, flags: 5
  modes[14]: name: "800x600", 800x600p, 72Hz, type: 64, flags: 5
  modes[15]: name: "800x600", 800x600p, 60Hz, type: 64, flags: 5
  modes[16]: name: "800x600", 800x600p, 56Hz, type: 64, flags: 5
  modes[17]: name: "720x576", 720x576p, 50Hz, type: 64, flags: 10
  modes[18]: name: "720x480", 720x480p, 60Hz, type: 64, flags: 10
  modes[19]: name: "720x480", 720x480p, 60Hz, type: 64, flags: 10
  modes[20]: name: "640x480", 640x480p, 75Hz, type: 64, flags: 10
  modes[21]: name: "640x480", 640x480p, 73Hz, type: 64, flags: 10
  modes[22]: name: "640x480", 640x480p, 67Hz, type: 64, flags: 10
  modes[23]: name: "640x480", 640x480p, 60Hz, type: 64, flags: 10
  modes[24]: name: "640x480", 640x480p, 60Hz, type: 64, flags: 10
  modes[25]: name: "720x400", 720x400p, 70Hz, type: 64, flags: 6
Display properties:
  1366 x 768, 60Hz
  340mm x 190mm
  pixel_ratio = 1.057276
Finding DRM encoder...
Creating GBM device
Querying EGL client extensions...
Getting EGL display for GBM device...
[... 31 lines cut ...]
Initializing Plugin Registry...
Initialized Services plugin.
[elm327plugin] elm_open: process doesn't have access to serial device "/dev/rfcomm0": No such file or directory
[elm327plugin] ELM327Plugin_init: ELM327 communication was not initialized successfully. elm327plugin won't supply any OBDII data. error code: No such file or directory
[elm327plugin] running pid query queue processor
[ERROR:flutter/shell/platform/embedder/embedder_surface_gl.cc(107)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
[ERROR:flutter/shell/platform/embedder/embedder_surface_gl.cc(107)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
flutter engine successfully started up.
Running IO thread...
Sending kAdd 0 to Flutter Engine
Sending kAdd 1 to Flutter Engine
Sending kAdd 2 to Flutter Engine
Running message loop...
Sending kAdd 3 to Flutter Engine
Sending kAdd 4 to Flutter Engine
Sending kAdd 5 to Flutter Engine
Sending kAdd 6 to Flutter Engine
Sending kAdd 7 to Flutter Engine
Sending kAdd 8 to Flutter Engine
Sending kAdd 9 to Flutter Engine
flutter: Observatory listening on http://127.0.0.1:46687/zmB1S5bxOE4=/
***PRESENT***
***PRESENT***

```

ardera, 2019-11-17:
There one more thing we can try, we can disable all GL ES extensions and override the GL version to 2.0 (using the environment variable you used before). Maybe the V3D driver reports some things as working even though they aren't, similiar to the VideoCore IV driver.

I'm afraid simple debugging ends after that though.

dluksza, 2019-11-17:
> we can disable all GL ES extensions and override the GL version to 2.0 (using the environment variable you used before

Just tired it and it didn't worked :|

I can give you ssh access to my PI4 if you would like. Also you could send/push your `flutter_assets` that are known to work so that I can verify it with something that works for sure.

ardera, 2019-11-18:
> I can give you ssh access to my PI4 if you would like.

That would certainly allow me to test all the things, but how are we gonna tell if something actually works? There seems to be no remote viewing solution for headless linux. There used to be dispmanx_vnc, but that's probably not working anymore, now that dispmanx is obsolete.

I also just bought a Raspberry Pi 4 (2GB version), it will arrive tomorrow.

My asset bundle is this: https://1drv.ms/u/s!AsNHmDqMr-aVgcgAhAGjCx7T2nkrmw?e=j8HLkR

dluksza, 2019-11-18:
> That would certainly allow me to test all the things, but how are we gonna tell if something actually works?

Well... we can get creative, eg. I can start a YT stream and point camera to the other monitor :D But since you are getting your own PI 4 it probably won't be necessary.

> I also just bought a Raspberry Pi 4 (2GB version), it will arrive tomorrow.

Cool! Can't wait for it! Thank you!

> My asset bundle is this: https://1drv.ms/u/s!AsNHmDqMr-aVgcgAhAGjCx7T2nkrmw?e=j8HLkR

This doesn't improve the situation, still nothing is shown :|

akilinomendez, 2019-11-18:
In case it helps, I run on raspberry pi 4 only change in flutter-pi.c 

snprintf(drm.device, sizeof(drm.device), "/dev/dri/card1");

And work perfectly.

ardera, 2019-11-18:
> In case it helps, I run on raspberry pi 4 only change in flutter-pi.c
> 
> snprintf(drm.device, sizeof(drm.device), "/dev/dri/card1");

Yeah, I already fixed that one in the `devel-pi4` branch. But it still doesn't work for @dluksza , while `kmscube`, which basically does the same low-level graphics stuff as flutter-pi, works perfectly.

> and make again.
> 
> Work perfectly.

Thanks! That's valuable information. That means the issue is probably not in flutter-pi, but somewhere else.

Would you mind if you both @dluksza @akilinomendez run `raspinfo` (install using `sudo apt install raspinfo`) and show me the output? Maybe the issue is the firmware or kernel version.

Also @dluksza, do you maybe have some other display lying around you can test? Maybe flutter doesn't like 1366x768. And I suppose you've already done that, but if you haven't, could you please do a full system upgrade using `apt update` & `apt upgrade`? You can also do a firmware update using `rpi-eeprom-update` (or `rpi-update`, not sure).

dluksza, 2019-11-18:
Here output from `raspinfo`:

[raspinfo.txt](https://github.com/ardera/flutter-pi/files/3860061/raspinfo.txt)

I did upgraded raspbian a few days ago, will do it right now. Upgrading didn't fix the issue.

Unfortunately I don't have any other display at hand, but will try to find another one in couple of days and will inform if it did worked.

> You can also do a firmware update using rpi-eeprom-update (or rpi-update, not sure)

The command is `rpi-upgrade`, I've run it a week or two ago. But will do it again right now... nope, didn't improve things. BTW. I'm running kernel 4.19.83-v7l+.

Also @akilinomendez, which version of raspbian are you using, did you installed X server or any other development libraries?

Maybe I've installed something inappropriate some kind of package, a lib or lib-dev that is making flutter upset.

BTW. I wanted to start today from scratch with new install of raspbian base, then give it a try to raspbinan with X pre-configured, but since we want to test the resolution, I'll hold of until I got a spare monitor to test on.

dluksza, 2019-11-19:
BTW. is it possible that flutter doesn't render the UI because of _unsupported_ touchscreen input device?

akilinomendez, 2019-11-19:
Upload my raspinfo.txt.

[raspinfo.txt](https://github.com/ardera/flutter-pi/files/3862455/raspinfo.txt)

dluksza, 2019-11-19:
There must be something with the display I'm using. I've connected Pi to another monitor and was able to see kernel buffer, when on mine it isn't showing anything while system is starting. Didn't had any input device at hand therefore was not able to start `flutter-pi` (since it was complaining about missing/wrong input device).

I wonder if using raspbian with configured X environment could improve things and show kernel buffer and/or flutter.

ardera, 2019-11-19:
> Upload my raspinfo.txt.
> 
> [raspinfo.txt](https://github.com/ardera/flutter-pi/files/3862455/raspinfo.txt)

Thanks!

> BTW. is it possible that flutter doesn't render the UI because of _unsupported_ touchscreen input device?

No, input devices are handled by flutter-pi. The flutter engine only sees the input events flutter-pi sends to it.

> Didn't had any input device at hand therefore was not able to start `flutter-pi` (since it was complaining about missing/wrong input device).

Yeah, sorry about that, running flutter-pi without input devices should really be allowed. I'm gonna fix that in the next update.

> There must be something with the display I'm using. I've connected Pi to another monitor and was able to see kernel buffer, when on mine it isn't showing anything while system is starting.

> I wonder if using raspbian with configured X environment could improve things and show kernel buffer and/or flutter.

Weird, but the kernel buffer is the only thing you're not seeing right? So the whole Raspbian startup is visible on screen?

BTW: which HDMI connector are you using? The one nearest to the USB-C (HDMI0) or the other one (HDMI1), or have you tried both? HDMI0 seems to be the more capable.

dluksza, 2019-11-19:
> Weird, but the kernel buffer is the only thing you're not seeing right? So the whole Raspbian startup is visible on screen?

No, I see completely nothing on my screen, no kernel buffer/messages, no raspbian startup, anything. The only think I can see is the `kmscube` if I start it over ssh. Other than that nothing, even login prompt is not visible. I did assumed that this is normal, but apparently it isn't.

> BTW: which HDMI connector are you using? The one nearest to the USB-C (HDMI0) or the other one (HDMI1), or have you tried both? HDMI0 seems to be the more capable.

Yes, I'm using HDMI0, one next   to the usb-c power cable.

Any hints? Or should I ditch this installation and start new one from scratch?

BTW. this is the [monitor](https://www.ebay.co.uk/itm/Hannspree-HT161HNB-15-6-inch-LED-1366-x-768-12ms-Response-Speakers-HDMI/382903381495) I'm using.

ardera, 2019-11-19:
> No, I see completely nothing on my screen, no kernel buffer/messages, no raspbian startup, anything. The only think I can see is the `kmscube` if I start it over ssh. Other than that nothing, even login prompt is not visible. I did assumed that this is normal, but apparently it isn't.

Yeah, that's definitely not normal.

Could really be a Raspbian or firmware bug. If you look at the [firmware](https://github.com/raspberrypi/firmware) and [raspbian](https://github.com/raspberrypi/linux) repos, there are many people reporting some kinds of graphics problems/bugs. Maybe this is one of them. Monitors with a native 1366x768 resolution aren't that common, so it's possible that you're just the first one to notice it.

But to be sure I'd start completely from scratch. You could also try hardcoding 1080p in `/boot/config.txt` and see if that changes anything, or enabling hdmi safe mode or maybe some other compatibility option. (see [here](https://www.raspberrypi.org/documentation/configuration/config-txt/video.md))

dluksza, 2019-11-19:
OK, I've found two options that shows kernel buffer:
```
hdmi_group=2
hdmi_mode=16
```
and
```
hdmi_group=2
hdmi_mode=39
```
`kmscube` works (as it does always), but `flutter-pi` still shows nothing. I've installed `x.org` and `xinit`, then tried `startx` and this is also showing nothing. As you said this may be bug in firmware :/ Will give a try to Raspbian Desktop :|

Any other hints?

dluksza, 2019-11-19:
BTW. When I start either `flutter-pi` or X, the monitor shows 'no signal' and then switches into 'power saving mode'.

dluksza, 2019-11-19:
I've managed to make `startx` working by _forcing_ resolution for HDM-1 output to 1360x768. This is done in `.xsessionrc` file by `xrandr --output HDM-1 --mode 1360x768`. Now X server starts properly, but still `flutter-pi` doesn't work :/ Of course `kmscube` works always :D

ardera, 2019-11-20:
The package that you can install to upgrade you raspbian lite to a raspbian desktop is called `raspberrypi-ui-mods`. If you install that and then reboot, normal raspbian desktop should show up.

> BTW. When I start either `flutter-pi` or X, the monitor shows 'no signal' and then switches into 'power saving mode'.

That means there must be a bug somewhere. The Raspberry Pi doesn't have the ability to switch off the HDMI signal, even if it wanted to. (It says so on [this](https://www.raspberrypi.org/documentation/configuration/config-txt/video.md) page under the `hdmi_blanking` option)

> 2\. kernel buffer (or frame buffer) doesn't work, I do see 4 raspberries in top left corner, then a raspbian splashscreen for a split of second and that's it, display blinks a few times and switches to _powersave_ mode

If splash screen is enabled, that behaviour is normal (up to the point where the display has no signal anymore). I too see only 4 raspberry's (no kernel buffer or raspbian startup log) and then the splash screen on startup, if I have splash screen enabled.

I think it's best to report this & all the things we've found out as a bug report on https://github.com/raspberrypi/linux. I'd especially mention that your screen shows no signal, since that for sure means something is wrong.

closed 2019-11-21 by ardera