ardera/flutter-pi #22: Keyboard Support

limbouser, 2020-01-01:
I have an app that use keyboard navigation. Is it possible to take input from keyboard in Flutter-pi.?

geofmureithi, 2020-01-02:
Haven't tried it but seems like a good option:
https://pub.dev/packages/virtual_keyboard

limbouser, 2020-01-02:
> Haven't tried it but seems like a good option:
> https://pub.dev/packages/virtual_keyboard

I'm speaking about using the Pi Keyboard to navigate within the options in app. I am not interested in a virtual keyboard.

ardera, 2020-01-02:
Native keyboard input is not yet supported. Should not be that hard to do though. Also, this project is open source, so feel free to add keyboard support yourself if you want to.
For now, VirtualKeyboard really is the only way to get some kind of keyboard input in flutter-pi.

commit 78014822 (2020-01-31): added raw keyboard input
resolves #22
closed 2020-01-31 by ardera