ardera/flutter-pi #23: State not reflected when there is no user input

dluksza, 2020-01-11:
When `setState` is called from _code_ instead of on user action (like mouse click or screen tap), first execution is not reflected in the UI.

Here is counter project modified to automatically increment counter value every 2s: https://github.com/dluksza/flutter_auto_refresh

On android emulator everything works fine UI shows `0, 1, 2, 3, 4, 5` etc. every 2s. But when the same code is run using `flutter-pi` on Rpi3 it shows `0, 2, 4, 6` etc every 4s. whereas on console values `1, 2, 3, 4` etc. are printed every 2s.

This also means that other _automatic_ UI updates will not be reflected until another `setState` is called.

ardera, 2020-01-12:
That's interesting. I actually think this is not caused by the missing user action, but because of some kind of bug in the vsync-waiter. Could be that somehow frames get dropped when no other frames are scheduled at that moment, or somehow 2 frames get drawn immediately after each other, so that the first one isn't visible.

If you press the action button to increment the counter, there's a bunch of frames being rendered so one wouldn't notice if a few were missing.

I'll look into it

ardera, 2020-01-12:
Implemented vsync some time ago, just forgot to upload it. oops
that vsync code changed rendering a lot, and seems like one of these changes has fixed the bug.

closed 2020-01-12 by ardera

dluksza, 2020-01-12:
Thank you! Can confirm this is fixed on latest `master` :)

dluksza, 2020-01-14:
Unfortunately this seems to break touch on my monitor. FAB in the counter app doesn't work anymore. Going back to commit 530049f74 fixes the with tap, but then `setState` doesn't work :|

PR #24 (2020-01-12): implement vsync
- reduce tearing a lot (seems like some remains though, don't know why)
- fixes #23 
- reworked input-loop a bit