ardera/flutter-pi #25: Changing screen orientation doesn't have any effect

dluksza, 2020-01-12:
Standard way of changing flutter application orientation doesn't work on Rpi. Setting:
```dart
SystemChrome.setPreferredOrientations([
    DeviceOrientation.portraitUp,
]);
```
doesn't affect screen layout. I've also tried with `DeviceOrientation.landscapeLeft` and there is no difference.

ardera, 2020-01-13:
Yeah the `SystemChrome` class is basically just an interface to the platform method channel. Specifically, you're calling the `SystemChrome.setPreferredOrientations` method on the `flutter/platform` method channel, which is implemented right [here](https://github.com/ardera/flutter-pi/blob/master/src/plugins/services-plugin.c#L55).

Haven't got around to implement it yet. Should not be that hard to do though

dluksza, 2020-01-13:
Would be nice to have this implemented soonish ;)

closed 2020-01-14 by ardera

dluksza, 2020-01-14:
Awesome! Thank you!