ardera/flutter-pi #26: build time and building on non rasp pi

gedw99, 2020-01-14:
this looks really intzeresting for a project we have to build a secure collaboration system for volunteers. I already use flutter on other projects.

The main thing i need is webrtc.

I have a few high level questions:

Can i build in docker on my desktop ?

How long does the build take on a rasp pi 4 ?

Whats the approach to write a plugin. for example connectivity or Video / Audio ?
I use libwebrtc and can build for rapi pi currently.

ardera, 2020-01-14:
> Can i build in docker on my desktop ?

have not tried that yet, and I've never used docker before so I'm afraid I can't tell. It builds really fast though on the Pi 4, so there's probably no reason to cross-compile.

> How long does the build take on a rasp pi 4 ?

About 2 seconds on a Pi 4. The embedder is pretty lightweight.

> Whats the approach to write a plugin. for example connectivity or Video / Audio ?
> I use libwebrtc and can build for rapi pi currently.

That's a little bit more complicated. Depending on the library (more specifically, the C-features the library uses) it's also possible to build a dart interface to that library using the [dart FFI](https://dart.dev/guides/libraries/c-interop). That may be preferable, since communication via platform channels can be annoying at times, and isn't that fast. (40ms between message being sent from flutter and message being received in flutter-pi).

If you can't use the dart FFI, you have to build a plugin for flutter-pi. That shouldn't be that hard to do. I'll probably create a wiki page for that in this repo. If you don't want to wait for that, you can also look how at the implementation of the services-plugin (for example) which implements the platform-side of the [flutter services library](https://api.flutter.dev/flutter/services/services-library.html).

flutter-pi plugins are _statically compiled_ with the rest of the flutter-pi sources. If you want to create a new plugin, just create some source files somewhere in the flutter-pi sources and add these sources to the Makefile.

Your plugin only needs to implement 2 functions: an `init` and `deinit` function. Inside these you can register/unregister listeners to platform channels. You tell flutter-pi about your plugin by adding your
- plugin name
- reference to your _init_ function
- reference to your _deinit_ function

to the [`hardcoded_plugins` table in the pluginregistry file](https://github.com/ardera/flutter-pi/blob/ffd2e6f36c992567d1c9662d38175a94fdd85a6a/src/pluginregistry.c#L26).

The specifics of how to register/unregister listeners and how to respond, decode platform messages are documented in the [`pluginregistry.h`](https://github.com/ardera/flutter-pi/blob/master/include/pluginregistry.h) and [`platformchannel.h`](https://github.com/ardera/flutter-pi/blob/master/include/platformchannel.h) headers.

Keep in mind flutter-pi, although it's mostly stable, is not really production-ready.

gedw99, 2020-01-15:
Thanks @ardera 

That build times are really encouraging.
Dart ffi is what I am working on now.

Will get back to you once I have something worked up and need tips

closed 2020-01-15 by gedw99

ardera, 2020-01-15:
> That build times are really encouraging.
> Dart ffi is what I am working on now.

I am sorry I just tried using dart ffi with flutter-pi, it's not working. Outputs the following error:
```
[ERROR:engine/src/flutter/shell/common/shell.cc(199)]
  Dart Error: error: import of dart:ffi is not supported in the current Dart runtime
```

I just investigated and this is because dart ffi does not yet support ARM fully: https://github.com/dart-lang/sdk/issues/36309