ardera/flutter-pi #27: vsync support broke touch actions

dluksza, 2020-01-14:
It seems that adding vsync support fixed the `setState` related bugs, but it also breaks the touch input.

The same (example) counter app works fine on commit 530049f7, but then on touch actions doesn't work on commit 804363d49.

ardera, 2020-01-14:
That's weird. I just tested it when I pushed the latest commit and it worked fine. Did you also try the latest commit?
Could please attach the complete commandline output of running flutter-pi? Maybe there's some hint there.

dluksza, 2020-01-14:
Here is the full output
```
engine_argv[0] = ./flutter-pi/out/flutter-pi
initializing display...
Finding a suitable DRM device, since none is given...
looking for a suitable DRM device from 1 available DRM devices...
  devices[0]: 
    available nodes: DRM_NODE_PRIMARY, DRM_NODE_RENDER
    nodes[DRM_NODE_PRIMARY] = "/dev/dri/card0"
    nodes[DRM_NODE_RENDER] = "/dev/dri/renderD128"
    bustype: DRM_BUS_PLATFORM
    businfo.fullname: /soc/gpu
    opening DRM device candidate at "/dev/dri/card0"...
    getting resources of DRM device candidate at "/dev/dri/card0"...
    flutter-pi chose "/dev/dri/card0" as its DRM device.
Finding a connected connector from 1 available connectors...
  connectors[0]: connected? yes, type: 0x0B (HDMI-A), 340mm x 190mm
Choosing DRM mode from 26 available modes...
  modes[0]: name: "1366x768", 1366x768p, 60Hz, type: 72, flags: 5
    this mode is preferred by DRM. (DRM_MODE_TYPE_PREFERRED)
  modes[1]: name: "1920x1080", 1920x1080p, 60Hz, type: 64, flags: 5
  modes[2]: name: "1920x1080", 1920x1080p, 60Hz, type: 64, flags: 5
  modes[3]: name: "1920x1080", 1920x1080p, 50Hz, type: 64, flags: 5
  modes[4]: name: "1280x800", 1280x800p, 60Hz, type: 64, flags: 9
  modes[5]: name: "1280x720", 1280x720p, 60Hz, type: 64, flags: 5
  modes[6]: name: "1280x720", 1280x720p, 60Hz, type: 64, flags: 5
  modes[7]: name: "1280x720", 1280x720p, 50Hz, type: 64, flags: 5
  modes[8]: name: "1024x768", 1024x768p, 75Hz, type: 64, flags: 5
  modes[9]: name: "1024x768", 1024x768p, 70Hz, type: 64, flags: 10
  modes[10]: name: "1024x768", 1024x768p, 66Hz, type: 0, flags: 6
  modes[11]: name: "1024x768", 1024x768p, 60Hz, type: 64, flags: 10
  modes[12]: name: "832x624", 832x624p, 75Hz, type: 64, flags: 10
  modes[13]: name: "800x600", 800x600p, 75Hz, type: 64, flags: 5
  modes[14]: name: "800x600", 800x600p, 72Hz, type: 64, flags: 5
  modes[15]: name: "800x600", 800x600p, 60Hz, type: 64, flags: 5
  modes[16]: name: "800x600", 800x600p, 56Hz, type: 64, flags: 5
  modes[17]: name: "720x576", 720x576p, 50Hz, type: 64, flags: 10
  modes[18]: name: "720x480", 720x480p, 60Hz, type: 64, flags: 10
  modes[19]: name: "720x480", 720x480p, 60Hz, type: 64, flags: 10
  modes[20]: name: "640x480", 640x480p, 75Hz, type: 64, flags: 10
  modes[21]: name: "640x480", 640x480p, 73Hz, type: 64, flags: 10
  modes[22]: name: "640x480", 640x480p, 67Hz, type: 64, flags: 10
  modes[23]: name: "640x480", 640x480p, 60Hz, type: 64, flags: 10
  modes[24]: name: "640x480", 640x480p, 60Hz, type: 64, flags: 10
  modes[25]: name: "720x400", 720x400p, 70Hz, type: 64, flags: 6
Display properties:
  1366 x 768, 60Hz
  340mm x 190mm
  pixel_ratio = 1.057276
Finding DRM encoder...
Creating GBM device
Querying EGL client extensions...
Getting EGL display for GBM device...
Initializing EGL...
Querying EGL display extensions...
Using display 27468240 with EGL version 1.4
===================================
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless EGL_EXT_platform_device"
  display extensions: "EGL_ANDROID_blob_cache EGL_ANDROID_native_fence_sync EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_WL_bind_wayland_display "
===================================
Binding OpenGL ES API...
Choosing EGL config...
Finding EGL configs with appropriate attributes...
Creating EGL context...
Creating EGL window surface...
===================================
OpenGL ES information:
[... 6 lines cut ...]
Swapping buffers...
Locking front buffer...
getting new framebuffer for BO...
Setting CRTC...
Clearing current context...
finished display setup!
Initializing Application...
Initializing Plugin Registry...
Initialized Services plugin.
[elm327plugin] elm_open: process doesn't have access to serial device "/dev/rfcomm0": No such file or directory
[elm327plugin] ELM327Plugin_init: ELM327 communication was not initialized successfully. elm327plugin won't supply any OBDII data. error code: No such file or directory
[elm327plugin] running pid query queue processor
[ERROR:engine/src/flutter/shell/platform/embedder/embedder_surface_gl.cc(107)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
flutter: Observatory listening on http://127.0.0.1:33741/V7TyMnhO6dc=/
detected VideoCore IV as underlying graphics chip, and VC4 as the driver.
Reporting modified GL_EXTENSIONS string that doesn't contain non-working extensions.
[ERROR:engine/src/flutter/shell/platform/embedder/embedder_surface_gl.cc(107)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
flutter engine successfully started up.
Initializing Input devices...
  input device 0: path="/dev/input/event0"
      ILITEK ILITEK-TP, connected via USB. vendor: 0x222A, product: 0x222A, version: 0x19F7C50
Running IO thread...
Running message loop...
```
I don't recall changing anything else. I've also verified this on android emulator and FAB action works as expected.

ardera, 2020-01-15:
I have no idea what could be causing this. I'll create a new branch `debug-touch` soon where I'll revert some of the changes from 80436ed49ce30b6ac6e1abb367558a0a9561bccd, can you then try that branch and tell me if it works?

ardera, 2020-01-15:
Just [pushed a commit to debug-touch](https://github.com/ardera/flutter-pi/tree/debug-touch) adding some logging to `on_user_input`. Can you try that one a send me the full commandline output?

dluksza, 2020-01-15:
Here is the output:
```
pi@raspberrypi:~ $ ./flutter-pi/out/flutter-pi counter/
engine_argv[0] = ./flutter-pi/out/flutter-pi
initializing display...
Finding a suitable DRM device, since none is given...
looking for a suitable DRM device from 1 available DRM devices...
  devices[0]: 
    available nodes: DRM_NODE_PRIMARY, DRM_NODE_RENDER
    nodes[DRM_NODE_PRIMARY] = "/dev/dri/card0"
    nodes[DRM_NODE_RENDER] = "/dev/dri/renderD128"
    bustype: DRM_BUS_PLATFORM
    businfo.fullname: /soc/gpu
    opening DRM device candidate at "/dev/dri/card0"...
    getting resources of DRM device candidate at "/dev/dri/card0"...
    flutter-pi chose "/dev/dri/card0" as its DRM device.
Finding a connected connector from 1 available connectors...
  connectors[0]: connected? yes, type: 0x0B (HDMI-A), 340mm x 190mm
Choosing DRM mode from 26 available modes...
  modes[0]: name: "1366x768", 1366x768p, 60Hz, type: 72, flags: 5
    this mode is preferred by DRM. (DRM_MODE_TYPE_PREFERRED)
  modes[1]: name: "1920x1080", 1920x1080p, 60Hz, type: 64, flags: 5
  modes[2]: name: "1920x1080", 1920x1080p, 60Hz, type: 64, flags: 5
  modes[3]: name: "1920x1080", 1920x1080p, 50Hz, type: 64, flags: 5
  modes[4]: name: "1280x800", 1280x800p, 60Hz, type: 64, flags: 9
  modes[5]: name: "1280x720", 1280x720p, 60Hz, type: 64, flags: 5
  modes[6]: name: "1280x720", 1280x720p, 60Hz, type: 64, flags: 5
  modes[7]: name: "1280x720", 1280x720p, 50Hz, type: 64, flags: 5
  modes[8]: name: "1024x768", 1024x768p, 75Hz, type: 64, flags: 5
  modes[9]: name: "1024x768", 1024x768p, 70Hz, type: 64, flags: 10
  modes[10]: name: "1024x768", 1024x768p, 66Hz, type: 0, flags: 6
  modes[11]: name: "1024x768", 1024x768p, 60Hz, type: 64, flags: 10
  modes[12]: name: "832x624", 832x624p, 75Hz, type: 64, flags: 10
  modes[13]: name: "800x600", 800x600p, 75Hz, type: 64, flags: 5
  modes[14]: name: "800x600", 800x600p, 72Hz, type: 64, flags: 5
  modes[15]: name: "800x600", 800x600p, 60Hz, type: 64, flags: 5
  modes[16]: name: "800x600", 800x600p, 56Hz, type: 64, flags: 5
  modes[17]: name: "720x576", 720x576p, 50Hz, type: 64, flags: 10
  modes[18]: name: "720x480", 720x480p, 60Hz, type: 64, flags: 10
  modes[19]: name: "720x480", 720x480p, 60Hz, type: 64, flags: 10
  modes[20]: name: "640x480", 640x480p, 75Hz, type: 64, flags: 10
  modes[21]: name: "640x480", 640x480p, 73Hz, type: 64, flags: 10
  modes[22]: name: "640x480", 640x480p, 67Hz, type: 64, flags: 10
  modes[23]: name: "640x480", 640x480p, 60Hz, type: 64, flags: 10
  modes[24]: name: "640x480", 640x480p, 60Hz, type: 64, flags: 10
  modes[25]: name: "720x400", 720x400p, 70Hz, type: 64, flags: 6
Display properties:
  1366 x 768, 60Hz
  340mm x 190mm
  pixel_ratio = 1.057276
Finding DRM encoder...
Creating GBM device
Querying EGL client extensions...
Getting EGL display for GBM device...
Initializing EGL...
Querying EGL display extensions...
Using display 20111840 with EGL version 1.4
===================================
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless EGL_EXT_platform_device"
  display extensions: "EGL_ANDROID_blob_cache EGL_ANDROID_native_fence_sync EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_WL_bind_wayland_display "
===================================
Binding OpenGL ES API...
Choosing EGL config...
Finding EGL configs with appropriate attributes...
Creating EGL context...
Creating EGL window surface...
===================================
[... 10 lines cut ...]
Setting CRTC...
Clearing current context...
finished display setup!
Initializing Application...
Initializing Plugin Registry...
Initialized Services plugin.
[elm327plugin] elm_open: process doesn't have access to serial device "/dev/rfcomm0": No such file or directory
[elm327plugin] ELM327Plugin_init: ELM327 communication was not initialized successfully. elm327plugin won't supply any OBDII data. error code: No such file or directory
[elm327plugin] running pid query queue processor
[ERROR:engine/src/flutter/shell/platform/embedder/embedder_surface_gl.cc(107)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
detected VideoCore IV as underlying graphics chip, and VC4 as the driver.
Reporting modified GL_EXTENSIONS string that doesn't contain non-working extensions.
[ERROR:engine/src/flutter/shell/platform/embedder/embedder_surface_gl.cc(107)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
flutter engine successfully started up.
[... cut ...]
  input device 0: path="/dev/input/event0"
      ILITEK ILITEK-TP, connected via USB. vendor: 0x222A, product: 0x222A, version: 0x12F3C50
Running IO thread...
Running message loop...
flutter: Observatory listening on http://127.0.0.1:44879/G8ve2jHK15o=/
```

Tap actions still doesn't work. Also nothing is logged when I tap on the screen.

Just out of curiosity, I've checked out commit 530049f74739c6d80c74c06fe45a60b2d7bec703 and FAB magically started working again, no other changes were required.

ardera, 2020-01-15:
nvm I think I found it. Was an off-by-one error while preparing the `nfds` arg for the `select` syscall, basically the input device that was opened last was never checked for new data. Didn't appear on my Pi because I have multiple input devices attached, and the one that was opened last wasn't my touchscreen. Gonna a push a fix to master soon

closed 2020-01-15 by ardera

dluksza, 2020-01-15:
Yeah!! Now it works! Also with the rotated screen! Thank you! :D

ardera, 2020-01-16:
no problem :smile:

commit 00e1415c (2020-01-15): fix off-by-one error while preparing input fd-set
fixes #27