ardera/flutter-pi #28: Segfault when pressing any keyboard key

dluksza, 2020-01-26:
When running my app on `flutter-pi` commit `e8f86132` pressing any keyboard key result in segfault error and application crash.

Within my app I don't have any text input fields. I've discovered this because wanted to switch back to text console.

What I also noticed is that mouse doesn't work. Both keyboard and mouse are connected to rpi at the same time (occupying 2 out of 4 USB ports). But now of them work.

Here is the output from running `flutter-pi`
```
engine_argv[0] = flutter-pi/out/flutter-pi
initializing display...
Finding a suitable DRM device, since none is given...
looking for a suitable DRM device from 1 available DRM devices...
  devices[0]: 
    available nodes: DRM_NODE_PRIMARY, DRM_NODE_RENDER
    nodes[DRM_NODE_PRIMARY] = "/dev/dri/card0"
    nodes[DRM_NODE_RENDER] = "/dev/dri/renderD128"
    bustype: DRM_BUS_PLATFORM
    businfo.fullname: /soc/gpu
    opening DRM device candidate at "/dev/dri/card0"...
    getting resources of DRM device candidate at "/dev/dri/card0"...
    flutter-pi chose "/dev/dri/card0" as its DRM device.
Finding a connected connector from 1 available connectors...
  connectors[0]: connected? yes, type: 0x0B (HDMI-A), 1600mm x 900mm
Choosing DRM mode from 23 available modes...
  modes[0]: name: "1920x1080", 1920x1080p, 60Hz, type: 72, flags: 5
    this mode is preferred by DRM. (DRM_MODE_TYPE_PREFERRED)
  modes[1]: name: "1920x1080", 1920x1080p, 60Hz, type: 64, flags: 5
  modes[2]: name: "1920x1080", 1920x1080p, 50Hz, type: 64, flags: 5
  modes[3]: name: "1920x1080", 1920x1080p, 30Hz, type: 64, flags: 5
  modes[4]: name: "1920x1080", 1920x1080p, 30Hz, type: 64, flags: 5
  modes[5]: name: "1920x1080", 1920x1080p, 25Hz, type: 64, flags: 5
  modes[6]: name: "1920x1080", 1920x1080p, 24Hz, type: 64, flags: 5
  modes[7]: name: "1920x1080", 1920x1080p, 24Hz, type: 64, flags: 5
  modes[8]: name: "1280x1024", 1280x1024p, 60Hz, type: 64, flags: 5
  modes[9]: name: "1360x768", 1360x768p, 60Hz, type: 64, flags: 5
  modes[10]: name: "1152x864", 1152x864p, 60Hz, type: 0, flags: 6
  modes[11]: name: "1280x720", 1280x720p, 60Hz, type: 64, flags: 9
  modes[12]: name: "1280x720", 1280x720p, 60Hz, type: 64, flags: 5
  modes[13]: name: "1280x720", 1280x720p, 60Hz, type: 64, flags: 5
  modes[14]: name: "1280x720", 1280x720p, 50Hz, type: 64, flags: 5
  modes[15]: name: "1024x768", 1024x768p, 60Hz, type: 64, flags: 10
  modes[16]: name: "800x600", 800x600p, 60Hz, type: 64, flags: 5
  modes[17]: name: "720x576", 720x576p, 50Hz, type: 64, flags: 10
  modes[18]: name: "720x480", 720x480p, 60Hz, type: 64, flags: 10
  modes[19]: name: "720x480", 720x480p, 60Hz, type: 64, flags: 10
  modes[20]: name: "640x480", 640x480p, 60Hz, type: 64, flags: 10
  modes[21]: name: "640x480", 640x480p, 60Hz, type: 64, flags: 10
  modes[22]: name: "720x400", 720x400p, 70Hz, type: 64, flags: 6
Display properties:
  1920 x 1080, 60Hz
  1600mm x 900mm
  pixel_ratio = 1.000000
Finding DRM encoder...
Creating GBM device
Querying EGL client extensions...
Getting EGL display for GBM device...
Initializing EGL...
Querying EGL display extensions...
Using display 8651104 with EGL version 1.4
===================================
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless EGL_EXT_platform_device"
  display extensions: "EGL_ANDROID_blob_cache EGL_ANDROID_native_fence_sync EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_WL_bind_wayland_display "
===================================
Binding OpenGL ES API...
Choosing EGL config...
Finding EGL configs with appropriate attributes...
Creating EGL context...
Creating EGL window surface...
===================================
OpenGL ES information:
  version: "OpenGL ES 2.0 Mesa 19.2.0-rc1"
  shading language version: "OpenGL ES GLSL ES 1.0.16"
  vendor: "Broadcom"
  renderer: "VC4 V3D 2.1"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_npot GL_OES_vertex_half_float GL_OES_EGL_image GL_OES_depth_texture GL_AMD_performance_monitor GL_OES_packed_depth_stencil GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_EXT_occlusion_query_boolean GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_separate_shader_objects GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_texture_border_clamp GL_KHR_context_flush_control GL_OES_draw_elements_base_vertex GL_OES_texture_border_clamp GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_KHR_parallel_shader_compile GL_MESA_tile_raster_order "
===================================
Swapping buffers...
Locking front buffer...
getting new framebuffer for BO...
Setting CRTC...
Clearing current context...
finished display setup!
Initializing Application...
Initializing Plugin Registry...
Initialized Services plugin.
[elm327plugin] elm_open: process doesn't have access to serial device "/dev/rfcomm0": No such file or directory
[elm327plugin] ELM327Plugin_init: ELM327 communication was not initialized successfully. elm327plugin won't supply any OBDII data. error code: No such file or directory
[elm327plugin] running pid query queue processor
[ERROR:engine/src/flutter/shell/platform/embedder/embedder_surface_gl.cc(107)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
detected VideoCore IV as underlying graphics chip, and VC4 as the driver.
Reporting modified GL_EXTENSIONS string that doesn't contain non-working extensions.
[ERROR:engine/src/flutter/shell/platform/embedder/embedder_surface_gl.cc(107)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
flutter engine successfully started up.
Initializing Input devices...
  input device 0: path="/dev/input/event0"
        USB Keyboard, connected via USB. vendor: 0x04D9, product: 0x04D9, version: 0x805C98
  input device 1: path="/dev/input/event1"
        USB Keyboard System Control, connected via USB. vendor: 0x04D9, product: 0x04D9, version: 0x805C98
  input device 2: path="/dev/input/event2"
        USB Keyboard Consumer Control, connected via USB. vendor: 0x04D9, product: 0x04D9, version: 0x805C98
  input device 3: path="/dev/input/event3"
      Logitech USB Optical Mouse, connected via USB. vendor: 0x046D, product: 0x046D, version: 0x805C98
Running IO thread...
Running message loop...
```

ardera, 2020-01-26:
Yeah sorry, forgot to initialize a variable to zero.

closed 2020-01-26 by ardera

dluksza, 2020-01-26:
OK, now `flutter-pi` doesn't crash, but keyboard input goes to the console. Also mouse still doesn't work, I can't even see mouse pointer. I've tried moved mouse and clicked both buttons many times but nothing changes. :|

ardera, 2020-01-26:
mouse support is working, mouse cursor support not 😅

The DRM implementation of the Raspberry Pi doesn't yet support (hardware) mouse cursors. It's of course possible to build a workaround, but that requires some fundamental changes to the way flutter-pi works right now. Flutter-pi is really simple right now. There's only one layer (flutter) that's being redrawn whenever it needs to. To add mouse cursor support, you'd need to introduce a new layer, and also a composer to interleave those two. Interleaving those two would of course be trivial, but the composer is not thaat trivial, since the composer API provided by the flutter engine also needs to support very complicated ways of interleaving platform contents with flutter contents.

Haven't looked into it that much though. Could be that there is an easier workaround too. I've seen this a low-prio in the past & actually my plan up until now was to wait until the Raspberry Pi team adds hardware cursor support. I though noone would really use flutter with a mouse 😄

btw my method of verifying that mouse support works was by adding `printf("x: %f" . . .)` calls to `on_user_input`

dluksza, 2020-01-26:
I understand that this is something that isn't trivial. I actually only need mouse pointer for testing. My main goal is to use `flutter-pi` with a touch screen.

Nevertheless, I'm guessing that tapping on the TextField and typing from keyboard will also not work, since keyboard is still _connected_ to text terminal.

ardera, 2020-01-26:
> Nevertheless, I'm guessing that tapping on the TextField and typing from keyboard will also not work,

Keyboard Input will not work, because of #22.

> since keyboard is still _connected_ to text terminal.

text terminal and application are the same entity in that respect, so you can't "disconnect" a keyboard from a terminal and "connect" it to an application.
the keyboard also _needs_ to be attached to terminal in order for text input to work. stdin is the only way to get text input (not keyboard/keycode input!) without manually parsing the Xkb config file, loading & applying keymaps.

commit 54069804 (2020-01-26): fixes #28