ardera/flutter-pi #29: mouse cursor support

ardera, 2020-01-26:
- wait for Raspberry Pi foundation to add hw cursor support to DRM, or
- use the embedder API to add a new layer on top of flutter (also a composer), or
- maybe there's some other way to get DRM to draw a cursor on top of the other planes, or
- build a package for flutter that provides software cursor support

ardera, 2020-05-10:
I thought about this for a while, but I'm not sure it's a wise choice to implement it.

The thing is, if I add mouse support, it will be without _mouse acceleration_ and without _any configuration options_. So the mouse will feel kinda raw, and not at all like what you're used to on a normal Desktop.

That means mouse support would only be useful for debugging purposes. But since touchscreen input seems to work quite well right now, I doubt anyone needs mouse input for debugging purposes.

That's why I think it's not that useful right now.

But of course, I'm open to discussion about that.

GioPan04, 2020-05-10:
Hi, in my opinion the mouse support is needed. How you said only for debugging purposes. For example, I don't have a touch screen display yet, and at the moment I'm stuck at the first screen because I can't interact with the app.

TL;DR
We need the mouse support

vcrtech, 2020-05-20:
> Hi, in my opinion the mouse support is needed. How you said only for debugging purposes. For example, I don't have a touch screen display yet, and at the moment I'm stuck at the first screen because I can't interact with the app.
> 
> TL;DR
> We need the mouse support

Agree. I'm currently investigating a borrowed RPI as a possible path forward on a project, but I'm not quite ready to make the investment on a touchscreen yet. Even if it's kind of nasty, basic mouse support would be nice to see how smooth/interactive flutter apps can be for those without touchscreens. Please consider it?

ardera, 2020-05-24:
Okay, that seems reasonable, I'll implement it when I have time

CAoTx, 2020-07-16:
> Okay, that seems reasonable, I'll implement it when I have time

That would be really nice, Thanks!

closed 2020-08-05 by ardera

ondrej1024, 2022-10-03:
So just to make sure: is mouse control supported with the latest flutter_pi release? Anything that needs to be configured to enable it ?
It doesn't seem to work here.

ardera, 2022-10-03:
It only works when the GPU supports hardware mouse cursors. If they're not supported you should see some error messages in the flutter-pi output.

You can use a software mouse cursor though when hw is not supported: https://gist.github.com/ardera/e7ddf7c8e477d258a26dfa9c40002626

PR #72 (2020-08-05): mouse cursor enhancements
- multiple cursor sizes
- 32x32, 48x48, 64x64, 96x96, 128x128
- cursor dpi scaling behaviour similiar to Windows
- cursor hotspot fixup
- rotate cursor in software
- closes #29