ardera/flutter-pi #30: > Nevertheless, I'm guessing that tapping on the TextField and typing from keyboard will also not work,

limbouser, 2020-01-29:
> Nevertheless, I'm guessing that tapping on the TextField and typing from keyboard will also not work,

Keyboard Input will not work, because of #22.

> since keyboard is still _connected_ to text terminal.

text terminal and application are the same entity in that respect, so you can't "disconnect" a keyboard from a terminal and "connect" it to an application.
the keyboard also _needs_ to be attached to terminal in order for text input to work. stdin is the only way to get text input (not keyboard/keycode input!) without manually parsing the Xkb config file, loading & applying keymaps.

_Originally posted by @ardera in https://github.com/ardera/flutter-pi/issues/28#issuecomment-578545274_

Can I have some more guidance on how I can add keyboard support? I am really interested in getting it done.

ardera, 2020-01-29:
I've just uploaded my current work on keyboard input in branch [feature-keyboard-input](https://github.com/ardera/flutter-pi/tree/feature-keyboard-input). Though I've only worked on text input yet, not raw keyboard input.

In #22, you mentioned your App uses Keyboard navigation? That means somewhere in your App there's a `RawKeyboardListener` listening for key events like `arrow key up`, `arrow key down`, right?

ardera, 2020-01-31:
keyboard input was added #31

closed 2020-01-31 by ardera