ardera/flutter-pi #33: 64bit Ubuntu 19.10 support?

Radagan, 2020-02-10:
I just found this project and I would very much like to add a Flutter based UI to my code running on RPi 3B+ and RPi 4 systems under Ubuntu 19.10 (64bit). I read in your docs that 64bit support would be easy but isn't there yet. Is this possible with flutter-pi in the near future?

ardera, 2020-02-10:
That's right, I can implement 64 bit support if you want
But are you running your ubuntu in headless mode? (i.e. ubuntu server, or ubuntu desktop with some modifications) If not, this project is not for you.
`flutter-pi` targets use cases where you want to run your flutter app fullscreen without any desktop environment running in the background (X11, Wayland). Otherwise, just use official linux desktop embedding and enjoy better plugin support.
You also need to make sure Ubuntu has hardware acceleration for graphics, that's a requirement for `flutter-pi`.

Radagan, 2020-02-10:
You have it exactly--I'm running headless. My intention is to have a local touchscreen display for a dedicated application running on a RPi using the official 7" touchscreen from Raspberry Pi. The application is written in C++, so dart/ffi will also be a part of the challenge...

I will have to check on hardware acceleration for graphics, but I believe it's in there. Ubuntu 19.10 has very good support for the pi, but I've only recently moved up from Ubuntu 19.04.

Radagan, 2020-02-10:
> Actually, this project should really be called `flutter-embedded-linux` because it doesn't really feature anything Pi-specific other than some hacks for OpenGL and the fact that I am testing it on a Pi 4.

That's very exciting! My application is only running on the RPi because it's a great board to prototype on--inexpensive, powerful, etc. But my intent is general embedded Linux.

ardera, 2020-02-10:
Okay, I'll start working on 64bit support then

> so dart/ffi will also be a part of the challenge...

I'd double check if `dart:ffi` works on your platform. Last time I tried (albeit on Raspbian) it failed because the arm hardfloat ABI was not supported at that point. It's supported now, but it probably hasn't landed in flutter stable yet. The check is simple, if it's not supported it will fail at the import of the `dart:ffi` library.

Radagan, 2020-02-17:
Thank you for being willing to add this. I don't mind working on dev or beta channels as this is a proof of concept for a project.

ardera, 2020-02-26:
should work now, but I didn't test it.

closed 2020-02-26 by ardera

Radagan, 2020-02-26:
Thank you, Hannes! 😎👍🏻

ardera, 2020-02-26:
no problem 😄