ardera/flutter-pi #34: Building on an embedded Linux Platform

limbouser, 2020-02-12:
I tried building flutter-pi on Raspbian and it worked. But when I tried to do the same on another embedded platform providing the necessary dependencies, an error stating that vc4_dri.so and kms_swrast_dri.so are not found came up. Isn't X11 a necessary dependency for building those? I tried building those SO's but get blocked since X11 libraries are not present. Mesa provides those SO's. When I checked, some libraries of X11 are also getting built in Raspberry Pi too.

ardera, 2020-02-12:
> an error stating that vc4_dri.so and kms_swrast_dri.so are not found came up. Isn't X11 a necessary dependency for building those?

`vc4_dri.so` is the userspace part of the VideoCore 4 driver. X11 is not a dependency for `vc4_dri.so` at runtime.

Mesa probably tries to load `kms_swrast_dri.so` (a software rasterizer) after that because the real driver (vc4) couldn't be loaded.

`vc4_dri.so` and `kms_swrast_dri.so` are both provided by the `libgl1-mesa-dri` package, so you can try installing that. I'm not sure why this wasn't installed too when you installed the `flutter-pi` dependencies.

> I tried building those SO's but get blocked since X11 libraries are not present. Mesa provides those SO's. When I checked, some libraries of X11 are also getting built in Raspberry Pi too.

Just because you need X11 libs at build-time for mesa, doesn't mean you need them at runtime too. If you run `ldd /path/to/flutter-pi` to check the run-time shared library dependencies of the `flutter-pi` binary, you'll see this:

```bash
pi@hpi4:~/devel $ ldd ./flutter-pi/out/flutter-pi
        linux-vdso.so.1 (0xbef74000)
        /usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0xb6fa0000)
        libgbm.so.1 => /usr/lib/arm-linux-gnueabihf/libgbm.so.1 (0xb6f70000)
        libdrm.so.2 => /usr/lib/arm-linux-gnueabihf/libdrm.so.2 (0xb6f51000)
        libGLESv2.so.2 => /usr/lib/arm-linux-gnueabihf/libGLESv2.so.2 (0xb6f32000)
        libEGL.so.1 => /usr/lib/arm-linux-gnueabihf/libEGL.so.1 (0xb6f13000)
        librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0xb6efc000)
        libflutter_engine.so => /usr/lib/libflutter_engine.so (0xb584d000)
        libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0xb5823000)
        libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0xb5810000)
        libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xb56c2000)
        libwayland-server.so.0 => /usr/lib/arm-linux-gnueabihf/libwayland-server.so.0 (0xb56a5000)
        libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0xb5623000)
        libexpat.so.1 => /lib/arm-linux-gnueabihf/libexpat.so.1 (0xb55e2000)
        /lib/ld-linux-armhf.so.3 (0xb6fb5000)
        libGLdispatch.so.0 => /usr/lib/arm-linux-gnueabihf/libGLdispatch.so.0 (0xb5563000)
        libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0xb5536000)
        libffi.so.6 => /usr/lib/arm-linux-gnueabihf/libffi.so.6 (0xb551e000)
```
(ldd lists transitive dependencies too)

There's no X11 libs in there. There _is_ `libwayland-server.so` though, that's probably because the mesa libs provided by the raspbian buster package repo were built with `--configure=drm,wayland`. (See [here](https://bbs.archlinux.org/viewtopic.php?id=216166))

If you don't want to have `libwayland-server.so` on your system, you probably need to compile mesa yourself with proper configuration. Maybe you can use [buildroot](https://buildroot.org/) for that.

limbouser, 2020-02-13:
Thanks for the help.

closed 2020-02-13 by limbouser