ardera/flutter-pi #35: Setting CRTC error

limbouser, 2020-02-13:
I am trying to build flutter-pi on a platform other than Raspbian. I got stuck at the following point.

`Swapping buffers...
Locking front buffer...
getting new framebuffer for BO...
Setting CRTC...
failed to set mode: Permission denied`

Why would this come up? Do you have any idea?

ardera, 2020-02-13:
Two possibilities that come to my mind:

- either you have insufficient permissions to access the device files in `/dev/dri/`. Try running `flutter-pi` as root or adding your user to the necessary groups. (2nd one is system-dependant, can't give you instructions for that)
- some other application is the DRM master and opened the device files under `/dev/dri/` before `flutter-pi` did. That could be an x-server, wayland, or some other device using graphics.

ardera, 2020-02-26:
Any news on this? If not I'll close this issue.

limbouser, 2020-02-26:
Yh. I'm closing it. This came up whenever I used an image with an UI even though minimal. Whatever, thanks for the help.

limbouser, 2020-03-23:
`drm_mode_config_cleanup` this is something that cameup when I was navigating on some help with this issue. Is their anyway this can be useful to you such that flutter-pi if run on a platform with an application using graphics can free it up and start the flutter application?

ardera, 2020-03-23:
Thanks for the input. I'm afraid `drm_mode_config_cleanup` is not a solution though. That function is defined inside the linux kernel to be used by drivers. It's not available to userspace.

> This came up whenever I used an image with an UI even though minimal. Whatever, thanks for the help.

I just now understood what you meant by this 😄 Using any linux OS with an enabled graphical user interface will probably not work, because of the 2nd thing I mentioned [in my previous comment](https://github.com/ardera/flutter-pi/issues/35#issuecomment-585764167). You need to disable the UI and make sure no X11- or wayland-server is starting up. There are many guides on that online. (Just google `<your distro> boot to console` or something like that)

This _may_ work, but I'm not sure:
```bash
sudo systemctl set-default multi-user.target --force
```

If you need graphical UI, you can't use flutter-pi. Consider using the [official linux desktop embedding for flutter](https://github.com/flutter/flutter/wiki/Desktop-shells).

closed 2020-04-25 by ardera