ardera/flutter-pi #37: ELM327 Plugin

limbouser, 2020-02-14:
What is the purpose of ELM327 plugin? Is it to let the app have bluetooth communication? Is it the ELM327 plugin that needs /dev/rfcomm0? Can the flutter-pi run without the same? Can ELM327  plugin be disabled by me if itsn't necessary?

ardera, 2020-02-14:
ELM327 plugin is a plugin for communicating with an [ELM327](https://en.wikipedia.org/wiki/ELM327). Basically the plugin allows you the query vehicle speed, RPM etc from an ELM327 connected via Bluetooth. I use it because I want to use flutter-pi in my car.

I'm gonna remove it soon anyways, but it can be disabled by removing the `-DBUILD_ELM327_PLUGIN` flag from the `Makefile`.

limbouser, 2020-02-26:
Yh. It worked. Thanks.

closed 2020-02-26 by limbouser