ardera/flutter-pi #38: Issue with external HDMI screen

ekuusi, 2020-02-17:
I am having the same issue as in https://github.com/ardera/flutter-pi/issues/16

That is, flutter-pi shows a blank screen when I try to start the app when the Pi is connected to this display: https://www.waveshare.com/10.1inch-HDMI-LCD-B-with-case.htm

Connecting the Raspberry Pi to an hp 1080p monitor, everything works fine, including the touch screen panel from the above Waveshare display, so I can control the app fine by touching the display when I pipe the image to the HP monitor.

I'm running flutter-pi on Raspberry Pi 4 with the latest version of Raspbian Buster Desktop. I've modified it to boot to command line and then running flutter-pi there. The Waveshare display works otherwise fine, I'm able to for example boot to chromium in kiosk-mode and run a Flutter for Web app on it fine.

Any ideas how to get the display to work on flutter-pi? My expertise on such hardware issues is limited so it's a bit challenging to troubleshoot this one.

Here's the log when starting the app from SSH:

```
pi@raspberrypi:~/flutter-pi/out $ ./flutter-pi /home/pi/uikit/flutter_assets
engine_argv[0] = ./flutter-pi
initializing display...
Finding a suitable DRM device, since none is given...
looking for a suitable DRM device from 2 available DRM devices...
  devices[0]:
    available nodes: DRM_NODE_PRIMARY,
    nodes[DRM_NODE_PRIMARY] = "/dev/dri/card1"
    bustype: DRM_BUS_PLATFORM
    businfo.fullname: /soc/gpu
    opening DRM device candidate at "/dev/dri/card1"...
    getting resources of DRM device candidate at "/dev/dri/card1"...
    flutter-pi chose "/dev/dri/card1" as its DRM device.
  devices[1]:
    available nodes: DRM_NODE_PRIMARY, DRM_NODE_RENDER
    nodes[DRM_NODE_PRIMARY] = "/dev/dri/card0"
    nodes[DRM_NODE_RENDER] = "/dev/dri/renderD128"
    bustype: DRM_BUS_PLATFORM
    businfo.fullname: /v3dbus/v3d@7ec04000
Finding a connected connector from 1 available connectors...
  connectors[0]: connected? yes, type: 0x0B (HDMI-A), 150mm x 100mm
Choosing DRM mode from 2 available modes...
  modes[0]: name: "1280x800", 1280x800p, 63Hz, type: 72, flags: 10
    this mode is preferred by DRM. (DRM_MODE_TYPE_PREFERRED)
  modes[1]: name: "1280x800", 1280x800p, 60Hz, type: 32, flags: 6
Display properties:
  1280 x 800, 63Hz
  150mm x 100mm
  pixel_ratio = 2.245614
Finding DRM encoder...
Creating GBM device
Querying EGL client extensions...
Getting EGL display for GBM device...
Initializing EGL...
Querying EGL display extensions...
Using display 0x5cc538 with EGL version 1.4
===================================
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless EGL_EXT_platform_device"
  display extensions: "EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
Binding OpenGL ES API...
Choosing EGL config...
Finding EGL configs with appropriate attributes...
Creating EGL context...
Creating EGL window surface...
===================================
OpenGL ES information:
  version: "OpenGL ES 3.0 Mesa 19.2.0-rc1"
  shading language version: "OpenGL ES GLSL ES 3.00"
  vendor: "Broadcom"
  renderer: "V3D 4.2"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_texture_border_clamp GL_KHR_context_flush_control GL_OES_draw_elements_base_vertex GL_OES_texture_border_clamp GL_OES_texture_stencil8 GL_EXT_float_blend GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_MESA_shader_integer_functions GL_KHR_parallel_shader_compile GL_EXT_texture_query_lod "
===================================
Swapping buffers...
Locking front buffer...
getting new framebuffer for BO...
Setting CRTC...
Clearing current context...
finished display setup!
Initializing Application...
Initializing Plugin Registry...
[services-plugin] init.
[text_input] init.
[raw_keyboard] init.
[test-plugin] init.
[elm327plugin] elm_open: process doesn't have access to serial device "/dev/rfcomm0": No such file or directory
[elm327plugin] ELM327Plugin_init: ELM327 communication was not initialized successfully. elm327plugin won't supply any OBDII data. error code: No such file or directory
[elm327plugin] running pid query queue processor
[ERROR:engine/src/flutter/shell/platform/embedder/embedder_surface_gl.cc(107)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
flutter: Observatory listening on http://127.0.0.1:35431/U7qsOz2kkTw=/
[ERROR:engine/src/flutter/shell/platform/embedder/embedder_surface_gl.cc(107)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
flutter engine successfully started up.
Initializing Input devices...
  input device 0: path="/dev/input/event0"
      WaveShare WS170120, connected via USB. vendor: 0x0EEF, product: 0x0005, version: 0x0110
Running IO thread...
Running message loop...
```

ekuusi, 2020-02-17:
For reference, here is what I get when having the exact same setup, only this time my HDMI cable is connected to HP's 27er monitor: https://support.hp.com/us-en/product/hp-value-27-inch-displays/3989902/document/c05078501/

```
pi@raspberrypi:~ $ ./flutter-pi/out/flutter-pi /home/pi/uikit/flutter_assets
engine_argv[0] = ./flutter-pi/out/flutter-pi
initializing display...
Finding a suitable DRM device, since none is given...
looking for a suitable DRM device from 2 available DRM devices...
  devices[0]:
    available nodes: DRM_NODE_PRIMARY,
    nodes[DRM_NODE_PRIMARY] = "/dev/dri/card1"
    bustype: DRM_BUS_PLATFORM
    businfo.fullname: /soc/gpu
    opening DRM device candidate at "/dev/dri/card1"...
    getting resources of DRM device candidate at "/dev/dri/card1"...
    flutter-pi chose "/dev/dri/card1" as its DRM device.
  devices[1]:
    available nodes: DRM_NODE_PRIMARY, DRM_NODE_RENDER
    nodes[DRM_NODE_PRIMARY] = "/dev/dri/card0"
    nodes[DRM_NODE_RENDER] = "/dev/dri/renderD128"
    bustype: DRM_BUS_PLATFORM
    businfo.fullname: /v3dbus/v3d@7ec04000
Finding a connected connector from 1 available connectors...
  connectors[0]: connected? yes, type: 0x0B (HDMI-A), 600mm x 340mm
Choosing DRM mode from 19 available modes...
  modes[0]: name: "1920x1080", 1920x1080p, 60Hz, type: 72, flags: 5
    this mode is preferred by DRM. (DRM_MODE_TYPE_PREFERRED)
  modes[1]: name: "1920x1080", 1920x1080p, 60Hz, type: 64, flags: 5
  modes[2]: name: "1920x1080", 1920x1080p, 50Hz, type: 64, flags: 5
  modes[3]: name: "1680x1050", 1680x1050p, 60Hz, type: 64, flags: 9
  modes[4]: name: "1600x900", 1600x900p, 60Hz, type: 64, flags: 5
  modes[5]: name: "1280x1024", 1280x1024p, 60Hz, type: 64, flags: 5
  modes[6]: name: "1440x900", 1440x900p, 60Hz, type: 64, flags: 9
  modes[7]: name: "1280x800", 1280x800p, 60Hz, type: 64, flags: 9
  modes[8]: name: "1280x720", 1280x720p, 60Hz, type: 64, flags: 5
  modes[9]: name: "1280x720", 1280x720p, 60Hz, type: 64, flags: 5
  modes[10]: name: "1280x720", 1280x720p, 50Hz, type: 64, flags: 5
  modes[11]: name: "1024x768", 1024x768p, 60Hz, type: 64, flags: 10
  modes[12]: name: "800x600", 800x600p, 60Hz, type: 64, flags: 5
  modes[13]: name: "720x576", 720x576p, 50Hz, type: 64, flags: 10
  modes[14]: name: "720x480", 720x480p, 60Hz, type: 64, flags: 10
  modes[15]: name: "720x480", 720x480p, 60Hz, type: 64, flags: 10
  modes[16]: name: "640x480", 640x480p, 60Hz, type: 64, flags: 10
  modes[17]: name: "640x480", 640x480p, 60Hz, type: 64, flags: 10
  modes[18]: name: "720x400", 720x400p, 70Hz, type: 64, flags: 6
Display properties:
  1920 x 1080, 60Hz
  600mm x 340mm
  pixel_ratio = 1.000000
Finding DRM encoder...
Creating GBM device
Querying EGL client extensions...
Getting EGL display for GBM device...
Initializing EGL...
Querying EGL display extensions...
Using display 0x1069930 with EGL version 1.4
===================================
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless EGL_EXT_platform_device"
  display extensions: "EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
Binding OpenGL ES API...
Choosing EGL config...
Finding EGL configs with appropriate attributes...
Creating EGL context...
Creating EGL window surface...
===================================
OpenGL ES information:
[... 6 lines cut ...]
Swapping buffers...
Locking front buffer...
getting new framebuffer for BO...
Setting CRTC...
Clearing current context...
finished display setup!
Initializing Application...
Initializing Plugin Registry...
[services-plugin] init.
[text_input] init.
[raw_keyboard] init.
[test-plugin] init.
[elm327plugin] elm_open: process doesn't have access to serial device "/dev/rfcomm0": No such file or directory
[elm327plugin] ELM327Plugin_init: ELM327 communication was not initialized successfully. elm327plugin won't supply any OBDII data. error code: No such file or directory
[elm327plugin] running pid query queue processor
[ERROR:engine/src/flutter/shell/platform/embedder/embedder_surface_gl.cc(107)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
[ERROR:engine/src/flutter/shell/platform/embedder/embedder_surface_gl.cc(107)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
flutter engine successfully started up.
Initializing Input devices...
  input device 0: path="/dev/input/event0"
      WaveShare WS170120, connected via USB. vendor: 0x0EEF, product: 0x0005, version: 0x0110
Running IO thread...
Running message loop...
flutter: Observatory listening on http://127.0.0.1:44013/OCrkZlf9XYc=/
```

ekuusi, 2020-02-17:
Further testing:

If I change the resolution in the HP monitor to anything other than 1920x1080 in raspi-config -> advanced -> resolution, I get the same issue as with the WaveShare display: The screen goes dark, then the screen comes back on (you see the backlight going off and then on), but the screen remains black until I stop flutter-pi with ctrl+c. After that the command line appears back again.

ekuusi, 2020-02-17:
Tried with a Samsung 4K monitor, that works as well. So far the WaveShare panel is the only one not working.

```
pi@raspberrypi:~ $ ./flutter-pi/out/flutter-pi /home/pi/uikit/flutter_assets
engine_argv[0] = ./flutter-pi/out/flutter-pi
initializing display...
Finding a suitable DRM device, since none is given...
looking for a suitable DRM device from 2 available DRM devices...
  devices[0]:
    available nodes: DRM_NODE_PRIMARY,
    nodes[DRM_NODE_PRIMARY] = "/dev/dri/card1"
    bustype: DRM_BUS_PLATFORM
    businfo.fullname: /soc/gpu
    opening DRM device candidate at "/dev/dri/card1"...
    getting resources of DRM device candidate at "/dev/dri/card1"...
    flutter-pi chose "/dev/dri/card1" as its DRM device.
  devices[1]:
    available nodes: DRM_NODE_PRIMARY, DRM_NODE_RENDER
    nodes[DRM_NODE_PRIMARY] = "/dev/dri/card0"
    nodes[DRM_NODE_RENDER] = "/dev/dri/renderD128"
    bustype: DRM_BUS_PLATFORM
    businfo.fullname: /v3dbus/v3d@7ec04000
Finding a connected connector from 1 available connectors...
  connectors[0]: connected? yes, type: 0x0B (HDMI-A), 610mm x 350mm
Choosing DRM mode from 40 available modes...
  modes[0]: name: "3840x2160", 3840x2160p, 30Hz, type: 72, flags: 5
    this mode is preferred by DRM. (DRM_MODE_TYPE_PREFERRED)
  modes[1]: name: "3840x2160", 3840x2160p, 30Hz, type: 64, flags: 5
  modes[2]: name: "3840x2160", 3840x2160p, 25Hz, type: 64, flags: 5
  modes[3]: name: "3840x2160", 3840x2160p, 24Hz, type: 64, flags: 5
  modes[4]: name: "3840x2160", 3840x2160p, 24Hz, type: 64, flags: 5
  modes[5]: name: "2560x1440", 2560x1440p, 60Hz, type: 64, flags: 9
  modes[6]: name: "1920x1080", 1920x1080p, 60Hz, type: 64, flags: 5
  modes[7]: name: "1920x1080", 1920x1080p, 60Hz, type: 64, flags: 5
  modes[8]: name: "1920x1080", 1920x1080p, 50Hz, type: 64, flags: 5
  modes[9]: name: "1920x1080", 1920x1080p, 30Hz, type: 64, flags: 5
  modes[10]: name: "1920x1080", 1920x1080p, 30Hz, type: 64, flags: 5
  modes[11]: name: "1920x1080", 1920x1080p, 24Hz, type: 64, flags: 5
  modes[12]: name: "1920x1080", 1920x1080p, 24Hz, type: 64, flags: 5
  modes[13]: name: "1680x1050", 1680x1050p, 60Hz, type: 64, flags: 9
  modes[14]: name: "1600x900", 1600x900p, 60Hz, type: 64, flags: 5
  modes[15]: name: "1280x1024", 1280x1024p, 75Hz, type: 64, flags: 5
  modes[16]: name: "1280x1024", 1280x1024p, 60Hz, type: 64, flags: 5
  modes[17]: name: "1440x900", 1440x900p, 60Hz, type: 64, flags: 9
  modes[18]: name: "1280x800", 1280x800p, 60Hz, type: 64, flags: 9
  modes[19]: name: "1152x864", 1152x864p, 75Hz, type: 64, flags: 5
  modes[20]: name: "1280x720", 1280x720p, 60Hz, type: 64, flags: 5
  modes[21]: name: "1280x720", 1280x720p, 60Hz, type: 64, flags: 5
  modes[22]: name: "1280x720", 1280x720p, 50Hz, type: 64, flags: 5
  modes[23]: name: "1024x768", 1024x768p, 75Hz, type: 64, flags: 5
  modes[24]: name: "1024x768", 1024x768p, 70Hz, type: 64, flags: 10
  modes[25]: name: "1024x768", 1024x768p, 60Hz, type: 64, flags: 10
  modes[26]: name: "832x624", 832x624p, 75Hz, type: 64, flags: 10
  modes[27]: name: "800x600", 800x600p, 75Hz, type: 64, flags: 5
  modes[28]: name: "800x600", 800x600p, 72Hz, type: 64, flags: 5
  modes[29]: name: "800x600", 800x600p, 60Hz, type: 64, flags: 5
  modes[30]: name: "800x600", 800x600p, 56Hz, type: 64, flags: 5
  modes[31]: name: "720x576", 720x576p, 50Hz, type: 64, flags: 10
  modes[32]: name: "720x480", 720x480p, 60Hz, type: 64, flags: 10
  modes[33]: name: "720x480", 720x480p, 60Hz, type: 64, flags: 10
  modes[34]: name: "640x480", 640x480p, 75Hz, type: 64, flags: 10
  modes[35]: name: "640x480", 640x480p, 73Hz, type: 64, flags: 10
  modes[36]: name: "640x480", 640x480p, 67Hz, type: 64, flags: 10
  modes[37]: name: "640x480", 640x480p, 60Hz, type: 64, flags: 10
  modes[38]: name: "640x480", 640x480p, 60Hz, type: 64, flags: 10
  modes[39]: name: "720x400", 720x400p, 70Hz, type: 64, flags: 6
Display properties:
  3840 x 2160, 30Hz
  610mm x 350mm
  pixel_ratio = 1.656601
[... 27 lines cut ...]
Swapping buffers...
Locking front buffer...
getting new framebuffer for BO...
Setting CRTC...
Clearing current context...
finished display setup!
Initializing Application...
Initializing Plugin Registry...
[services-plugin] init.
[text_input] init.
[raw_keyboard] init.
[test-plugin] init.
[elm327plugin] elm_open: process doesn't have access to serial device "/dev/rfcomm0": No such file or directory
[elm327plugin] ELM327Plugin_init: ELM327 communication was not initialized successfully. elm327plugin won't supply any OBDII data. error code: No such file or directory
[elm327plugin] running pid query queue processor
[ERROR:engine/src/flutter/shell/platform/embedder/embedder_surface_gl.cc(107)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
flutter: Observatory listening on http://127.0.0.1:38099/-F5b1wdfC3c=/
[ERROR:engine/src/flutter/shell/platform/embedder/embedder_surface_gl.cc(107)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
flutter engine successfully started up.
Initializing Input devices...
  input device 0: path="/dev/input/event0"
      WaveShare WS170120, connected via USB. vendor: 0x0EEF, product: 0x0005, version: 0x0110
Running IO thread...
Running message loop...
```

ardera, 2020-02-17:
Just tested it, I got the same problem with my monitor, albeit with the default 1080p60 resolution of my monitor, so it seems it's not only when configuring a non-default HDMI mode. It definitely worked before, so there's probably a regression somewhere in `flutter-pi`. Should probably test `flutter-pi` with monitors more often. (Usually I'm using the Raspberry Pi 7" display)

Actually the resolution configuration using `raspi-config --> Advanced --> Resolution` is quite hacky. It seems that `raspi-config` just tells the Raspberry Pi firmware that it should configure the HDMI output for this resolution, but it doesn't tell the linux-side about it at all. I think it's possible to do it in a _non-hacky_ way as well by specifying a custom HDMI EDID file in `/boot/config.txt`, but that's not really relevant as HDMI config seems to not be the cause.

I think it's unrelated to #16 though, that issue was (or still is) caused by a raspbian kernel / firmware bug. See [here](https://github.com/raspberrypi/linux/issues/3221). He couldn't even boot into Raspbian Desktop without modifications.

ardera, 2020-02-17:
Okay the error seems to happen in [this line](https://github.com/ardera/flutter-pi/blob/370c8e90a7b3f1e234e30bc32121202a642c2263/src/flutter-pi.c#L518), which is part of the vsync mechanism.

Normally, `drmCrtcGetSequence` outputs the time of the last vblank in it's `ns` argument, in this case it doesn't and throws an `EBADF` instead, even though the specified file descriptor is _absolutely_ not an invalid file descriptor. (This results in flutter waiting for vsync for eternity, no frame ever gets displayed)

I think for now I can build a workaround that disables vsync if `drmCrtcGetSequence` does not work, though that'll result in a lot of tearing.

ardera, 2020-02-17:
Pushed the workaround.

Will have to do some investigation to see whether this is a kernel/firmware bug or some misconfiguration

ekuusi, 2020-02-18:
The screen works now, but there's a nasty side effect. I get this crash pretty soon after scrolling views in flutter:
```
failed to queue page flip: Device or resource busy
failed to queue page flip: Device or resource busy
failed to queue page flip: Device or resource busy
Segmentation fault
```

Doesn't happen always when scrolling, only when there's a hiccup for any reason (which happens pretty quick when using the UIKit demo I'm using for testing: https://github.com/iampawan/Flutter-UI-Kit ).  Doesn't happen on the monitors that worked before, only on the problematic one.

One quick question while discussing displays: since I can't use xrandr with flutter-pi, I don't know of a way to rotate the screen. Do you know of any way to rotate the screen 90 deg to portrait mode? Would help a lot not to have to develop everything sideways...

limbouser, 2020-02-18:
An error similar to 
Duplicate of #38 was present. When I tried the latest commit, a error  `Could not get last VBlank timestamp. drmCrtcGetSequence` came up.

ardera, 2020-02-18:
> The screen works now, but there's a nasty side effect. I get this crash pretty soon after scrolling views in flutter:

my bad, some parts of the vsync mechanism mistakenly weren't disabled by the workaround. Should work now.

> When I tried the latest commit, a error `Could not get last VBlank timestamp. drmCrtcGetSequence` came up.

Wow `drmCrtcGetSequence` seems to be really buggy. I changed the behaviour so that `flutter-pi` will not exit when `drmCrtcGetSequence` returns an error.
In your case, it should now print something like:
```
WARNING: Could not get last vblank timestamp. <error code>
         VSync will be disabled.
         . . .
```

Can you tell me what error code it prints in your case? So whats the text after `Could not get last vblank timestamp.`?

ekuusi, 2020-02-18:
Can confirm that it works now, no crashes. Your work is super impressive.

Sorry to go a bit off-topic, but since we're discussing displays and as I've banged my head against this all day, quick question: is there a way to flip the HDMI screen 90 degrees somehow for flutter-pi?

Since we have to use Fake-KMS, /boot/config.txt -> display_hdmi_rotate=1 doesn't work (it flips the screen but doesn't adjust resolution), and as we're not using X can't use xrandr, which is how it can be done in Desktop.

ardera, 2020-02-18:
> Can confirm that it works now, no crashes. Your work is super impressive.

Thank you!

> Sorry to go a bit off-topic, but since we're discussing displays and as I've banged my head against this all day, quick question: is there a way to flip the HDMI screen 90 degrees somehow for flutter-pi?

This is actually integrated into flutter, just use [`SystemChrome.setPreferredOrientations`](https://api.flutter.dev/flutter/services/SystemChrome/setPreferredOrientations.html).

The default [`DeviceOrientation`](https://api.flutter.dev/flutter/services/DeviceOrientation-class.html) is `landscapeLeft`. If you want to rotate 90 degrees clockwise, you need to set the orientation to `portraitUp`. Just call
```dart
await SystemChrome.setPreferredOrientations(const {DeviceOrientation.portraitUp});
```
somewhere in your code.

ekuusi, 2020-02-18:
You're my hero. I love you.

ardera, 2020-02-18:
> You're my hero. I love you.

You're welcome 😄

limbouser, 2020-02-19:
`engine_argv[0] = flutter-pi
initializing display...
Finding a suitable DRM device, since none is given...
looking for a suitable DRM device from 1 available DRM devices...
  devices[0]: 
    available nodes: DRM_NODE_PRIMARY, DRM_NODE_RENDER
    nodes[DRM_NODE_PRIMARY] = "/dev/dri/card0"
    nodes[DRM_NODE_RENDER] = "/dev/dri/renderD128"
    bustype: DRM_BUS_PLATFORM
    businfo.fullname: /soc/gpu
    opening DRM device candidate at "/dev/dri/card0"...
    getting resources of DRM device candidate at "/dev/dri/card0"...
    flutter-pi chose "/dev/dri/card0" as its DRM device.
Finding a connected connector from 2 available connectors...
  connectors[0]: connected? yes, type: 0x0B (HDMI-A), 1220mm x 690mm
  connectors[1]: connected? unknown, type: 0x05, 0mm x 0mm
Choosing DRM mode from 22 available modes...
  modes[0]: name: "1920x1080", 1920x1080p, 60Hz, type: 72, flags: 5
    this mode is preferred by DRM. (DRM_MODE_TYPE_PREFERRED)
  modes[1]: name: "1920x1080", 1920x1080p, 60Hz, type: 64, flags: 5
  modes[2]: name: "1920x1080i", 1920x1080i, 60Hz, type: 64, flags: 21
  modes[3]: name: "1920x1080i", 1920x1080i, 60Hz, type: 64, flags: 21
  modes[4]: name: "1920x1080", 1920x1080p, 50Hz, type: 64, flags: 5
  modes[5]: name: "1920x1080i", 1920x1080i, 50Hz, type: 64, flags: 21
  modes[6]: name: "1280x1024", 1280x1024p, 60Hz, type: 64, flags: 5
  modes[7]: name: "1440x900", 1440x900p, 60Hz, type: 64, flags: 9
  modes[8]: name: "1360x768", 1360x768p, 60Hz, type: 64, flags: 5
  modes[9]: name: "1280x720", 1280x720p, 60Hz, type: 64, flags: 5
  modes[10]: name: "1280x720", 1280x720p, 60Hz, type: 64, flags: 5
  modes[11]: name: "1280x720", 1280x720p, 50Hz, type: 64, flags: 5
  modes[12]: name: "1024x768", 1024x768p, 60Hz, type: 64, flags: 10
  modes[13]: name: "800x600", 800x600p, 60Hz, type: 64, flags: 5
  modes[14]: name: "720x576", 720x576p, 50Hz, type: 64, flags: 10
  modes[15]: name: "720x576i", 720x576i, 50Hz, type: 64, flags: 4122
  modes[16]: name: "720x480", 720x480p, 60Hz, type: 64, flags: 10
  modes[17]: name: "720x480", 720x480p, 60Hz, type: 64, flags: 10
  modes[18]: name: "720x480i", 720x480i, 60Hz, type: 64, flags: 4122
  modes[19]: name: "720x480i", 720x480i, 60Hz, type: 64, flags: 4122
  modes[20]: name: "640x480", 640x480p, 60Hz, type: 64, flags: 10
  modes[21]: name: "640x480", 640x480p, 60Hz, type: 64, flags: 10
Display properties:
  1920 x 1080, 60Hz
  1220mm x 690mm
  pixel_ratio = 1.000000
Finding DRM encoder...
Creating GBM device
Querying EGL client extensions...
Getting EGL display for GBM device...
Initializing EGL...
Querying EGL display extensions...
Using display 0x9f33a0 with EGL version 1.4
===================================
EGL information:
  version: 1.4 (DRI2)
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_client_extensions EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm"
  display extensions: "EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_WL_bind_wayland_display "
===================================
Binding OpenGL ES API...
Choosing EGL config...
Finding EGL configs with appropriate attributes...
Creating EGL context...
Creating EGL window surface...
===================================
OpenGL ES information:
  version: "OpenGL ES 2.0 Mesa 17.3.8"
  shading language version: "OpenGL ES GLSL ES 1.0.16"
  vendor: "Broadcom"
  renderer: "VC4 V3D 2.1"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_npot GL_OES_vertex_half_float GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_separate_shader_objects GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_texture_border_clamp GL_KHR_context_flush_control GL_OES_draw_elements_base_vertex GL_OES_texture_border_clamp GL_KHR_no_error GL_MESA_tile_raster_order "
[... 11 lines cut ...]
[raw_keyboard] init.
[test-plugin] init.
WARNING: Could not get last vblank timestamp. Invalid argument
         VSync will be disabled.
         See https://github.com/ardera/flutter-pi/issues/38 for more info.
[... cut ...]
Running message loop...
flutter: Observatory listening on http://127.0.0.1:39759/CPJfiHHKWWk=/
`

This is how my error log looks like. The screen goes dark/blank after this until ctrl+C is pressed. I'm not working on a Raspbian Image but on a Pi image of another embedded linux project. (Will check if I'm getting the same error when I use a Raspbian Image). 

Also could you include the flutter_assets of any very basic flutter application. Just to ensure that this isn't an error caused by any version mismatches.

ekuusi, 2020-02-19:
> Also could you include the flutter_assets of any very basic flutter application. Just to ensure that this isn't an error caused by any version mismatches.

Clone this: https://github.com/iampawan/Flutter-UI-Kit
And just add the "debugDefaultTargetPlatformOverride = TargetPlatform.fuchsia;" into main.dart. I've used that for testing.

ardera, 2020-02-19:
>     WARNING: Could not get last vblank timestamp. Invalid argument

`Invalid Argument` (or more precisely `EINVAL`) is most probably caused by an outdated or not supported linux kernel. You should check (with `uname  -a`) that you have at least kernel version `4.15`.
Mesa version `17.3.8` is two years old too. Really seems like your distribution is kinda old. Graphics on Raspberry Pi is pretty cutting-edge (on the 4 at least), I'd always make sure to get the latest kernel and Mesa.

These are not the cause of your problem though, but they may cause problems later on.

> [ERROR:flutter/shell/common/shell.cc(178)] Dart Error: Can't load Kernel binary: Invalid kernel binary format version.
>     [ERROR:flutter/shell/common/engine.cc(188)] Could not prepare to run the [...]

This is the actual problem. Two possible causes:
- A recent update to the `stable` flutter SDK changed the engine version, maybe that's an incompatible change. I'll test that and upload new binaries to the `engine-binaries` branch if necessary.
- You used a flutter SDK that's not compatible with your engine version, so for example you switched your SDK to `dev` `beta` or `master` channel, or you used and outdated `stable` channel SDK (outdated meaning anything older than `1.12`)

> Also could you include the flutter_assets of any very basic flutter application. Just to ensure that this isn't an error caused by any version mismatches.

Yeah, @ekuusi's asset bundle will work. Btw you can create a minimal flutter app with the `flutter create <project name>` command.

limbouser, 2020-02-19:
How did you create the flutter engine(`libflutter_engine.so`)? Can you provide the steps for the same? I tried the article by Chinmay Garde in medium.com . But get blocked at an error while creating libcxx. 
As you mentioned in your article, even though building engine is a bit hard it's a great option. I would like to do that actually.

Will your engine version 1.9.1 still work? The build I'm using doesn't support the latest version of flutter engine(1.12.3) Some GLIBC errors.

ekuusi, 2020-02-19:
@ardera Just to verify (even though it's pretty obvious), there indeed is some tearing happening without vsync. It's not horrible, and mostly only noticeable when you scroll up / down, but def something that would be worth looking into if there's a solution that "fixes" these problematic monitors. I tested various apps with both the HP monitor and the problematic one.

Other than that, the performance on a RPi4 is really impressive, very smooth even on a 4k monitor. I did notice that for example in this demo app: https://github.com/scitbiz/flutter_pokedex the pokemon don't load (they did when building for my phone) and some background color animations didn't trigger (the color just instantly changes), even though I didn't notice any platform specific plugins. Not a big deal really, just curious why the difference, as other demo apps I tried seemed to work 1:1.

ardera, 2020-02-19:
@limbouser

> How did you create the flutter engine(`libflutter_engine.so`)? Can you provide the steps for the same? I tried the article by Chinmay Garde in medium.com . But get blocked at an error while creating [...]

Compiling the engine is a complicated topic. There are so many things that can go wrong doing it, many things that are platform-dependant, so I'm sorry but you're on your own. I support everyone that has problems with `flutter-pi`, but not with the flutter engine. There are binaries for flutter 1.9 on the `engine-binaries` branch though, it's the 2nd-latest commit, and of course feel free to comment on the article by Chinmay Garde, or write an issue on the [`flutter/engine` repo](https://github.com/flutter/engine). Actually some guy wanted to do a write-up of the compilation process some time ago, but I haven't heard from him since.

I can give you my cross compilation script though, [link here](https://gist.github.com/ardera/06e8057aa04eb3e28516612737d1e6d2). I compile in WSL because compiling on-target takes _ages_. This needs a cloned repo of GNU binutils, LLVM 9 _and_ LLVM 10. _Use at your own risk,_ I'm not gonna provide any support for it either.    
    
.     

@ekuusi

> @ardera Just to verify (even though it's pretty obvious), there indeed is some tearing happening without vsync. It's not horrible, and mostly only noticeable when you scroll up / down, but def [...]

I agree. Actually I think the `drmCrtcGetSequence` call is not that important, just need to find a way to trick the flutter engine into doing the right thing without vblank timestamps.

> Other than that, the performance on a RPi4 is really impressive, very smooth even on a 4k monitor.

Yeah I think so too, it's really great.

> I did notice that for example in this demo app: https://github.com/scitbiz/flutter_pokedex the pokemon don't load (they did when building for my phone) and some background color animations didn't [...]

While the App doesn't depend on any plugins directly, it _does_ indirectly. i.e. it depends on [`cached_network_image`](https://pub.dev/packages/cached_network_image), which depends on [`flutter_cache_manager`](https://pub.dev/packages/flutter_cache_manager), which depends on [`sqflite`](https://github.com/tekartik/sqflite/tree/master/sqflite) which includes some native code.

> EDIT: Also, am I missing something, because I'm building assets with --precompiled and --release:
> `flutter build bundle --precompiled --release` but the apps still show the debug banner and any overflow warnings are still displayed. Do I need to remove those manually in code, for example with
> `debugShowCheckedModeBanner: false` or is there something I'm doing wrong with the build command?

`flutter-pi` only support debug mode right now. It's not that much work for the `release` or `profile` mode support, but I saw it as low-priority because performance seems to be well enough just using `debug` mode

> I guess the overflow errors are supposed to always be there, just annoying when doing quick demos with layouts not optimized for all monitors. Quick googling didn't turn out any oneliner solutions of hiding the overflow errors.

Did you try the [`OverflowBox` widget](https://api.flutter.dev/flutter/widgets/OverflowBox-class.html)? It lets it's child overflow, but I'm not sure whether it will hide the overflow errors.

ekuusi, 2020-02-19:
Ok, cool. The debug indeed doesn't seem to be an issue at the moment. Though eventually would be useful to be able to squeeze that extra performance out.

I also tested https://rive.app/ demos (https://github.com/2d-inc/Flare-Flutter/tree/stable/example), the animations work and are smooth as butter. Using htop it was really neat seeing that when there's nothing going on in the app the Pi has one core at under 2% and everything else at 0, and while playing Rive / Flare animations just a couple of cores under 30%

So, decided to go with flutter-pi for embedded so I can live that code (and animate) once deploy everywhere dream for real .. (if they hopefully get flutter for web production ready this year)

ardera, 2020-02-20:
> Though eventually would be useful to be able to squeeze that extra performance out.

I agree.

> So, decided to go with flutter-pi for embedded so I can live that code (and animate) once deploy everywhere dream for real .. (if they hopefully get flutter for web production ready this year)

Nice! I've got to say though that flutter-pi is not really production ready. Could be that there are bugs or security risks, I don't provide any guarantees for that, it's more like an experiment right now. I'm not sure `flutter-pi` is the best way to solve the underlying problem either (flutter on linux with X or wayland), but I haven't found a better solution yet.

closed 2020-10-13 by ardera