ardera/flutter-pi #40: Flutter SDK Version

limbouser, 2020-02-26:
Is stable branch of flutter SDK used to build the flutter assets? I don't find a stable SDK release with version 1.12.3 [here](https://flutter.dev/docs/development/tools/sdk/releases).
Is the flutter SDK version and flutter  engine version used by `flutter-pi` in anyway related?
Would the keyboard support code of `flutter-pi` work if the engine-version is kept at 1.9.1? (I found out that it wouldn't but what would be the reason for that?)

ardera, 2020-02-26:
> Is stable branch of flutter SDK used to build the flutter assets? I don't find a stable SDK release with version 1.12.3 [here](https://flutter.dev/docs/development/tools/sdk/releases).
> Is the flutter SDK version and flutter engine version used by `flutter-pi` in anyway related?

If you use the most recent _stable_ flutter SDK, just use the most recent engine binaries I provide in the `engine-binaries` branch. If they don't work, I'll update them. Generally, any flutter SDK 1.12 will probably work with any flutter engine 1.12.

> Would the keyboard support code of `flutter-pi` work if the engine-version is kept at 1.9.1? (I found out that it wouldn't but what would be the reason for that?)

If it doesn't work, it's probably because some interface for keyboard support in the flutter engine changed.

Though it could be that this is/was because I accidentally disabled text input (not raw keyboard input though) in the last update, I fixed that in bc18152e95daa5d1245c8706120ad67e5f0681f7.

limbouser, 2020-02-26:
Does flutter have a stable release 1.12.3? If yes, why is it not their in the stable releases list? Just curious about the version you have use d to build the flutter engine binaries.

ardera, 2020-02-27:
The flutter SDK does have a stable release 1.12.3, but it was replaced by newer stable versions. They probably didn't keep the older stable versions listed because the changes were non-breaking. You can see a list of all flutter versions on the [github releases page of the flutter sdk](https://github.com/flutter/flutter/releases).

closed 2020-03-02 by limbouser

goudon007, 2020-04-27:
@ardera which flutter sdk version did you use ? I used the web version 1.12.3.
And I am not able to `flutter` build bundle'.
I keep getting errors about 'wrong full snapshot version, expected bla bla'
Please help me.

limbouser, 2020-04-27:
Just to keep it very simple if you have a Windows machine, try this 
`https://storage.googleapis.com/flutter_infra/releases/stable/windows/flutter_windows_v1.12.13+hotfix.9-stable.zip`

And run the `flutter_console.bat` , move to the flutter directory examples, any example and run
`flutter build bundle`

goudon007, 2020-04-27:
my question was with regard to the flutter-pi using Raspberry pi. So i did all the steps mentioned in the article. 
Then i installed flutter 1.12.3 web and then tried to 'flutter build bundle' the hello world project and then it failed with error 'wrong full snapshot version, expected bla bla'
Then I installed the latest verion of flutter instead. When i tried to build bundle, this time the error was  related to "Snapshot not compatible with the current VM configuration: the snapshot requires 'release use_bare_instructions no-"asserts bla bla bla"
Do you have any advices if its not a windows pc. ?

ardera, 2020-04-27:
> So i did all the steps mentioned in the article.

What article do you mean? Do you mean the readme of my project? Or did you build the engine yourself and used Chinmay Garde's article on medium for that?

> Then i installed flutter 1.12.3 web and then tried to 'flutter build bundle' the hello world project and then it failed with error 'wrong full snapshot version, expected bla bla'

What do you mean by flutter _web_? Do you mean you installed it via the webpage?

You should use the very latest stable flutter SDK for building the asset bundle. `1.12.3` is just the version the SDK was at when I built the engine binaries that you can find in the `engine-binaries` branch, but that same engine still works with newer SDKs.

> Then i installed flutter 1.12.3 web and then tried to 'flutter build bundle' the hello world project and then it failed with error 'wrong full snapshot version, expected bla bla'
> Then I installed [...]

The `bla bla` part is actually very important. That's where it says why the dart VM can't execute some snapshot. Could you post the complete commandline output of you running `flutter build bundle` (except your trying to use the SDK on ARM, see next paragraph)

> Do you have any advices if its not a windows pc. ?

There shouldn't be anything OS-specific when building the asset bundle. But keep in mind that the flutter SDK doesn't yet support ARM as a host platform. So running the SDK on an ARM machine, which includes the Raspberry Pi, won't work. You need to use a x86 (64bit) machine for building instead.

goudon007, 2020-04-28:
Thanks for your input. I am not a professional at programming the PI. I just followed the instructions from your description. I was assuming I should also build bundles from the raspberry pi. 
Now I built them using windows and copied to my PI 3 and it instantly worked. Many thanks for the code and your support.

ardera, 2020-04-28:
Thanks, glad I could help!

goudon007, 2020-05-01:
I have on more question. If I am to flash my raspberry pi with Ubunut core OS, would the flutter-pi still work ? Or is there an other thing I have to keep in mind while doing this ?

ardera, 2020-05-01:
I'm not that familiar with Ubuntu Core OS. If it contains the drivers for the VideoCore VI (or VideoCore IV if Pi 3) (which is pretty likely), it should work. Other than that, there are no specific dependencies or requirements for non-Raspbian based distributions, I think.