ardera/flutter-pi #45: Portrait - LCD - does not work

boeledi, 2020-03-14:
Hi,

First of all so many thanks for your 'flutter-pi'.  It really works very well.

There is only 1 problem I am currently facing using the "standard" 7" LCD display.

Flutter-pi only works in Landscape mode (even if the lcd_rotate=2).
Also, "await SystemChrome.setPreferredOrientations([DeviceOrientation.portraitUp, DeviceOrientation.portraitDown]);" prevents the application from running.

Would you have any solution to this?

In advance, many thanks

Didier

ardera, 2020-03-16:
Okay, I'll try to reproduce your error.

> even if the lcd_rotate=2

I recommend not using the linux or raspberry pi firmware settings to rotate the display, the rotation is entirely done by flutter-pi. Also, at least in my experience, most of the display settings inside `config.txt` are kinda hacky and glitchy

bottee, 2020-03-19:
Hi,

first of all, this is a really great project. Thank you very much for making it available to everybody!

I have the same problem here. It happens without any modification in `config.tx`. If I start my flutter app in landscape mode everything works like a charm.
But if I add `await SystemChrome.setPreferredOrientations([DeviceOrientation.portraitUp, DeviceOrientation.portraitDown]);` to my flutter application it stops after the startup animation and changes to dark colors.

I have a Raspi4 with 4GB Ram and the official 7" LCD touch Dislplay.

Do you have any hints how to find the cause?

gustavsl, 2020-03-19:
I can confirm this same issue - not on a Raspberry Pi but on a i.MX 6 device.
Flutter application displays black screen after adding setPreferredOrientations.

ardera, 2020-03-20:
Weird, it works fine for me. I created a brand-new barebones flutter project using `flutter create` and put the call to `setPreferredOrientations` into the `initState` and `build` methods.

Can anyone send me some more code so I can reproduce the issue?

gustavsl, 2020-03-22:
I'm using the example from https://github.com/google/flutter-desktop-embedding/tree/master/example adding `await SystemChrome.setPreferredOrientations([DeviceOrientation.portraitUp]);`before ´ runApp(new MyApp());´.

ardera, 2020-03-22:
When I put the `SystemChrome.setPreferredOrientations` call before `runApp`, at least for me, this error message is printed:
```text
[ERROR:engine/src/flutter/lib/ui/ui_dart_state.cc(157)] Unhandled Exception: ServicesBinding.defaultBinaryMessenger was accessed before the binding was initialized.
If you're running an application and need to access the binary messenger before `runApp()` has been called (for example, during plugin initialization), then you need to explicitly call the `WidgetsFlutterBinding.ensureInitialized()` first.
If you're running a test, you can call the `TestWidgetsFlutterBinding.ensureInitialized()` as the first line in your test's `main()` method to initialize the binding.
#0      defaultBinaryMessenger.<anonymous closure> (package:flutter/src/services/binary_messenger.dart:76:7)
#1      defaultBinaryMessenger (package:flutter/src/services/binary_messenger.dart:89:4)
#2      MethodChannel.binaryMessenger (package:flutter/src/services/platform_channel.dart:140:62)
#3      MethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart:314:35)
#4      OptionalMethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart:429:36)
#5      SystemChrome.setPreferredOrientations (package:flutter/src/services/system_chrome.dart:236:35)
#6      main (package:orientation/main.dart:8:6)
#7      _runMainZoned.<anonymous closure>.<anonymous closure> (dart:ui/hooks.dart:239:25)
#8      _rootRun (dart:async/zone.dart:1126:13)
#9      _CustomZone.run (dart:async/zone.dart:1023:19)
#10     _runZoned (dart:async/zone.dart:1518:10)
#11     runZoned (dart:async/zone.dart:1502:12)
#12     _runMainZoned.<anonymous closure> (dart:ui/hooks.dart:231:5)
#13     _startIsolate.<anonymous closure> (dart:isolate-patch/isolate_patch.dart:307:19)
#14     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:174:12)
```

As it says in the error message, just don't use `SystemChrome.setPreferredOrientations` before the `runApp` call.

`runApp` initializes the bindings to the platform side. `SystemChrome.setPreferredOrientations` needs these bindings to be initialized, since it uses these bindings to tell the platform side about the preferred orientations. I think, in general, it's a bad idea to do flutter-specific stuff before the `runApp` call.

bottee, 2020-05-12:
Sorry for this late response. I tried to create a minimal version to reproduce the issue, as my project is quite sensitive. As a result I found that the issue is caused by Flutter. So from my side the issue is not an flutter_pi issue. I'm sorry for the inconvenience. 

And thank you for this great project!

closed 2020-06-14 by ardera