ardera/flutter-pi #46: Difference between Flutter_wayland and flutter-pi

limbouser, 2020-03-18:
Apart from the fact that one is designed for Pi and other for a 64bit machine, what is the difference between [flutter_wayland](https://github.com/chinmaygarde/flutter_wayland) and flutter-pi?

Does flutter-pi run on a machine if vc4graphics isn't available?

ardera, 2020-03-19:
The difference is basically abstraction. For _flutter_wayland_, it looks like this:
`DRM (linux kernel) <-- wayland <-- flutter_wayland <-- flutter`

For _flutter_pi_, it looks like this:
`DRM (linux kernel) <-- flutter_pi <-- flutter`

So, basically, for _flutter_wayland_, there's an additional layer between the linux kernel and flutter_wayland, while flutter_pi communicates with the linux kernel directly.

There's some advantages to this (flutter_pi's) approach, but also some drawbacks.

Advantages:

- resources. It's just plain wasteful to start a whole wayland server for a single, full-screen application.
- faster (in general, and also startup) because of less overhead.
- easier to implement platform-specific stuff. For example, in the past, the Raspberry Pi used a different graphics stack. IIRC, you could only get hardware acceleration for OpenGL by using that proprietary graphics stack. If you want to make `flutter-pi` compatible with some platform that still uses such a non-standard grahpics stack, you could just modify the `init_graphics` procedure of flutter-pi to get it to work. It's not that easy with flutter_wayland though, as you'd have to update wayland for that.

Disadvantages:

- wayland does a lot of stuff other than graphics. `flutter_pi` has to implement that stuff by itself. For example, there's no way to just change the speed of your mouse for flutter-pi, because there's no layer of abstraction in flutter-pi that'd implement such configuration. Keyboard layouts is similiar, that's why there's this limitation to the text input from the controlling console (see readme).
- you can't run flutter-pi alongside your x- or wayland-server.

limbouser, 2020-03-30:
Doesn't Wayland alone exist? Can I run flutter-pi somewhere if I can't afford to have vc4-drm? What changes wouldnI have to make to make this thing work on plain Wayland?

ardera, 2020-03-30:
> Doesn't Wayland alone exist?

I don't understand what you mean by this.

> Can I run flutter-pi somewhere if I can't afford to have vc4-drm?

As I wrote above, if it has working DRM support, `flutter-pi` should run. That's untested though. I don't understand what you mean by "afford to have vc4-drm".

> What changes wouldnI have to make to make this thing work on plain Wayland?

The sole purpose of flutter-pi is to run without X or wayland, so you could probably discard most of the `flutter-pi.c`, `console_keyboard.c` files. No idea what specific changes would be necessary though.

closed 2020-04-25 by ardera