ardera/flutter-pi #47: Guidance on writing new plugins

mortenboye, 2020-03-19:
What is the recommended approach to writing new plugins for flutter-pi?
Looking at the gpiod plugin, it looks like a dart package (no native code) combined with a gpiod.c file in the plugins folder.
What are the steps necessary to link and build the .c file and make flutter-pi aware of new plugins?

ardera, 2020-03-19:
> Looking at the gpiod plugin, it looks like a dart package (no native code) combined with a gpiod.c file in the plugins folder.

Exactly, a flutter-pi plugin is just:

- a dart package for the dart-side implementation
- typically, a `my_plugin.c` file inside `src/plugins` and a `my_plugin.h` file inside `include/plugins` (if you need more files, just create a `my_plugin` folder inside `src/plugins` and `include/plugins` which contains your plugin source files)
- some code inside `src/plugin_registry.c` to make your plugin known to the plugin registry
- of course, you also need to update the `Makefile` so it compiles your `src/plugins/my_plugin.c` file

It's also "convention" to implement a flag, like `BUILD_MY_PLUGIN`, so people can choose when compiling `flutter-pi` whether your plugin should be built and included in the finished binary. (To be more precise, the `BUILD_MY_PLUGIN` flag should do nothing more than enable/disable the registration of your plugin to the plugin_registry, see [here](https://github.com/ardera/flutter-pi/blob/520dc58cee2c613899c721cb49231fc8eafd3f33/src/pluginregistry.c#L12) and [here](https://github.com/ardera/flutter-pi/blob/520dc58cee2c613899c721cb49231fc8eafd3f33/src/pluginregistry.c#L44))

> What are the steps necessary to link and build the .c file and make flutter-pi aware of new plugins?

So, in practice, you'll have to add 6 lines to `pluginregistry.c` to make your plugin known to `flutter-pi`:
```C
// my_plugin.h should contain the definitions of `my_plugin_init`
// and `my_plugin_deinit`
#ifdef BUILD_MY_PLUGIN
#    include <plugins/my_plugin.h>
#endif

// . . .

struct flutterpi_plugin hardcoded_plugins[] = {

// . . . the other plugins' registrations here

#ifdef BUILD_MY_PLUGIN
    {.name = "my_plugin", .init = my_plugin_init, .deinit = my_plugin_deinit},
#endif
};
```

Then you need to update the `Makefile` so your plugin gets built:

1) Add `src/plugins/my_plugin.c` to the definition of the `SOURCES` variable
2) Add the `-DBUILD_MY_PLUGIN` c-flag to the definition of the `REAL_CFLAGS` variable

mortenboye, 2020-03-24:
Perfect, works! Thanks, closing...

closed 2020-03-24 by mortenboye