ardera/flutter-pi #50: pixel ratio change

queirozrphl, 2020-03-27:
Hi folks, first of all, thank you very much for this engine. It's amazing work. 

I'm trying tu use this engine, but in portrait mode, then I do the change in the flutter code, but the pixel ratio of the display dont change, causing some mistakes in the view. The pixel ratio keeps in 1.00 even though I do changed it on rpi config. 

I guess you can help me with this little thing. Thanks

> Choosing DRM mode from 28 available modes...
  modes[0]: name: "1024x600", 1024x600p, 60Hz, type: 72, flags: 6
    this mode is preferred by DRM. (DRM_MODE_TYPE_PREFERRED)
  modes[1]: name: "1920x1080", 1920x1080p, 60Hz, type: 64, flags: 5
  modes[2]: name: "1920x1080", 1920x1080p, 60Hz, type: 64, flags: 5
  modes[3]: name: "1920x1080", 1920x1080p, 50Hz, type: 64, flags: 5
  modes[4]: name: "1280x1024", 1280x1024p, 75Hz, type: 64, flags: 5
  modes[5]: name: "1280x1024", 1280x1024p, 60Hz, type: 64, flags: 5
  modes[6]: name: "1440x900", 1440x900p, 75Hz, type: 64, flags: 6
  modes[7]: name: "1440x900", 1440x900p, 60Hz, type: 64, flags: 9
  modes[8]: name: "1152x864", 1152x864p, 75Hz, type: 64, flags: 5
  modes[9]: name: "1280x720", 1280x720p, 60Hz, type: 64, flags: 5
  modes[10]: name: "1280x720", 1280x720p, 60Hz, type: 64, flags: 5
  modes[11]: name: "1280x720", 1280x720p, 50Hz, type: 64, flags: 5
  modes[12]: name: "1024x768", 1024x768p, 75Hz, type: 64, flags: 5
  modes[13]: name: "1024x768", 1024x768p, 70Hz, type: 64, flags: 10
  modes[14]: name: "1024x768", 1024x768p, 60Hz, type: 64, flags: 10
  modes[15]: name: "832x624", 832x624p, 75Hz, type: 64, flags: 10
  modes[16]: name: "800x600", 800x600p, 75Hz, type: 64, flags: 5
  modes[17]: name: "800x600", 800x600p, 72Hz, type: 64, flags: 5
  modes[18]: name: "800x600", 800x600p, 60Hz, type: 64, flags: 5
  modes[19]: name: "800x600", 800x600p, 56Hz, type: 64, flags: 5
  modes[20]: name: "720x576", 720x576p, 50Hz, type: 64, flags: 10
  modes[21]: name: "720x480", 720x480p, 60Hz, type: 64, flags: 10
  modes[22]: name: "720x480", 720x480p, 60Hz, type: 64, flags: 10
  modes[23]: name: "640x480", 640x480p, 75Hz, type: 64, flags: 10
  modes[24]: name: "640x480", 640x480p, 73Hz, type: 64, flags: 10
  modes[25]: name: "640x480", 640x480p, 60Hz, type: 64, flags: 10
  modes[26]: name: "640x480", 640x480p, 60Hz, type: 64, flags: 10
  modes[27]: name: "720x400", 720x400p, 70Hz, type: 64, flags: 6
Display properties:
  1024 x 600, 60Hz
  410mm x 260mm
  pixel_ratio = 1.000000
Finding DRM encoder...
Creating GBM device
Querying EGL client extensions...
Getting EGL display for GBM device...
Initializing EGL...
Querying EGL display extensions...
Using display 0x590aa0 with EGL version 1.4

ardera, 2020-03-27:
> Hi folks, first of all, thank you very much for this engine. It's amazing work.

Thanks!

> but the pixel ratio of the display dont change, causing some mistakes in the view. The pixel ratio keeps in 1.00 even though I do changed it on rpi config.

I'm afraid I don't understand what you mean. Why should [flutter's device pixel ratio](https://api.flutter.dev/flutter/dart-ui/Window/devicePixelRatio.html) change on rotation? The device pixel ratio is calculated using the resolution of your display and the physical size of your display, it's not affected by rotation

There's also no way to change the pixel ratio in `raspi-config`. You can change the resolution in `raspi-config`, but I don't recommend doing that because the way `raspi-config` does it is really hacky and may result in some "glitchy" behaviour. The correct way to change your resolution is to supply a custom EDID file for your display, see [here](https://www.raspberrypi.org/documentation/configuration/config-txt/video.md)

> The pixel ratio keeps in 1.00 even though I do changed it on rpi config.

If you calculate the pixel ratio for your display according to [this documentation](https://api.flutter.dev/flutter/dart-ui/Window/devicePixelRatio.html), it's `0.657`. But I read somewhere that flutter can't handle device pixel ratios below 1 properly, so `flutter-pi` rounds up all pixel ratios below 1. That's why your pixel ratio is 1.00.

Though, I never actually tested whether flutter can handle pixel ratios below 1 myself. I'll do that now

queirozrphl, 2020-03-27:
Hey,

> There's also no way to change the pixel ratio in `raspi-config`. You can change the resolution in `raspi-config`, but I don't recommend doing that because the way `raspi-config` does it is really [...]

I do understand that. The problem is exactaly this, even when I make changes on there, the display properties of the DRM device dont changes. That doesnt changes with pixel_ratio > 1 either < 0. So I was wondering if this isn't a problem with openGL. It's not very clear to me why does this happens.

The error is that the flutter application can't be center properly in portrait mode. 

> But I read somewhere that flutter can't handle device pixel ratios below 1 properly, so `flutter-pi` rounds up all pixel ratios below 1. That's why your pixel ratio is 1.00.

Thanks for this information!

closed 2020-08-06 by queirozrphl