ardera/flutter-pi #51: Resolution problems (resolution too low in Flutter)

ekuusi, 2020-03-27:
Hi again, flutter-pi is working great, still loving it.

Just one issue, thought to post here before I dig deeper myself to figure out what's going on. I'm still using the same touchscreen I used earlier in https://github.com/ardera/flutter-pi/issues/38 - a WaveShare 1280x800 screen. Starting flutter-pi, everything seems to be fine and in the startup logs it says: 

```Display properties:
  1280 x 800, 63Hz
  150mm x 100mm
  pixel_ratio = 2.245614
```

But the resolution in Flutter is most definitely not 1280x800, it looks to be closer to 640x400, everything looking very different than when running the app on my 1280x800 emulator. Any ideas what might be the issue? I know the panel is capable of 1280x800, but I'm wondering whether the panel has some issues again as we had the vsync problem as well.

I have a superior screen on its way that I want to switch to anyways, but it would be good to understand why the resolution is so low when starting flutter-pi and whether I can somehow force it to 1280x800.

ekuusi, 2020-03-27:
I attached the Pi to my gaming monitor with:
```
Display properties:
  1920 x 1080, 60Hz
  530mm x 300mm
  pixel_ratio = 1.000000
```

And there everything looks as it should on a 1080p screen.

So I'm guessing the culprit is the pixel_ratio that is not 1 on the WaveShare display. Can I change the pixel ratio somehow to 1, or is it a firmware / hardware thing that's not editable?

ardera, 2020-03-27:
> So I'm guessing the culprit is the pixel_ratio that is not 1 on the WaveShare display.

I think so too. `flutter-pi` calculates the pixel ratio based on the resolution and the physical screen size. Some displays don't report a valid physical screen size and thus the resulting pixel ratio is bogus. It seems like your screen is one of them. (On [this](https://www.waveshare.com/10.1inch-HDMI-LCD-B-with-case.htm) page, which you posted in another issue, it says your screen is `216.96mm x 135.60mm`, not `150mm x 100mm` as reported by your display.)

> Can I change the pixel ratio somehow to 1, or is it a firmware / hardware thing that's not editable?

Actually the correct pixel ratio for your screen is 1.553. Right now, there are two ways you can fix this:

- You can hardcode your pixel ratio value by modifying [this line](https://github.com/ardera/flutter-pi/blob/520dc58cee2c613899c721cb49231fc8eafd3f33/src/flutter-pi.c#L99) line,
- or alternatively, you can hardcode your physical screen size by modifying [these two lines](https://github.com/ardera/flutter-pi/blob/520dc58cee2c613899c721cb49231fc8eafd3f33/src/flutter-pi.c#L91), so `flutter-pi` calculates the correct pixel ratio, which is probably better.

I think it'd be a good idea to add a commandline option for specifying a custom screen size.

ekuusi, 2020-03-27:
Changing the pixel ratio fixed it, thanks! Custom screen size would be really useful, yes. I could imagine the need to downscale if you have a very high res screen.

This is ok to close by me, if you don't want to keep it open to remind for the commandline option.

ardera, 2020-03-27:
> I could imagine the need to downscale if you have a very high res screen.

I'm not so sure about that specific case though, the sole purpose of the pixel ratio is to compensate for high / low resolutions

I'll close this for now

closed 2020-03-27 by ardera