ardera/flutter-pi #52: Segmentation fault on bigger tv screens

ffrm, 2020-03-31:
Hi, @ardera!

First of all I'd like to thank for developing this project. I was able to run a project which I've been planning for a long time. And now, using the flutter I could achieve the best perfomance ever!

I've developed my project and got a working version using a forked version of your repo. It's is in the [working](https://github.com/ffrm/flutter-pi/tree/working) branch.

It's running fine on my Samsung TV 32 inch, but when I try to run it on a bigger TV screen I get `Segmentation fault` error without any screen. Full log:

```
Using mouse input from mouse /dev/input/mouse0
Asset bundle path: ./flutter_assets/build/flutter_assets/
engine_argv[0] = ./libs/flutter-pi/out/flutter_pi
initializing display...
Opening DRM device...
Getting DRM resources...
Finding a connected connector...
Choosing DRM mode...
Display properties:
  0 x 0, 0Hz
  1210mm x 680mm
  pixel_ratio = 1.367100
Finding DRM encoder...
Creating GBM device
Querying EGL client extensions...
Getting EGL display for GBM device...
Initializing EGL...
Querying EGL display extensions...
Using display 4942280 with EGL version 1.4
===================================
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless EGL_EXT_platform_device"
  display extensions: "EGL_ANDROID_blob_cache EGL_ANDROID_native_fence_sync EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
Binding OpenGL ES API...
Choosing EGL config...
Finding EGL configs with appropriate attributes...
Creating EGL context...
Creating EGL window surface...
===================================
OpenGL ES 2.x information:
  version: "OpenGL ES 2.0 Mesa 19.3.2"
  shading language version: "OpenGL ES GLSL ES 1.0.16"
  vendor: "Broadcom"
  renderer: "VC4 V3D 2.1"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_npot GL_OES_vertex_half_float GL_OES_EGL_image GL_OES_depth_texture GL_AMD_performance_monitor GL_OES_packed_depth_stencil GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_EXT_occlusion_query_boolean GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_separate_shader_objects GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_texture_border_clamp GL_KHR_context_flush_control GL_OES_draw_elements_base_vertex GL_OES_texture_border_clamp GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_KHR_parallel_shader_compile GL_MESA_tile_raster_order "
===================================
Swapping buffers...
Locking front buffer...
getting new framebuffer for BO...
Segmentation fault
```

Do you have any clue about it?

Thanks in advance!

ardera, 2020-03-31:
> First of all I'd like to thank for developing this project. I was able to run a project which I've been planning for a long time. And now, using the flutter I could achieve the best perfomance ever!

Glad to hear, good luck with your project! ;)

> It's running fine on my Samsung TV 32 inch, but when I try to run it on a bigger TV screen I get `Segmentation fault` error without any screen. Full log:

> Do you have any clue about it?

`flutter-pi` has received a lot of updates in the past few months, there's a good chance one of these updates fixes your problem (so just pull my `master` to your `working` branch). Or is there a reason you use this specific commit of `flutter-pi`?

ffrm, 2020-03-31:
Yes, you are 100% right!

In the past I've tried to rebuild `flutter-pi` with the latest updates from master, but I was getting some erros. I tried to rebuild it now and seems that it was missing the `gpiod` and `libgpiod-dev` dependencies. After installation it worked perfectly fine on both TVs. And even better, it gave a better perfomance for the app and turned on some animations (which wasn't working before).

Again, thank you for your help with this project!

closed 2020-03-31 by ffrm