ardera/flutter-pi #57: How to build asset bundle

chetrf, 2020-05-06:
Thanks for making this available to everyone.  I seem to be stuck on how to build the asset bundle.  You call out the need to use a flutter SDK that is compatible with the engine version you are using.  I am going to attempt to use the engine from your engine-binaries branch.  Do I install a compatible SDK on my regular computer, build the assets bundle, and them move them to the Pi?  Can you point me to info that would help me take that step?  Again thanks very much.

ardera, 2020-05-06:
> Do I install a compatible SDK on my regular computer, build the assets bundle, and them move them to the Pi?

Exactly

> Can you point me to info that would help me take that step?

Unfortunately, there's not that much info about that. I found [this](https://wiki.loliot.net/docs/lang/flutter/engine/flutter-engine-for-linux-arm64/#example) (though he uses a flutter-pi instead of vanilla flutter-pi)
So here are some rough steps:

1. install flutter using the [instructions on the webpage](https://flutter.dev/docs/get-started/install)
2. if you're using windows: install SSH using [these instructions](https://docs.microsoft.com/en-us/windows-server/administration/openssh/openssh_install_firstuse)
3. it's best you setup public-key authentication using [these instructions](https://www.raspberrypi.org/documentation/remote-access/ssh/passwordless.md), so you don't have to enter your Pis password every time you transfer an asset bundle
4. open the command line / terminal and enter `cd <the path to your app directory>`. Keep in mind you need to surround the path in quotes if it contains spaces. So for example, `cd C:\Users\Test\Desktop\flutter_test\` would be a possible command. 
5. from this point on you can follow the instructions on [the page I mentioned](https://wiki.loliot.net/docs/lang/flutter/engine/flutter-engine-for-linux-arm64/#example), starting from the second command.
6. (for running flutter-pi on the target, just run `flutter-pi ./flutter_assets` or `sudo flutter-pi ./flutter_assets` if you are getting an error without sudo)

chetrf, 2020-05-06:
OK, very helpful feedback.  Will give this a go.  Thanks very much.

chetrf, 2020-05-07:
So very cool!  it is working.

For the benefit of others, here are few more details that I did understand at first:

I followed your directions on the engine-binaries branch to get your prebuilt binaries.
After moving the .h and .so files per the instructions on that branch I deleted the cloned flutter-pi directory.  

Then I cloned your master branch to create flutter-pi again.  Then I cd'd into flutter-pi and ran make.

After that I edited my .profile file and added the flutter-pi executable to my path.

I already had a garden variety flutter SDK install on my Mac.  I made sure I had a working app as tested in an iOS simulator.  Then inside my flutter app directory on my Mac I ran "flutter build bundle"

Then I used scp to copy over the build/flutter_assets directory.

Then at the command line on my pi I ran 

flutter-pi /home/pi/flutter_assets/

and it all jumped to life and is working.  I have 4 buttons on my simple app and they all seem to work nicely with the standard pi 7 inch touchscreen.

Also in case anyone is curios, I am using the latest Buster Lite version of Raspian.

This sooo cool!  You are awesome for putting this out there for others.  Thank you.  Now I can dev away on my computer.  When I am happy with my app just build the asset bundle and push it on over and run it on the pi.

ardera, 2020-05-07:
Thanks! I'm glad it worked!

closed 2020-06-09 by ardera