ardera/flutter-pi #60: Can't run release builds

being-yash-t, 2020-05-24:
First thanks for providing flutter-pi it helps a lot.
As everyone of us know the difference in performance of flutter between release & debug mode.
I tried adding --release to the "flutter build bundle" and tried to run the app still was is debug mode, I also tried adding --release to the "flutter-pi my_project/build/flutter_assets" still can't figure out how to build/run in release mode.

DisDis, 2020-05-24:
It is working only for debug mode

being-yash-t, 2020-05-24:
> It is working only for debug mode

Is it not possible to run it in release mode?

DisDis, 2020-05-24:
Please read comment https://medium.com/@chinmaygarde/the-aot-snapshot-is-generated-on-the-host-via-a-tool-called-gen-snapshot-you-can-find-this-in-2043f461b77a

DisDis, 2020-05-24:
@being-yash-t 
However it is working very fast and very smoothly in JIT mode. For example i im making photoframe (pi4 + 2560*1600px display). it work very fast.

being-yash-t, 2020-05-24:
@DisDis 
Thank you for your reply. I was getting about 64°c or more and hardly around 30 fps on a rpi3, thats the reason why I wanted to get a release builds.
Anyways will close this issue here now.

closed 2020-05-24 by being-yash-t

ardera, 2020-05-24:
> I was getting about 64°c or more and hardly around 30 fps on a rpi3, thats the reason why I wanted to get a release builds.

That's not normal. Last time I tried on my Pi 3 A+ I had 60fps. (in the default flutter create example. Don't know about flutter_gallery) I'll test again, maybe some update caused a performance regression.

being-yash-t, 2020-05-24:
> That's not normal. Last time I tried on my Pi 3 A+ I had 60fps. (in the default flutter create example. Don't know about flutter_gallery) I'll test again, maybe some update caused a performance [...]

Yeah, its okay when it is less complex ui, I had to seprate some logic. Now getting accross 35-45fps & moved to 64bit kernel