ardera/flutter-pi #61: Multitouch results into first finger not being released.

dbeulen, 2020-06-08:
I'm using flutter pi on a raspberry pi official touchscreen. However, when you touch the screen whenever a different finger is pressing a widget, the widget will never be released any more. This is tested on a slider and a card, but on both widgets the issue exists.

Flutter version 1.17.3
Flutter-pi version: latest commit on master 94f38d3
Flutter-pi engine: latest on engine binaries branch dcf0d22

To reproduce:
1. Flutter create test_app
2. Replace homepage build method with this code:
```
@override
  Widget build(BuildContext context) {
    // This method is rerun every time setState is called, for instance as done
    // by the _incrementCounter method above.
    //
    // The Flutter framework has been optimized to make rerunning build methods
    // fast, so that you can just rebuild anything that needs updating rather
    // than having to individually change instances of widgets.
    return Scaffold(
      appBar: AppBar(
        // Here we take the value from the MyHomePage object that was created by
        // the App.build method, and use it to set our appbar title.
        title: Text(widget.title),
      ),
      body: Center(
        // Center is a layout widget. It takes a single child and positions it
        // in the middle of the parent.
        child: Slider(
          value: _sliderValue,
          min: 0.0,
          max: 1.0,
          activeColor: Colors.red,
          inactiveColor: Colors.black,
          onChanged: (double newValue) {
            setState(() => _sliderValue = newValue);
          },
        ),
      ),
      floatingActionButton: FloatingActionButton(
        onPressed: _incrementCounter,
        tooltip: 'Increment',
        child: Icon(Icons.add),
      ), // This trailing comma makes auto-formatting nicer for build methods.
    );
  }
```
3. Build and run the application
4. With one finger, press the slider and move it for a bit (a dark red shadow will appear around the slider)
5. While still pressing the slider, touch the screen with a second finger.
6. Release both fingers. 
7. The shadow will still be shown around the slider, indicating it is still pressed. 

With a Card this issue is a bigger problem, because i cannot recognise a new touch after the problem arises.

ardera, 2020-06-08:
Okay, it's on my to-do list.

dbeulen, 2020-06-08:
Great, if theres anything i can help you with, just let me know. I had a look on the code, but just didn't even know where to start looking.

ardera, 2020-06-08:
@dbeulen
That's not your fault at all, the whole I/O "subsystem" is a complete mess right now. I tried integrating it all into one function that "does it all" (`on_evdev_input`) but that wasn't a good choice after all.

matzesoft, 2020-07-08:
Can we have a option to disable multitouch until you fixed the bug to prevent unwanted inputs?

ardera, 2020-07-08:
> Can we have a option to disable multitouch until you fixed the bug to prevent unwanted inputs?

I just now uploaded the [`fix/touchscreen-input`](https://github.com/ardera/flutter-pi/compare/fix/touchscreen-input) branch (which is based on the `feature/video_player_plugin` branch) that fixes touchscreen input. I still have to fix orientation on that branch, that's why it isn't merged into master yet. (The fix involved basically completely rewriting `flutter-pi.c`)

dbeulen, 2020-07-08:
We didnt do a very clean job of removing multitouch, but it works for now....

matzesoft, 2020-07-22:
So in first place, thanks both for your work!

@ardera I tried to compile the `fix/touchscreen-input` branch but always run into the same error.
```
pi@raspberrypi:~/sdks/flutter-pi $ make
Package libsystemd was not found in the pkg-config search path.
Perhaps you should add the directory containing `libsystemd.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libsystemd' found
Package libinput was not found in the pkg-config search path.
Perhaps you should add the directory containing `libinput.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libinput' found
Package libudev was not found in the pkg-config search path.
Perhaps you should add the directory containing `libudev.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libudev' found
cc -c -I./include  -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_GPIOD_PLUGIN -DBUILD_SPIDEV_PLUGIN -DBUILD_TEST_PLUGIN -DBUILD_OMXPLAYER_VIDEO_PLAYER_PLUGIN -O0 -ggdb  src/flutter-pi.c -o out/obj/flutter-pi.o
In file included from src/flutter-pi.c:25:
/usr/include/xf86drm.h:40:10: fatal error: drm.h: No such file or directory
 #include <drm.h>
          ^~~~~~~
compilation terminated.
make: *** [Makefile:41: out/obj/flutter-pi.o] Error 1
```
I am running on the lite version so maybe that's just me missing some packages.

ardera, 2020-07-22:
> I never had such problems with the master branch but I am running on the lite version so maybe that's just me missing some packages.

Exactly, you need to have the following additional packages installed: `libsystemd-dev`, `libudev-dev`, `libinput-dev`

The `drm.h: No such file or directory` error is weird. If it still ocurrs after installing the packages you can try invoking `make` with the `--always-make` flag.

The dependency on `libudev` is another reason why I haven't merged into master yet, udev sadly is not present on all embedded linux devices.

matzesoft, 2020-07-23:
> Exactly, you need to have the following additional packages installed: libsystemd-dev, libudev-dev, libinput-dev

I tried this but haven't put "-dev" behind the requested package name. Just because I'm interested - when does a library get this "-dev" sign?

But with these packages building worked fine until I wanted to start my app. The [main project](https://github.com/matzesoft/kart-project) I'm working on just crashed right after starting it:
```
[flutter-pi] WARNING: could not make stdin raw
flutter: Observatory listening on http://127.0.0.1:38329/x58Rzlh9VMc=/
[flutter-pi] Error handling platform message. plugin_registry_on_platform_message: Invalid argument
on_get_transformation
create_backing_store
present_layers_callback
on_get_transformation
create_backing_store
present_layers_callback
collect_window_surface_backing_store
destroying window surface backing store FBO
on_get_transformation
create_backing_store
present_layers_callback
on_get_transformation
create_backing_store
present_layers_callback
collect_window_surface_backing_store
destroying window surface backing store FBO
on_get_transformation
create_backing_store
present_layers_callback
on_get_transformation
create_backing_store
present_layers_callback
collect_window_surface_backing_store
destroying window surface backing store FBO
on_get_transformation
create_backing_store
present_layers_callback
on_get_transformation
create_backing_store
present_layers_callback
collect_window_surface_backing_store
destroying window surface backing store FBO
on_get_transformation
create_backing_store
present_layers_callback
Segmentation fault
```

Output of the default Flutter app was this:
```
[flutter-pi] WARNING: could not make stdin raw
flutter: Observatory listening on http://127.0.0.1:35811/T2GiZOLlQJk=/
[flutter-pi] Error handling platform message. plugin_registry_on_platform_message: Invalid argument
on_get_transformation
create_backing_store
present_layers_callback
on_get_transformation
create_backing_store
present_layers_callback
collect_window_surface_backing_store
destroying window surface backing store FBO
```
[image]

So this has probably something to do with orientation which you said is not implemented yet.

PR #70 (2020-08-05): Lots of additions, refactors
- resolves #65 
- fixes #61 
- partially implements #29 
- resolves #8 (finally)

- migrate to `libsystemd` for the main event loop
- support udev for hotplugging support, but only on platforms where it's present
- small improvements to AOT support
- support for platform views
- implement hardware composition using some glorious EGL hacks
- support hardware-accelerated video playback using omxplayer and the standard video_player package.
- mouse cursors are now supported, but only hardware mouse cursors. Included in flutter-pi are 64x64 mouse cursor skins
closed 2020-08-05 by ardera