ardera/flutter-pi #62: Running as root

matzesoft, 2020-06-09:
I'm currently trying to build a own library to communicate with the GPIOs to use UART and PWM. I am using the wiringPi library because it supports both interfaces. The problem is that wiringPi needs root permission to initialize PWM. I already searched for other solutions to run wiringPi without root but didn't find a lot.

So I just came to the point to run flutter-pi as root which doesn't seem to work.

```shell
pi@raspberrypi:~/projects/kart_project $ sudo flutter-pi .
sudo: flutter-pi: command not found
```

I was extra confused when I read this in README.md at the _Fixing the GPU permissions_ part:
> Otherwise, you'll need to run flutter-pi with sudo.

So is this a problem of flutter-pi or is there something wrong with my machine?

DisDis, 2020-06-09:
use:
`sudo ./flutter-pi .`

DisDis, 2020-06-09:
But you should exec (one time)
`sudo usermod -a -G render pi`
afte that we can exec just:
`flutter-pi .`

ardera, 2020-06-09:
> So is this a problem of flutter-pi or is there something wrong with my machine?

Neither, it's a user error 😄

You need to remember `sudo` is interpreting the argument you give it as a completely different user, namely `root`. `root` has it's own user directory (`/root`), similiar to the `pi` user, which has `/home/pi` as its user directory.

So everything you put into `/home/pi/.bashrc`, `/home/pi/.profile` or `/home/pi/bin` (et al.) will _not_ work when using `sudo`. (Because the shell looks for these file in `/root` instead). So in your case, you probably put the path to flutter-pi into your `PATH` variable inside `/home/pi/.bashrc`, or you created a symlink in `/home/pi/bin` to point to the flutter-pi executable, so you don't have to invoke flutter-pi with the path every time. But this does not work when running using `sudo`.

You need to use the path of the `flutter-pi` executable when running using `sudo`. So for example, `sudo ./flutter-pi` or `sudo /home/pi/devel/flutter-pi`. Or you modify the `.bashrc` or similiar of the `root` user, but that may be a security hazard.

matzesoft, 2020-06-09:
Oh thank you! I am quite new to Linux so thanks for the explanation. Works fine :)

closed 2020-06-09 by matzesoft