ardera/flutter-pi #64: How to specify DRM device?

tobiasht, 2020-06-23:
I am trying to run the flutter program on a 7inch display, but when I run the code I get this error:

```
./flutter-pi/out/flutter-pi ./dots-clock-master/dots_clock/build/flutter_assets

engine_argv[0] = ./flutter-pi/out/flutter-pi
initializing display...
Finding a suitable DRM device, since none is given...
could not query drm device list: No such file or directory
```

Seems like it doesn´t find the screen, even though I am currently using the display and it is showing the CLI perfectly. Is there a way to specify the "DRM" device? Any tips are much appreciated.

ardera, 2020-06-23:
Can you provide more details on the device & system you are using? Are you using a raspberry pi, if yes what model, OS, etc

If you're using a Raspberry Pi, can you give me complete output of the `raspinfo` command? (You may need to install it using `sudo apt install raspinfo`)

> Seems like it doesn´t find the screen, even though I am currently using the display and it is showing the CLI perfectly

Almost, a DRM device is basically a GPU. If it finds the GPU (/DRM device), but doesn't find a screen you'd see some other error (`couldn't find a connected connector` or so). DRM devices are located inside `/dev/dri`. Can you execute `ls /dev/dri` and give me the output of that?

> Is there a way to specify the "DRM" device? Any tips are much appreciated.

There was a way in earlier versions. Though I don't think you'd get any further with the ability to manually specify the DRM device, since this error is more likely caused by no DRM devices being present on your system at all

tobiasht, 2020-06-24:
Thank you so much for the help. I managed to solve the problem by restoring the raspberry pi and using another driver for it.

closed 2020-06-24 by tobiasht