ardera/flutter-pi #66: Errors running the pre-built engines.

Clancey, 2020-07-08:
I am getting the following output when trying to run on a Raspbery Pi 3 B. I followed all the instructions for using your existing pre-built engine. Any thoughts?

```cc -c -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_ELM327_PLUGIN -DBUILD_GPIOD_PLUGIN -DBUILD_SPIDEV_PLUGIN -DBUILD_TEST_PLUGIN -ggdb  -lgbm -ldrm -lGLESv2 -lEGL -lrt -lflutter_engine -lpthread -ldl  src/flutter-pi.c -o out/obj/flutter-pi.o
cd cc -c -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_ELM327_PLUGIN -DBUILD_GPIOD_PLUGIN -DBUILD_SPIDEV_PLUGIN -DBUILD_TEST_PLUGIN -ggdb  -lgbm -ldrm -lGLESv2 -lEGL -lrt -lflutter_engine -lpthread -ldl  src/platformchannel.c -o out/obj/platformchannel.o
cc -c -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_ELM327_PLUGIN -DBUILD_GPIOD_PLUGIN -DBUILD_SPIDEV_PLUGIN -DBUILD_TEST_PLUGIN -ggdb  -lgbm -ldrm -lGLESv2 -lEGL -lrt -lflutter_engine -lpthread -ldl  src/pluginregistry.c -o out/obj/pluginregistry.o
oucc -c -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_ELM327_PLUGIN -DBUILD_GPIOD_PLUGIN -DBUILD_SPIDEV_PLUGIN -DBUILD_TEST_PLUGIN -ggdb  -lgbm -ldrm -lGLESv2 -lEGL -lrt -lflutter_engine -lpthread -ldl  src/console_keyboard.c -o out/obj/console_keyboard.o
cc -c -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_ELM327_PLUGIN -DBUILD_GPIOD_PLUGIN -DBUILD_SPIDEV_PLUGIN -DBUILD_TEST_PLUGIN -ggdb  -lgbm -ldrm -lGLESv2 -lEGL -lrt -lflutter_engine -lpthread -ldl  src/plugins/elm327plugin.c -o out/obj/plugins/elm327plugin.o
cc -c -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_ELM327_PLUGIN -DBUILD_GPIOD_PLUGIN -DBUILD_SPIDEV_PLUGIN -DBUILD_TEST_PLUGIN -ggdb  -lgbm -ldrm -lGLESv2 -lEGL -lrt -lflutter_engine -lpthread -ldl  src/plugins/services.c -o out/obj/plugins/services.o
cc -c -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_ELM327_PLUGIN -DBUILD_GPIOD_PLUGIN -DBUILD_SPIDEV_PLUGIN -DBUILD_TEST_PLUGIN -ggdb  -lgbm -ldrm -lGLESv2 -lEGL -lrt -lflutter_engine -lpthread -ldl  src/plugins/testplugin.c -o out/obj/plugins/testplugin.o
cc -c -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_ELM327_PLUGIN -DBUILD_GPIOD_PLUGIN -DBUILD_SPIDEV_PLUGIN -DBUILD_TEST_PLUGIN -ggdb  -lgbm -ldrm -lGLESv2 -lEGL -lrt -lflutter_engine -lpthread -ldl  src/plugins/text_input.c -o out/obj/plugins/text_input.o
tcc -c -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_ELM327_PLUGIN -DBUILD_GPIOD_PLUGIN -DBUILD_SPIDEV_PLUGIN -DBUILD_TEST_PLUGIN -ggdb  -lgbm -ldrm -lGLESv2 -lEGL -lrt -lflutter_engine -lpthread -ldl  src/plugins/raw_keyboard.c -o out/obj/plugins/raw_keyboard.o

cc -c -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_ELM327_PLUGIN -DBUILD_GPIOD_PLUGIN -DBUILD_SPIDEV_PLUGIN -DBUILD_TEST_PLUGIN -ggdb  -lgbm -ldrm -lGLESv2 -lEGL -lrt -lflutter_engine -lpthread -ldl  src/plugins/gpiod.c -o out/obj/plugins/gpiod.o
cc -c -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_ELM327_PLUGIN -DBUILD_GPIOD_PLUGIN -DBUILD_SPIDEV_PLUGIN -DBUILD_TEST_PLUGIN -ggdb  -lgbm -ldrm -lGLESv2 -lEGL -lrt -lflutter_engine -lpthread -ldl  src/plugins/spidev.c -o out/obj/plugins/spidev.o
cc -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_ELM327_PLUGIN -DBUILD_GPIOD_PLUGIN -DBUILD_SPIDEV_PLUGIN -DBUILD_TEST_PLUGIN -ggdb  -lgbm -ldrm -lGLESv2 -lEGL -lrt -lflutter_engine -lpthread -ldl  out/obj/flutter-pi.o out/obj/platformchannel.o out/obj/pluginregistry.o out/obj/console_keyboard.o out/obj/plugins/elm327plugin.o out/obj/plugins/services.o out/obj/plugins/testplugin.o out/obj/plugins/text_input.o out/obj/plugins/raw_keyboard.o out/obj/plugins/gpiod.o out/obj/plugins/spidev.o -o out/flutter-pi
gcc out/obj/*.o out/obj/plugins/*.o -shared -o out/libFlutter.so
pi@raspberrypi:~/flutter/flutter-pi $ cd out
pi@raspberrypi:~/flutter/flutter-pi/out $ ./flutter-pi flutter_assets/
engine_argv[0] = ./flutter-pi
initializing display...
Finding a suitable DRM device, since none is given...
looking for a suitable DRM device from 1 available DRM devices...
  devices[0]: 
    available nodes: DRM_NODE_PRIMARY, DRM_NODE_RENDER
    nodes[DRM_NODE_PRIMARY] = "/dev/dri/card0"
    nodes[DRM_NODE_RENDER] = "/dev/dri/renderD128"
    bustype: DRM_BUS_PLATFORM
    businfo.fullname: /soc/gpu
    opening DRM device candidate at "/dev/dri/card0"...
    getting resources of DRM device candidate at "/dev/dri/card0"...
    flutter-pi chose "/dev/dri/card0" as its DRM device.
Finding a connected connector from 1 available connectors...
  connectors[0]: connected? yes, type: 0x0B (HDMI-A), 600mm x 340mm
Choosing DRM mode from 16 available modes...
  modes[0]: name: "1920x1080", 1920x1080p, 60Hz, type: 64, flags: 5
  modes[1]: name: "1920x1080", 1920x1080p, 60Hz, type: 64, flags: 5
  modes[2]: name: "1920x1080", 1920x1080p, 30Hz, type: 64, flags: 5
  modes[3]: name: "1920x1080", 1920x1080p, 30Hz, type: 64, flags: 5
  modes[4]: name: "1600x900", 1600x900p, 60Hz, type: 64, flags: 5
  modes[5]: name: "1280x1024", 1280x1024p, 60Hz, type: 64, flags: 5
  modes[6]: name: "1280x800", 1280x800p, 60Hz, type: 64, flags: 9
  modes[7]: name: "1152x864", 1152x864p, 60Hz, type: 0, flags: 6
  modes[8]: name: "1280x720", 1280x720p, 60Hz, type: 64, flags: 5
  modes[9]: name: "1280x720", 1280x720p, 60Hz, type: 64, flags: 5
  modes[10]: name: "1024x768", 1024x768p, 60Hz, type: 64, flags: 10
  modes[11]: name: "800x600", 800x600p, 60Hz, type: 64, flags: 5
  modes[12]: name: "720x480", 720x480p, 60Hz, type: 64, flags: 10
  modes[13]: name: "720x480", 720x480p, 60Hz, type: 64, flags: 10
  modes[14]: name: "640x480", 640x480p, 60Hz, type: 64, flags: 10
  modes[15]: name: "640x480", 640x480p, 60Hz, type: 64, flags: 10
Display properties:
  1920 x 1080, 60Hz
  600mm x 340mm
  pixel_ratio = 1.000000
Finding DRM encoder...
Creating GBM device
Querying EGL client extensions...
Getting EGL display for GBM device...
Initializing EGL...
Querying EGL display extensions...
Using display 0x9197a0 with EGL version 1.4
===================================
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless EGL_EXT_platform_device"
  display extensions: "EGL_ANDROID_blob_cache EGL_ANDROID_native_fence_sync EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
Binding OpenGL ES API...
Choosing EGL config...
[... 32 lines cut ...]
[flutter_spidev] Done.
[elm327plugin] running pid query queue processor
WARNING: Kernel didn't return a valid vblank timestamp. (timestamp == 0)
         VSync will be disabled.
         See https://github.com/ardera/flutter-pi/issues/38 for more info.
[ERROR:engine/src/flutter/shell/platform/embedder/embedder_surface_gl.cc(107)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
detected VideoCore IV as underlying graphics chip, and VC4 as the driver.
Reporting modified GL_EXTENSIONS string that doesn't contain non-working extensions.
[ERROR:engine/src/flutter/shell/platform/embedder/embedder_surface_gl.cc(107)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
flutter engine successfully started up.
Initializing Input devices...
  input device 0: path="/dev/input/event0"
      Cooler Master Technology Inc. MASTERKEYS PRO S, connected via USB. vendor: 0x2516, product: 0x003C, version: 0x0111
  input device 1: path="/dev/input/event1"
[... cut ...]
      Cooler Master Technology Inc. MASTERKEYS PRO S Keyboard, connected via USB. vendor: 0x2516, product: 0x003C, version: 0x0111
[ERROR:engine/src/flutter/shell/common/shell.cc(213)] Dart Error: Can't load Kernel binary: Invalid kernel binary format version.
[ERROR:engine/src/flutter/shell/common/engine.cc(200)] Could not prepare to run the isolate.
[ERROR:engine/src/flutter/shell/common/engine.cc(139)] Engine not prepare and launch isolate.
Running IO thread...
[ERROR:engine/src/flutter/shell/common/shell.cc(485)] Could not launch engine with configuration.
Running message loop...```

GioPan04, 2020-07-08:
With which flutter version you compiled the app?

Clancey, 2020-07-08:
Latest stable.

GioPan04, 2020-07-08:
The prebuilt engines are compiled for flutter 1.17 if I remember correctly. Try to built yourself for your flutter version

ardera, 2020-07-08:
Works fine for me with latest stable.

The only way this error can happen is when there is a version mismatch between flutter SDK and flutter engine. Minor version mismatches are okay, i.e. a 1.17.0 engine should work with a 1.17.5 SDK (it's not impossible that it won't work, in which case I'll just upload updated binaries.). The latest engine binaries I provide are for 1.17.0.

Are you sure the flutter engine you have installed on your target are the latest ones I provide on `engine-binaries`?

Clancey, 2020-07-08:
So I had the right engine, but it looks like the build machine was on dev channel, Sorry about that!

closed 2020-07-08 by Clancey