ardera/flutter-pi #67: What is the memory footprint?

nanfang2000, 2020-07-13:
app size and memory usage?

DisDis, 2020-07-13:
I have application
[image]
RaPi4 (4Gb Ram)
flutter_assets ~ 31Mb

nanfang2000, 2020-07-16:
@DisDis Thank you for you info!

nanfang2000, 2020-07-16:
338MB-63MB=275MB! That is a surprise number.
My test on Flutter Mac desktop app shows the release version is only 70MB.
the debug version is bigger which is 200+MB. So I guess your number is based on debug version? @DisDis 

I would like to porting flutter to a low end device which only has 128MB memory. so now I'm wondering whether it is feasible.

ardera, 2020-07-16:
> So I guess your number is based on debug version?

Yeah flutter-pi only supports debug mode right now. Though I'm working on AOT support.

DisDis, 2020-07-30:
After enabling AOT.
I have 200mb-74mb=126Mb
[image]

ardera, 2020-08-06:
I have a small test app that is just a Scaffold with an AppBar and a Chewie Player playing back an MP4 fetched via web.

- flutter-pi process: virt 214M, res 52M, shr 35M.
- separate omxplayer process that actually plays back the video: res 39M, shr 21M

closed 2020-10-08 by ardera