ardera/flutter-pi #71: SEGFAULT when hotplugging a mouse and libinput is using udev as a backend

ardera, 2020-08-05:
When libinput is using udev as a backend and a mouse is plugged in while flutter-pi is running, some time later a call to `libinput_dispatch` will result in a SEGFAULT because libinput somehow tries to invoke a function pointer that has a NULL value.

hoppler, 2020-10-31:
Are there any updates on this bug as it happens to me too.

If i use your raspberrypi-fast-ts in combination with a script to turn of the screen after n seconds per  echo 1 | sudo tee /sys/class/backlight/rpi_backlight/bl_power the raspberrypi-fast-ts sometimes crashes because of an i2c error, the systemd service for it restarts it and than the flutter-pi crashes too.

So i guess the segv happens when ever the list of devices changes, either per usb-keyboard or per reinitialisation of the virtual-devices of raspberrypi-fast-ts after a restart of the service.

####
You can even force this error with the raspberrypi-fast-ts by 
1. "systemctl disable raspberrypi-fast-ts.service" and 
2. change config.txt to use the default touchscreen driver 
3. reboort
4. after starting the flutter app 
5. run "systemctl start raspberrypi-fast-ts". 

Then you have a loop of flutter-app starting and crashing.  (as long as you have configured the flutter-app to restart on failure per systemd service)

ardera, 2020-10-31:
> If i use your raspberrypi-fast-ts in combination with a script to turn of the screen after n seconds per echo 1 | sudo tee /sys/class/backlight/rpi_backlight/bl_power the raspberrypi-fast-ts [...]

I should probably add some documentation for that, changing the backlight via the sysfs while `raspberrypi-fast-ts` is running causes the firmware and raspbian to try to use the I2C device concurrently, which can cause _lots_ of problems. I'll add backlight supports for `raspberrypi-fast-ts` so you don't need to use the sysfs device.

I haven't debugged it thoroughly, but it seems to be caused by libinput itself. I'll look into it

hoppler, 2020-10-31:
fyi: is was using this library to turn the screen off: https://github.com/timothyhollabaugh/pi-touchscreen-timeout/blob/master/timeout.c

with gdb it says its a nullpointer dereferencing inside of libinput always on the same address inside of libinput.
my guess is, with the change of available inputs (either by pluggin in/out a keyboard or restarting the `raspberrypi-fast-ts` service) the internal table of libinput for your handle gets invalid. a workaround could be to listen to those changes and before accessing libinput again to re-setup the connection to it.

closed 2020-10-31 by ardera

ardera, 2020-10-31:
> my guess is, with the change of available inputs (either by pluggin in/out a keyboard or restarting the `raspberrypi-fast-ts` service) the internal table of libinput for your handle gets invalid. a [...]

Libinput supports hotplugging, it'd be odd if it couldn't handle a simple device unplug.
It was actually a lot easier, I passed a stack pointer for the device open/close interface to libinput, because it was a `const struct` and I thought libinput would be copying it internally. It didn't.

hoppler, 2020-10-31:
tried it and it works. thx for the fast fix^^

fyi: running the flutter-pi with --no-text-input now causes a new segv^^

Thread 1 "flutter-pi" received signal SIGSEGV, Segmentation fault.
0x00030e0c in keyboard_state_new (config=0x0, keymap_override=0x0, compose_table_override=0x0) at src/keyboard.c:263
xkb_state = xkb_state_new(keymap_override != NULL ? keymap_override : config->default_keymap);

(gdb) bt
#0  0x00030e0c in keyboard_state_new (config=0x0, keymap_override=0x0, compose_table_override=0x0)  at src/keyboard.c:263
#1  0x0001c468 in on_libinput_ready (s=0x143a28, fd=9, revents=1, userdata=0x0) at src/flutter-pi.c:1962
#2  0x76ebd6a4 in  () at /lib/arm-linux-gnueabihf/libsystemd.so.0

ardera, 2020-10-31:
> Thread 1 "flutter-pi" received signal SIGSEGV, Segmentation fault.
> 0x00030e0c in keyboard_state_new (config=0x0, keymap_override=0x0, compose_table_override=0x0) at src/keyboard.c:263
> xkb_state = xkb_state_new(keymap_override != NULL ? keymap_override : config->default_keymap);

Thanks, fixed it. Though I should really just remove the `--no-text-input` flag. It was useful when flutter-pi was still using stdin for text/keyboard input, since it would switch it to raw/non-canonical mode which could cause problems on some occasions. Now, flutter-pi uses libinput for keyboard input, and converts the keyboard input to text input using `libxkbcommon` and the `/etc/default/keyboard` config file, so text input doesn't cause problems anymore.

hoppler, 2020-10-31:
thx again for the fast response. keep up the good work^^

PR #114 (2020-10-31): fix libinput segfault
- fixes #71