ardera/flutter-pi #73: drm not found

queirozrphl, 2020-08-06:
Hi folks, yesterday I have faced some issue that I never seem. 
When I was installing the flutter engine, with make, I got this error:

```
pi@raspberrypi:~/flutter-engine/flutter-pi $ make
Package libinput was not found in the pkg-config search path.
Perhaps you should add the directory containing `libinput.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libinput' found
cc -c -I./include  -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_GPIOD_PLUGIN -DBUILD_SPIDEV_PLUGIN -DBUILD_TEST_PLUGIN -DBUILD_OMXPLAYER_VIDEO_PLAYER_PLUGIN -O0 -ggdb  src/flutter-pi.c -o out/obj/flutter-pi.o
In file included from src/flutter-pi.c:30:
/usr/include/xf86drm.h:40:10: fatal error: drm.h: No such file or directory
 #include <drm.h>
```

I don't understand why this happened, I've installed this many times and that is the first time. 
Had someone got this error?

DisDis, 2020-08-06:
See https://github.com/ardera/flutter-pi/issues/65#issuecomment-669480141

queirozrphl, 2020-08-06:
Thany you very much! @DisDis 

> sudo apt-get install libsystemd-dev libinput-dev

Solved this problem
Sorry, I had not seem this question before.

closed 2020-08-06 by queirozrphl