ardera/flutter-pi #77: Memory leak.

DisDis, 2020-08-09:
Application eat all memory and crash.
I look at Observatory.
It has 3 isolate and consume 70+40+45 = 155mb
The consumption of isolates is always stable and does not grow, but the memory of consumption of the entire process increases.
But the process consume 428Mb. And step by step eat memory.
[image]
Any idea how to understand what's going on?
Before 1.20 and the flutter-pi update it was ok. :(

DisDis, 2020-08-09:
it is debug mode.

DisDis, 2020-08-09:
After 5 min
[image]

ardera, 2020-08-09:
Can you give me more details about your application? I think I experience memory leaks too, but maybe about 10MB with flutter_gallery and some usage. That's bad too and I'll fix it, but I'll probably be able to fix your specific issue faster if I knew what makes your app special / why your specific case leaks memory so much.

DisDis, 2020-08-09:
https://github.com/DisDis/dslideshow
1. create 'images' folder
2. add any image(jpeg,png, etc).

DisDis, 2020-08-14:
last master fixes it

queirozrphl, 2020-08-16:
I'm facing a similar problem, how did you solve this?

DisDis, 2020-08-16:
> I'm facing a similar problem, how did you solve this?

I checked the latest master and the problem is fixed. (Process memory is stable)
@queirozrphl

queirozrphl, 2020-08-16:
I've created a simple app, with SingleTickerProviderStateMixin. 
After around 40min, they consumed lot of memory, around 802mb. 

[image]

I'm using 1.20.0 with the latest binary provided

ardera, 2020-08-17:
Can you make sure this is flutter-pi related by checking the process memory stats in the dart observatory?
Or just send me the source code so I can reproduce it.

ardera, 2020-08-19:
I built myself an app with a `SingleTickerProviderStateMixin` (which drives an Animation), but couldn't reproduce the issue. I let it run over night and the virtual memory usage increased from 312M to 318M.

My code:
```dart
import 'package:flutter/material.dart';

void main() {
  runApp(MyApp());
}

class MyApp extends StatelessWidget {
  @override
  Widget build(BuildContext context) {
    return MaterialApp(
      title: 'Flutter Demo',
      theme: ThemeData(
        primarySwatch: Colors.blue,
        visualDensity: VisualDensity.adaptivePlatformDensity,
      ),
      home: MyHomePage(title: 'Flutter Demo Home Page'),
    );
  }
}

class MyHomePage extends StatefulWidget {
  MyHomePage({Key key, this.title}) : super(key: key);

  final String title;

  @override
  _MyHomePageState createState() => _MyHomePageState();
}

class _MyHomePageState extends State<MyHomePage> with SingleTickerProviderStateMixin {
  AnimationController _controller;

  @override
  void initState() {
    _controller = AnimationController(
      vsync: this,
      lowerBound: 0.0,
      upperBound: 1.0,
      duration: Duration(days: 1)
    );
    _controller.forward();
  }

  @override
  Widget build(BuildContext context) {
    return Scaffold(
      appBar: AppBar(
        title: Text(widget.title),
      ),
      body: Center(
        child: AnimatedBuilder(
          animation: _controller,
          builder: (context, child) {
            return Text("animation value: ${_controller.value}");
          }
        )
      ),
    );
  }
}

```

queirozrphl, 2020-08-19:
I do updated to last engine version and the problem vanish. 
I've tried to reproduce, but wasnt possible. 
I appreciate your attempts

DisDis, 2020-08-20:
the problem exists. the memory leak is very slow.

closed 2020-08-20 by ardera