ardera/flutter-pi #82: Unable to build the Asset bundle

bensinghbeno, 2020-08-16:
Hello Ardera,

I have succesfully performed the below steps,
(1) Setting Up Rpi
(2) Building & running fluttter-pi

I do not understand how to build a compatible flutter bundle.
Do you mean to build the bundle on the host?
I tried to build the bundle on the host for my simple app - https://github.com/bensinghbeno/design-engine/tree/master/projects/android/flutter_ImageDisplay &  I copied the assets to rpi.

On the Rpi, flutter-pi command says that the bundle snapshot is incompatible as below,

pi@raspberrypi:~/dev/flutter-pi/out $ ./flutter-pi /home/pi/dev/flutter-apps/flutter_assets
[flutter-pi] WARNING: display didn't provide valid physical dimensions.
             The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
===================================
display mode:
  resolution: 720 x 480
  refresh rate: 60Hz
  physical size: 0mm x 0mm
  flutter device pixel ratio: 1.000000
===================================
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless EGL_EXT_platform_device"
  display extensions: "EGL_ANDROID_blob_cache EGL_ANDROID_native_fence_sync EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
OpenGL ES information:
  version: "OpenGL ES 2.0 Mesa 19.3.2"
  shading language version: "OpenGL ES GLSL ES 1.0.16"
  vendor: "Broadcom"
  renderer: "VC4 V3D 2.1"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_npot GL_OES_vertex_half_float GL_OES_EGL_image GL_OES_depth_texture GL_AMD_performance_monitor GL_OES_packed_depth_stencil GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_EXT_occlusion_query_boolean GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_separate_shader_objects GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_texture_border_clamp GL_KHR_context_flush_control GL_OES_draw_elements_base_vertex GL_OES_texture_border_clamp GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_KHR_parallel_shader_compile GL_MESA_tile_raster_order "
===================================
detected VideoCore IV as underlying graphics chip, and VC4 as the driver.
Reporting modified GL_EXTENSIONS string that doesn't contain non-working extensions.
flutter: Observatory listening on http://127.0.0.1:42159/MABXKxyuYdA=/
[ERROR:engine/src/flutter/shell/common/shell.cc(209)] Dart Error: Can't load Kernel binary: Invalid kernel binary format version.
[ERROR:engine/src/flutter/shell/common/engine.cc(205)] Could not prepare to run the isolate.
[ERROR:engine/src/flutter/shell/common/engine.cc(144)] Engine not prepare and launch isolate.
[ERROR:engine/src/flutter/shell/common/shell.cc(486)] Could not launch engine with configuration.

In your repo I see that -  " You need to use a flutter SDK that's compatible to the engine version you're using."
Also I see that you are using flutter gallery...even though flutter-pi does not support it. These 2 are really confusing for me...sorry :)

How do i build a bundle that's compatible with rpi ?.. Kindly help!!

ardera, 2020-08-18:
> In your repo I see that - " You need to use a flutter SDK that's compatible to the engine version you're using."

This means that you can't use for example engine binaries with version 1.9 and a flutter SDK with version 1.20. If you do, the `Invalid Kernel binary format version` error will ocurr, like it did for you.
If you use the 1.20 engine binaries on your Raspberry Pi, you also need to use flutter 1.20 on your host machine to build the asset bundle.
In general, the engine binaries I provide are the latest stable ones. So to build your asset bundle, just use the latest stable SDK.

> Also I see that you are using flutter gallery...even though flutter-pi does not support it.

This is actually outdated, flutter-pi does now support flutter gallery (Or more precisely, flutter gallery now works better with flutter-pi). You just need to be careful to choose the right version of flutter_gallery, since it's developed against flutter master.

> How do i build a bundle that's compatible with rpi ?.. Kindly help!!

It's actually very easy:
1. select the `stable` channel for your flutter SDK. cmd: `flutter channel stable`
2. make sure your flutter SDK is up to date. cmd: `flutter upgrade`
3. build the bundle of your app. cmd: `flutter build bundle`

bensinghbeno, 2020-08-19:
Thanks a lot...uve explained pretty nicely...Let me go for it like u said n get back :)

bensinghbeno, 2020-08-20:
Hi @ardera ,

Thanks a lot.. The debug build works like a charm :) ..
 My next hurdle is when I try to build the release version app.so.

At Step -3, I launch dart from my linux machine, but get the below error,

flutter/bin/cache/dart-sdk/bin$ ./dart   ./snapshots/frontend_server.dart.snapshot   --sdk-root /home/ben/my/dev/opensrc/flutters/chinmaygarde/flutter/bin/cache/artifacts/engine/common/flutter_patched_sdk_product   --target=flutter   --aot   --tfa   -Ddart.vm.product=true   --packages .packages   --output-dill /home/ben/my/dev/opensrc/flutters/chinmaygarde/build/kernel_snapshot.dill   --verbose   --depfile /home/ben/my/dev/opensrc/flutters/chinmaygarde/build/kernel_snapshot.d   --package:flutter_app1/main.dart

Unhandled exception:
FormatException: Scheme not starting with alphabetic character (at character 1)
--package:flutter_app1/main.dart
^

#0      _Uri._fail (dart:core/uri.dart:1628:5)
#1      _Uri._makeScheme (dart:core/uri.dart:2167:7)
#2      new _Uri.notSimple (dart:core/uri.dart:1459:18)
#3      Uri.parse (dart:core/uri.dart:1022:17)
#4      parseUri (package:front_end/src/fasta/resolve_input_uri.dart:27:14)
#5      resolveInputUri (package:front_end/src/fasta/resolve_input_uri.dart:12:11)
#6      FrontendCompiler.compile (file:///b/s/w/ir/cache/builder/src/third_party/dart/pkg/frontend_server/lib/frontend_server.dart:395:19)
#7      starter (file:///b/s/w/ir/cache/builder/src/third_party/dart/pkg/frontend_server/lib/frontend_server.dart:1430:27)
#8      main (file:///b/s/w/ir/cache/builder/src/third_party/dart/pkg/frontend_server/bin/frontend_server_starter.dart:9:20)
#9      _startIsolate.<anonymous closure> (dart:isolate-patch/isolate_patch.dart:299:32)
#10     _RawReceivePortImpl._handleMessage (dart:isolate-patch/isolate_patch.dart:168:12)

**I hope I have put my package name correctly as "flutter_app1" as I found it in the Android Manifest.xml as "    package="com.example.flutter_app1".

Kindly let me know what Ive done wrong!!**