ardera/flutter-pi #83: Slow Start / Performance/  Launch of app even in release mode

bensinghbeno, 2020-08-24:
Dear Ardera,
Thanks for your great support with release build earlier.

I have built assets for my small flutter app that loads an image file and displays the image (simple app).
https://github.com/bensinghbeno/design-engine/blob/master/projects/android/flutter_ImageDisplay/lib/main.dart

Although I see that the performance of the app is very slow (Takes 9 Seconds to show UI). Even though we disable UI and run as CLI. I have also removed x11 completely.

 I have used the commanline option "--disable-observatory" that improves the launch of the app by 1 second.

The results of command are below,

pi@raspberrypi:~/dev/flutter-pi/out $ ./flutter-pi ../../flutter_assets_rel/ --release --disable-observatory
[flutter-pi] WARNING: display didn't provide valid physical dimensions.
             The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
===================================
display mode:
  resolution: 1920 x 1080
  refresh rate: 60Hz
  physical size: 0mm x 0mm
  flutter device pixel ratio: 1.000000
===================================
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless EGL_EXT_platform_device"
  display extensions: "EGL_ANDROID_blob_cache EGL_ANDROID_native_fence_sync EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
OpenGL ES information:
  version: "OpenGL ES 2.0 Mesa 19.3.2"
  shading language version: "OpenGL ES GLSL ES 1.0.16"
  vendor: "Broadcom"
  renderer: "VC4 V3D 2.1"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_npot GL_OES_vertex_half_float GL_OES_EGL_image GL_OES_depth_texture GL_AMD_performance_monitor GL_OES_packed_depth_stencil GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_EXT_occlusion_query_boolean GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_separate_shader_objects GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_texture_border_clamp GL_KHR_context_flush_control GL_OES_draw_elements_base_vertex GL_OES_texture_border_clamp GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_KHR_parallel_shader_compile GL_MESA_tile_raster_order "
===================================
detected VideoCore IV as underlying graphics chip, and VC4 as the driver.
Reporting modified GL_EXTENSIONS string that doesn't contain non-working extensions.
[flutter-pi] Error handling platform message. plugin_registry_on_platform_message: Invalid argument

UI SHOWS UP After this,, It takes >= 9 Seconds :( ..

Kindly let me know if you can take a look !!! Thanks again!!

ardera, 2020-08-24:
The time flutter-pi takes to launch an app is mostly spend on disk IO. So I recommend a fast SD-card if you haven't already got one. I personally use a SanDisk Extreme Plus SD-card, that's the fastest sd-card for random access according to some benchmarking site.

Starting flutter_gallery in release mode takes about 1 second for me, in debug mode 5.5 seconds.

bensinghbeno, 2020-08-25:
Hmm..I can check it...I beleive u have a Rpi4 ?. Mine is rpi3. Also the performance of Qt app on the same sd card / raspbian / same application logic  is 8 times faster . Are there any other ways u can think of to optimize the performance?

DisDis, 2020-08-25:
@bensinghbeno 
Are you sure the release mode is working?
flutter-pi help info:
> flutter-pi [options] <asset bundle path> [flutter engine options]

You exec: 

> ./flutter-pi ../../flutter_assets_rel/ **--release** --disable-observatory

Try:
./flutter-pi **--release** ../../flutter_assets_rel/ --disable-observatory

bensinghbeno, 2020-08-25:
@DisDis : Wow...youv'e got eagle eyes... You are exactly right.

After trying as below,
./flutter-pi --release ../../flutter_assets_rel/ --disable-observatory 

The app launches in 3 seconds....Thank you so much !!!

closed 2020-08-25 by bensinghbeno

ardera, 2020-08-25:
> You exec:
> 
> > ./flutter-pi ../../flutter_assets_rel/ **--release** --disable-observatory

Oh, I didn't notice that!
Yeah, also the "debug" text in the top right corner of the UI is a good indicator of whether your app is running in debug or release mode.

bensinghbeno, 2020-08-25:
Good to know..Thanks @ardera ...Ill share out my performance analysis with you guys soon :)