ardera/flutter-pi #84: Hello, Could not load libflutter_engine.so, dlopen: Success

vcxk, 2020-09-01:
I'm trying flutter-pi, but have some trouble.

I use flutter 1.20.2 to build bundle

When I run ./flutter-pi ~/Downloads/build/build/flutter_assets/.

It reports

> [flutter-pi] WARNING: display didn't provide valid physical dimensions.
             The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
===================================
display mode:
  resolution: 720 x 480
  refresh rate: 60Hz
  physical size: 0mm x 0mm
  flutter device pixel ratio: 1.000000
===================================
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless EGL_EXT_platform_device"
  display extensions: "EGL_ANDROID_blob_cache EGL_ANDROID_native_fence_sync EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
OpenGL ES information:
  version: "OpenGL ES 2.0 Mesa 19.3.2"
  shading language version: "OpenGL ES GLSL ES 1.0.16"
  vendor: "Broadcom"
  renderer: "VC4 V3D 2.1"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_npot GL_OES_vertex_half_float GL_OES_EGL_image GL_OES_depth_texture GL_AMD_performance_monitor GL_OES_packed_depth_stencil GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_EXT_occlusion_query_boolean GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_separate_shader_objects GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_texture_border_clamp GL_KHR_context_flush_control GL_OES_draw_elements_base_vertex GL_OES_texture_border_clamp GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_KHR_parallel_shader_compile GL_MESA_tile_raster_order "
===================================
[flutter-pi] Warning: Could not load libflutter_engine.so.debug. Trying to open libflutter_engine.so...
[flutter-pi] Could not load libflutter_engine.so. dlopen: Success
[flutter-pi] Could not find a fitting libflutter_engine.

ardera, 2020-09-01:
This means `libflutter_engine.so.debug` is not installed.
The instructions for installing it are [here](https://github.com/ardera/flutter-pi/blob/engine-binaries/README.md). I also just realized I had a small mistake in the instructions (more precisely, the shell commands), that may be the reason why it didn't work for you.
Can you try again with the updated instructions?

vcxk, 2020-09-01:
Oh,It works. I have put the so to /usr/local/lib ./out and assets directory.
Thank you.

vcxk, 2020-09-01:
enn..
The flutter gallery hang up after it report draw call returned Cannot allocate memory. Expect corruption.
The full output is here:

> [flutter-pi] WARNING: display didn't provide valid physical dimensions.
             The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
===================================
display mode:
  resolution: 1280 x 800
  refresh rate: 60Hz
  physical size: 0mm x 0mm
  flutter device pixel ratio: 1.000000
===================================
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless EGL_EXT_platform_device"
  display extensions: "EGL_ANDROID_blob_cache EGL_ANDROID_native_fence_sync EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
OpenGL ES information:
  version: "OpenGL ES 2.0 Mesa 19.3.2"
  shading language version: "OpenGL ES GLSL ES 1.0.16"
  vendor: "Broadcom"
  renderer: "VC4 V3D 2.1"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_npot GL_OES_vertex_half_float GL_OES_EGL_image GL_OES_depth_texture GL_AMD_performance_monitor GL_OES_packed_depth_stencil GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_EXT_occlusion_query_boolean GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_separate_shader_objects GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_texture_border_clamp GL_KHR_context_flush_control GL_OES_draw_elements_base_vertex GL_OES_texture_border_clamp GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_KHR_parallel_shader_compile GL_MESA_tile_raster_order "
===================================
detected VideoCore IV as underlying graphics chip, and VC4 as the driver.
Reporting modified GL_EXTENSIONS string that doesn't contain non-working extensions.
flutter: Observatory listening on http://127.0.0.1:43993/8-K2wwqtmk0=/
[flutter-pi] Error handling platform message. plugin_registry_on_platform_message: Invalid argument
[compositor] Preferred framebuffer depth for hardware cursor is not supported by flutter-pi.
[flutter-pi] Error handling platform message. plugin_registry_on_platform_message: Invalid argument
[flutter-pi] Error handling platform message. plugin_registry_on_platform_message: Invalid argument
[flutter-pi] Error handling platform message. plugin_registry_on_platform_message: Invalid argument
[flutter-pi] Error handling platform message. plugin_registry_on_platform_message: Invalid argument
[flutter-pi] Error handling platform message. plugin_registry_on_platform_message: Invalid argument
flutter: ColorScheme#fc923(primary: Color(0xffff8383), primaryVariant: Color(0xff1cdec9), secondary: Color(0xff4d1f7c), secondaryVariant: Color(0xff451b6f), surface: Color(0xff1f1929), background: Color(0xff241e30), error: Color(0xffffffff), onSecondary: Color(0xffffffff), onSurface: Color(0xffffffff), onBackground: Color(0x0dffffff), brightness: Brightness.dark)
flutter: ColorScheme#fc923(primary: Color(0xffff8383), primaryVariant: Color(0xff1cdec9), secondary: Color(0xff4d1f7c), secondaryVariant: Color(0xff451b6f), surface: Color(0xff1f1929), background: Color(0xff241e30), error: Color(0xffffffff), onSecondary: Color(0xffffffff), onSurface: Color(0xffffffff), onBackground: Color(0x0dffffff), brightness: Brightness.dark)
[ERROR:engine/src/flutter/lib/ui/ui_dart_state.cc(166)] Unhandled Exception: MissingPluginException(No implementation found for method canLaunch on channel plugins.flutter.io/url_launcher)
#0      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:159:7)
<asynchronous suspension>
#1      MethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart:334:12)
#2      MethodChannelUrlLauncher.canLaunch (package:url_launcher_platform_interface/method_channel_url_launcher.dart:18:21)
#3      canLaunch (package:url_launcher/url_launcher.dart:116:45)
#4      _GalleryDemoPageState._showDocumentation (package:gallery/pages/demo.dart:173:15)
[... cut ...]
detected VideoCore IV as underlying graphics chip, and VC4 as the driver.
Reporting modified GL_EXTENSIONS string that doesn't contain non-working extensions.
flutter: Observatory listening on http://127.0.0.1:46175/lSFYvGVjD-A=/
Draw call returned Invalid argument.  Expect corruption.
[flutter-pi] Error handling platform message. plugin_registry_on_platform_message: Invalid argument
[compositor] Preferred framebuffer depth for hardware cursor is not supported by flutter-pi.

And after reboot ,It works again.

vcxk, 2020-09-01:
draw call return .... again 
and some images disappear

ardera, 2020-09-01:
Yeah then you may have too little memory on your device to run flutter gallery in debug mode.
Please make sure you've given the GPU 16M (or 64M if you want to use the omxplayer_video_player plugin) of RAM in `raspi-config`.
If that doesn't work, the only thing left to try is building the app in release mode. Release mode has a lot less memory usage.

vcxk, 2020-09-01:
but it need reboot device and then i can reopen application. emm... i will try it tommorow。

closed 2020-09-14 by vcxk

MobileAppGwt, 2021-01-21:
> This means `libflutter_engine.so.debug` is not installed.
> The instructions for installing it are [here](https://github.com/ardera/flutter-pi/blob/engine-binaries/README.md). I also just realized I [...]

Hi @vcxk @ardera I have also face this same error, could anyone explain in more detail.

Thanks in advance :)

ardera, 2021-01-21:
> Hi @vcxk @ardera I have also face this same error, could anyone explain in more detail.

Hi, what error do you have?
The `dlopen: Success` error or the cannot allocate memory error?

- If you have the `dlopen` error, make sure you've installed the engine binaries correctly (correct architecture, correct location)

- If you have the out of memory error, there really isn't a single way to solve this. You can try lowering the GPU memory to 16M in `raspi-config` (instructions in the readme) and using release mode instead (instructions in readme too).

Tek-hp, 2022-10-31:
> debug

I tired to follow the linked documentation, Can you specify more clearly
?