ardera/flutter-pi #85: Problems with: Building the app.so (for running your app in Release/Profile mode)

CAoTx, 2020-09-01:
I hope that this isn't an issue but rather a failure of me. 

I'm trying to build the app.so to run the app in release mode on my Raspberry. 
Anyhow, when I come to step 4 - Building the app.so, I get the following error: 

```
Unable to open file build/kernel_snapshot.dill
../../../engine/src/third_party/dart/runtime/vm/dart_api_impl.cc: 2153: error: Dart_ExitScope expects there to be a current isolate. Did you forget to call Dart_CreateIsolateGroup or Dart_EnterIsolate?
Aborted (core dumped)
sbey@nunki:~/Projects/
```

The following commands were made:
3. Inside the app directory: 
```
~/flutter/bin/cache/dart-sdk/bin/dart \
        ~/flutter/bin/cache/dart-sdk/bin/snapshots/frontend_server.dart.snapshot \
        --sdk-root ~/flutter/bin/cache/artifacts/engine/common/flutter_patched_sdk_product \
        --target=flutter \
        --aot \
        --tfa \
        -Ddart.vm.product=true \
        --packages .packages \
        --output-dill build\kernel_snapshot.dill \
        --verbose \
        --depfile build\kernel_snapshot.d \
        package:flutterReference/main.dart
```
4. After downloading the engine_binary tree:
```
~/engine-binaries/gen_snapshot_linux_x64 \
        --causal_async_stacks \
        --deterministic \
        --snapshot_kind=app-aot-elf \
        --elf=build/app.so \
        --strip \
        --sim_use_hardfp \
        --no-use-integer-division \ 
        build/kernel_snapshot.dill
```

CAoTx, 2020-09-01:
I moved the both files **buildkernel_snapshot.d** & **buildkernel_snapshot.dill** into the build directory and renamed the buildkernel_snapshot.dill to **kernel_snapshot.dill**. 

Now the 4. runs through 

Copied to the Raspberry. 
On Rpi - inside the build folder:

When running
`flutter-pi --release flutter_assets`

The following error occurs: 
`[flutter-pi] Could not find "app.so" file inside "flutter_assets/", which is required for release and profile mode.`

After copying app.so from **build** into **build/flutter_assets** everything works fine.

I'm still testing if this is running as release version or not. Until then I will suggest the following improvements to help others avoid these problems and avoid the need for U to help them: 

1. Clear out that **gen_snapshot_linux_x64** has to be made executable: 
`sudo chmod +x gen_snapshot_linux_x64`
2. Copy the **buildkernel_snapshot.d** & **buildkernel_snapshot.dill** into the build folder
3. Rename the **buildkernel_snapshot.dill** into **kernel_snapshot.dill**
4. After Step 4, the **app.so** lays in the **build** folder rather than in the expected **build/flutter_assets** folder, copy it into **flutter_assets**

ardera, 2020-09-01:
> ```
> ~/flutter/bin/cache/dart-sdk/bin/dart \
>         ~/flutter/bin/cache/dart-sdk/bin/snapshots/frontend_server.dart.snapshot \
>         --sdk-root [...]

You need to change all `\` to `/` if you're running under linux. So `--output-dill build\kernel_snapshot.dill` should be changed into `--output-dill build/kernel_snapshot.dill` on linux.

> **After that, the both files buildkernel_snapshot.d & buildkernel_snapshot.dill were created in the root directory of my app rather than in the build folder. Is that right?**

It's not right, but it's also not really a problem. `build/kernel_snapshot.dill` is just the default paths. The name of the file and it's path is pretty irrelevant. The only thing that matters is that you give the `gen_snapshot_linux_x64` the correct path to the `.dill` file. (Which is where you went wrong, your `.dill` file was called `buildkernel_snapshot.dill` but the path you gave to `gen_snapshot_linux_x64` was `build/kernel_snapshot.dill`)

closed 2020-10-08 by ardera