ardera/flutter-pi #87: gen_snapshot_linux_x64: cannot execute binary file: Exec format error

JLiao98, 2020-09-09:
I use the gen_snapshot_linux_x64 executable that you provide in the engine-binaries branch. However it does not work and it shows that it cannot execute binary file.

I used `chmod 777 gen_snapshot_linux_x64` too.

JLiao98, 2020-09-09:
For the step of: Building the app.so (for running your app in Release/Profile mode), is a little bit confusing. Do you mind giving a walkthrough?

And also, I built the kernel snapshot on Mac OS, do I expect to build `app.so` on Mac too or Linux on my RPi?

Thank you!

ardera, 2020-09-09:
> I use the gen_snapshot_linux_x64 executable that you provide in the engine-binaries branch. However it does not work and it shows that it cannot execute binary file.

> And also, I built the kernel snapshot on Mac OS, do I expect to build `app.so` on Mac too or Linux on my RPi?

The `gen_snapshot_linux_x64` executable sadly needs to be executed on linux running on a x64 machine. This is because that `gen_snapshot` executable is actually only a side-product of compiling the engine. If I wanted to build a `gen_snapshot` that can be run on MacOS or Windows, I'd have to maintain multiple cross-compiling toolchains. You can try running the `gen_snapshot_linux_x64` in a virtual machine though

The machine you invoke `gen_snapshot` on is irrelevant though, it always builds an `app.so` that can be used on the Raspberry Pi. (More precisely, an `app.so` that is compiled for ARMv7 and uses hardfp ABI). The `app.so` target architecture, ABI, etc was specified when compiling the `gen_snapshot` executable.

Btw, that `app.so` is the only file containing platform-specific code. The `kernel_snapshot.dill`, `kernel_blob.bin` file etc only contain dart bytecode.

> For the step of: Building the app.so (for running your app in Release/Profile mode), is a little bit confusing. Do you mind giving a walkthrough?

The text I wrote in the readme was meant as a walktrough 😅 If you tell me what you find confusing I can try to clarify it

JLiao98, 2020-09-09:
> > I use the gen_snapshot_linux_x64 executable that you provide in the engine-binaries branch. However it does not work and it shows that it cannot execute binary file.
> 
> > And also, I built the kernel snapshot on Mac OS, do I expect to build `app.so` on Mac too or Linux on my RPi?
> 
> The `gen_snapshot_linux_x64` executable sadly needs to be executed on linux running on a x64 machine. This is because that `gen_snapshot` executable is actually only a side-product of compiling the engine. If I wanted to build a `gen_snapshot` that can be run on MacOS or Windows, I'd have to maintain multiple cross-compiling toolchains. You can try running the `gen_snapshot_linux_x64` in a virtual machine though
> 
> Regarding the outputted `app.so`, the machine you invoke `gen_snapshot` on is irrelevant. It always builds an `app.so` that can be used on the Raspberry Pi. (More precisely, an `app.so` that is compiled for ARMv7 and uses hardfp ABI). The `app.so` target architecture, ABI, etc was specified when compiling the `gen_snapshot` executable.
> 
> Btw, that `app.so` is the only file containing platform-specific code. The `kernel_snapshot.dill`, `kernel_blob.bin` file etc only contain dart bytecode.
> 
> > For the step of: Building the app.so (for running your app in Release/Profile mode), is a little bit confusing. Do you mind giving a walkthrough?
> 
> The text I wrote in the readme was meant as a walkthrough 😅 If you tell me what you find confusing I can try to clarify it

Thanks. I think I got it. I can try to compile `gen_snapshot` on Mac OS. 

And I have another question. I run the Flutter app on `flutter-pi` from SSH, and I saw that there is a debugger running: `flutter: Observatory listening on http://127.0.0.1:41511/tcgmfJ-m1bA=/`. However, I can't access this page and I think that would be a really good feature for debugging since debugging on `flutter-pi` is difficult.

And also, how can I quit the current flutter app and go back to the console? My current way is to close the SSH connection and it will close. But I also use the SEGFAULT bug to terminate by plugging in a keyboard lol.

ardera, 2020-09-09:
> And I have another question. I run the Flutter app on `flutter-pi` from SSH, and I saw that there is a debugger running: `flutter: Observatory listening on http://127.0.0.1:41511/tcgmfJ-m1bA=/`. [...]

Exactly, that's the dart observatory. If you would enter this URL on a browser on the Pi, you would see it. "127.0.0.1" always means localhost, so if you enter that url on your Mac, it tries accessing :41511 on your mac, and that fails of course.

There are 2 engine options that help us out here: `--observatory-host <ip addr>` and `--observatory-port <port>`
In your case, just find out the local IP address of your pi (maybe using `ifconfig`). You can use any port you want, like 8080.

The engine options are specified after the asset bundle path in the flutter-pi invocation, so with `192.168.178.11` as the IP address and 8080 as the port it'd look like this:
`flutter-pi ~/devel/my_flutter_assets --observatory-host 192.168.178.11 --observatory-port 8080`

JLiao98, 2020-09-09:
> Exactly, that's the dart observatory. If you would enter this URL on a browser on the Pi, you would see it. "127.0.0.1" always means localhost, so if you enter that url on your Mac, it tries [...]

I tried putting `--observatory-host <ip addr>` and `--observatory-port <port>` when I ran the app on flutter-pi. But still I cannot access the dart observatory. I tried all these IP addresses like: http://<Pi_IP_addr>:<Pi_Port>/2Cark3_13Yo=/, http://<Pi_IP_addr>:<Pi_Port>. What's the correct way to access to the dart observatory?

ardera, 2020-09-11:
Sorry, the engine options need to be `--observatory-host=<ip addr>` and `--observatory-port=<port>`. That was my bad, I always forget about the equals sign.

closed 2020-10-08 by ardera