ardera/flutter-pi #88: Question: release with Raspberry4 support?

apiraino, 2020-09-10:
Hi!

I'm trying to run the release binary from november 2019, I get this error:

```
pi@raspberrypi:~ $ ./flutter-pi -m /dev/input/event0 -t /dev/input/event0 flutter_gallery_assets/flutter_assets/ -r 90
Using mouse input from mouse /dev/input/event0
Using touchscreen input from /dev/input/event0
Asset bundle path: flutter_gallery_assets/flutter_assets/
engine_argv[0] = ./flutter-pi
engine_argv[1] = -r
engine_argv[2] = 90
initializing display...
Opening DRM device...
Getting DRM resources...
/dev/dri/card0 doesn't look like a modeset device
```

Possibly the binary still does not support the Rpi4 (possibly related to [this issue](https://github.com/ardera/flutter-pi/issues/16)) and I have to manually compile, correct?

I have run all the steps to build the Flutter app assets, `app.so` and downloaded the engine binaries from [this branch](https://github.com/ardera/flutter-pi/tree/engine-binaries).

I have customized the raspi according [to your instructions](https://github.com/ardera/flutter-pi/tree/master#configuring-your-raspberry-pi).

thanks for your project :+1:

falkorichter, 2020-09-10:
@apiraino did you try to compile the latest master? there is a make file :)

ardera, 2020-09-10:
The november 2019 is very very outdated. It doesn't support the Pi 4 and it also doesn't support rotation. (`-r 90`).

But be aware that the invocation is `flutter-pi <flutter-pi options> <asset bundle path> <flutter engine options>`. So even if your `flutter-pi` was latest master, the `-r 90` argument would not be recognized since you specified it as a flutter engine option, after the asset bundle path. It's a flutter-pi options though, so it needs to be before the asset bundle path. The flutter-pi options are the ones listed in the usage. flutter engine options are the one you can find in the `switches.cc` file (link in the usage), for example `--observatory-host`, `--observatory-port`

closed 2020-09-10 by apiraino