ardera/flutter-pi #89: Cannot build app.so with Dart 2.9.2: "Can't load Kernel binary: Invalid SDK hash"

apiraino, 2020-09-10:
Hey,

I've updated the Dart+Flutter SDK to the very latest _stable_:

```
$ flutter --version
Flutter 1.20.3 • channel stable • https://github.com/flutter/flutter.git
Framework • revision 216dee60c0 (9 days ago) • 2020-09-01 12:24:47 -0700
Engine • revision d1bc06f032
Tools • Dart 2.9.2
```

I can compile the kernel snapshot, but I can't compile anymore the `app.so`.

```
$ gen_snapshot_linux_x64 --causal_async_stacks --deterministic --snapshot_kind=app-aot-elf --elf=build/app.so --strip --sim_use_hardfp --no-use-integer-division build/kernel_snapshot.dill 
Can't load Kernel binary: Invalid SDK hash.
```

Can you detail a little bit how is the `gen_snapshot_linux_x64` binary you provide is generated and why it fails? Do I have to match the exact same Dart SDk version as yours?

Thanks!

apiraino, 2020-09-10:
Possibly relatd [to this](https://github.com/dart-lang/pub/issues/2593)

ardera, 2020-09-11:
Hey,

> I can compile the kernel snapshot, but I can't compile anymore the `app.so`.
>
> ```
> $ gen_snapshot_linux_x64 \
>   --causal_async_stacks \
>   --deterministic \
>   --snapshot_kind=app-aot-elf \ [...]
That's not your fault. `gen_snapshot_linux_x64` is a side-product of the flutter engine compilation. A flutter engine version (which is a commit **hash**) maps one-to-one to an SDK version (which is a commit **hash** as well). So one specific version of `gen_snapshot_linux_x64` can only accept `kernel_snapshot.dill`'s from one specific version of the flutter SDK. This error message is basically saying that the version of the flutter SDK that built this `kernel_snapshot.dill` is wrong & not the version it expects to find.

**TL;DR** there was an update of the stable flutter SDK 9 days ago and I haven't yet updated the engine binaries. I'm compiling right now. These version checks are new and I haven't yet found a way to get notified of flutter stable updates

apiraino, 2020-09-11:
Ah thank you so much for clarifying, now the whole picture is getting clearer (see also the answer I received in the above linked dart pub issue). I was confused because I thought that "stable is stable" and a minor version update would make no difference :-)

Next release of the `gen_snapshot_linux_x64` you're going to do, it would be great if you can pin the _exact_ SDK version (or version range) that can be used.

Also, being able to build that by myself would be a huge plus, but I understand that documenting the process will take some time :-)

Thank you so much!

ardera, 2020-09-11:
> Next release of the `gen_snapshot_linux_x64` you're going to do, it would be great if you can pin the _exact_ SDK version (or version range) that can be used.

That's a good idea, I'll do that

> Also, being able to build that by myself would be a huge plus, but I understand that documenting the process will take some time :-)

To do that, just build the engine (in release mode). After you've done that, the `gen_snapshot` executable is located inside the `clang_x64` directory in the output directory.

Building the engine is kinda complicated and takes a long time. #49 is an open issue about that, you can find some instructions there.

ardera, 2020-09-11:
> Next release of the `gen_snapshot_linux_x64` you're going to do, it would be great if you can pin the _exact_ SDK version (or version range) that can be used.

Some info about that: `Invalid SDK hash` actually means `Invalid dart SDK hash`. I can (and will) give the exact dart SDK version that's compatible with this engine in the repo (see `dart-sdk.version`). It's impossible to specify exact flutter SDK version ranges though. Best I can do is a lower bound, i.e. specifying that flutter SDK 1.20.3 started using engine X and dart SDK Y.

Btw you can also see the engine version your flutter SDK is using when running `flutter upgrade`.

apiraino, 2020-09-11:
I have downloaded the new binary and successfully build the `app.so`, great job! Now I know that I must not change a single bit in this workflow :-)

In the following days I will try running my Flutter app and see if I can make run on the Rpi4. If it's ok for you, I'd keep this issue open in case I have further questions. I will close it anyway when I'll be done.

Thanks :+1:

ardera, 2020-09-12:
> In the following days I will try running my Flutter app and see if I can make run on the Rpi4. If it's ok for you, I'd keep this issue open in case I have further questions. I will close it anyway when I'll be done.
> 
> Thanks 👍

Yep, sure. No problem

closed 2020-10-08 by ardera