ardera/flutter-pi #91: Flutter app don't show anything

shiipou, 2020-09-16:
Hi, I've build flutter-pi on my raspberry pi with all dependency on the README of the project, but my screen didn't show the flutter app on the raspberry pi. The Flutter app was a fresh new app generated with `flutter create launcher` and patched with :
```dart
import 'package:flutter/foundation.dart'
[...]
debugDefaultTargetPlatformOverride = TargetPlatform.fuchsia;
```

I can only see my mouse cursor in black on top of my terminal ouput.

Can anyone help me ?

[image]

Here all commands I've run on my RPI : 
```bash
pi@raspberrypi:~ $ history 
    1  raspi-config 
    2  sudo raspi-config 
    3  git clone https://github.com/ardera/flutter-pi
    4  sudo apt install build-essential
    5  sudo apt install git
    6  git clone https://github.com/ardera/flutter-pi
    7  cd flutter
    8  cd flutter-pi/
    9  ls
   10  sudo apt install libgl1-mesa-dev libgles2-mesa-dev libegl-mesa0 libdrm-dev libgbm-dev
   11  sudo apt install ttf-mscorefonts-installer fontconfig
   12  sudo fc-cache
   13  sudo apt-get install gpiod libgpiod-dev libsystemd-dev libinput-dev libudev-dev
   14  make
   15  git clone --depth 1 --branch engine-binaries https://github.com/ardera/flutter-pi.git engine-binaries
   16  cd engine-binaries
   17  sudo cp ./libflutter_engine.so.* ./icudtl.dat /usr/lib
   18  sudo cp ./flutter_embedder.h /usr/include
   19  cd ..
   20  make
   21  sudo usermod -aG render
   22  sudo usermod -aG render $USER
   23  mv flutter-pi/out/flutter-pi /usr/bin
   24  sudo mv flutter-pi/out/flutter-pi /usr/bin
   25  flutter-pi flutter_assets/
   26  flutter-pi launcher/build/flutter_assets/
   27  sudo reboot
   28  flutter-pi launcher/build/flutter_assets/
   29  flutter-pi -i /dev/input/event{0,1} -i /dev/input/event{2,3} launcher/build/flutter_assets/
   30  usermod -aG render pi
   31  sudo usermod -aG render pi
   32  reboot 
   33  flutter-pi launcher/build/flutter_assets/
   34  history 
```

shiipou, 2020-09-16:
Ok after some investigation of my own, This issue was now fixed.
I've just need to switch back to stable with `flutter channel stable` and run `flutter upgrade`.

closed 2020-09-16 by shiipou

ardera, 2020-09-16:
As a sidenote, the `debugDefaultTargetPlatformOverride = TargetPlatform.fuchsia;` is no longer necessary, since flutter supports linux out of the box now. It's best to not specify any override at all now.

shiipou, 2020-09-16:
> As a sidenote, the `debugDefaultTargetPlatformOverride = TargetPlatform.fuchsia;` is no longer necessary, since flutter supports linux out of the box now. It's best to not specify any override at all now.

Ok thank you for the note. That look better for me.