ardera/flutter-pi #95: Running the given steps on MAC

sachin-puranik, 2020-10-07:
Hi,
Thanks you have done great work. I have a question though.
I am able to run few steps on MAC (Specifically till step 3)
and unable to execute step 4 as gen_snapshot_linux_x64 is not available for MAC.

Can you help or guide on this?

ardera, 2020-10-08:
Yeah, sadly I can't provide the `gen_snapshot` binary for macOS.

There's a few things that may help:
- use `noah` or a VM to run it under macOS (see [this stackoverflow question](https://unix.stackexchange.com/questions/212754/is-there-a-way-to-run-a-linux-binary-on-macos))
- use actual linux to execute the `gen_snapshot_linux_x64` natively (live cd/usb, or a real install)
- run the app in debug mode instead
- build the engine yourself on your macOS machine, which will yield a `gen_snapshot` that will run under macOS

vlidholt, 2020-11-08:
Great work on this project @ardera!

I finally got it working on my Mac. Some things I tried:

- Noah seems to be a no-go, it hasn't been maintained for years and couldn't get it working at all
- Debug mode works, but performance is slow for production apps
- I tried building the engine, and got a native gen_snapshot. This would have been a good solution, if it didn't produce the app.so file in an incompatible format. Couldn't find any way to build it so it makes the snapshot compatible with Raspberry. Any ideas if this is possible?

I ended up using VirtualBox with Ubuntu, which works, but feels like overkill for the task. It's possible to run commands remotely from the Mac terminal using VBoxManage, so can at least incorporate it in a build script.

For step 3, I used the following command:

    dart /Users/viktor/Projects/flutter/bin/cache/dart-sdk/bin/snapshots/frontend_server.dart.snapshot --sdk-root /Users/viktor/Projects/flutter/bin/cache/artifacts/engine/common/flutter_patched_sdk_product --target=flutter --aot --tfa -Ddart.vm.product=true --packages .packages --output-dill build/kernel_snapshot.dill --verbose --depfile build/kernel_snapshot.d lib/main.dart

ardera, 2020-11-08:
> Great work on this project @ardera!

Thanks, I'm glad you like it!

> * I tried building the engine, and got a native gen_snapshot. This would have been a good solution, if it didn't produce the app.so file in an incompatible format. Couldn't find any way to build it so it makes the snapshot compatible with Raspberry. Any ideas if this is possible?

The target of `gen_snapshot` matches the engine target you configured using `gn`. I always give `gn` the following target args:
```
--target-triple arm-linux-gnueabihf
--target-os linux
--linux-cpu arm
--arm-float-abi hard
```
(of course, I also specified other flags like `--runtime-mode release`, `--no-lto`, `--embedder-for-target` etc)

I'd compile the mac binaries myself, but I don't have a single mac machine to test the binaries. Also, I fear it's going to be a configuration nightmare, since I have to differentiate between 3 architectures - the build machine (linux x86-64), host machine (macos x86-64) and target machine (linux armhf). Often cross compilation with 2 architectures is difficult enough.

Also, when you clone the engine sources, you need to make sure you're using the latest stable sources (instead of latest master), like this:
```
gclient sync --revision=src/flutter@$(wget https://raw.githubusercontent.com/flutter/flutter/stable/bin/internal/engine.version -q -O -)
```

vlidholt, 2020-11-08:
Thank you @ardera!

Anywhere I can see the exact flags you are passing to gn? Will post an update as soon as I've tried this out.

vlidholt, 2020-11-08:
Did some more experimentation. It looks like it might not be viable on a Mac, or at least very hard. When running `gn` I get the following error:
`Missing sysroot (/Users/viktor/Projects/engine/src/build/linux/debian_sid_amd64-sysroot). To fix, run: build/linux/sysroot_scripts/install-sysroot.py --arch=x64`
Trying to run the sysroot install script doesn't do anything, probably because it needs to download some linux stuff I'm guessing by eyeing the install script quickly.

ardera, 2020-11-08:
I think it's possible to build the `gen_snapshot` without compiling the whole engine: https://medium.com/flutter-community/cross-compiling-dart-apps-f88e69824639 (See the code block where it says `# gen_snapshot (simarm)`). I think it shouldn't even require a cross-SDK or sysroot

Right now I'm trying to setup a OSX VM so I can build the `gen_snapshot` too and distribute them with the other engine binaries