ardera/flutter-pi #96: [flutter-pi] Could not find a suitable DRM encoder.

matgrz1993, 2020-10-08:
Hello,
First of all, thanks a lot for your job!
I would be very grateful if you could help me with the DRM encoder problem when running flutter_gallery.

I've done all the steps regarding installation flutter-pi, but when I try to run flutter_galary on my
Raspberry Pi 4, flutter-pi throws below error:

`
pi@raspberrypi:~/Projects/flutter_gallery/build/flutter_assets $ /home/pi/flutter-pi/out/flutter-pi .
[flutter-pi] WARNING: display didn't provide valid physical dimensions.
             The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
[flutter-pi] Could not find a suitable DRM encoder.
`

ardera, 2020-10-08:
I just pushed some stuff to the develop branch that should fix this. Can you try that out and see if it fixes it?

matgrz1993, 2020-10-09:
```
pi@raspberrypi:~/flutter-pi $ git status
Your branch is up to date with 'origin/develop'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)

        engine-binaries/

nothing added to commit but untracked files present (use "git add" to track)

pi@raspberrypi:~/flutter-pi $ git pull
Already up to date.

pi@raspberrypi:~/flutter-pi $ git log
commit 9b11f127fb46cb08e70b2a7cdfe8eaa8de977d5f (HEAD)
Author: Renzo Olivares <rmolivares@renzo-olivares.dev>
Date:   Wed Jul 29 11:55:00 2020 -0700

    Shared y axis reverse fix (#236)

    * Add reverse property to PageTransitionSwitcher that dictates whether the new widget transitions under or over the old child

    * Remove duration that was used in testing

    * update code segments

pi@raspberrypi:~/flutter-pi $ cd ~/Projects/flutter_gallery/build/flutter_assets

pi@raspberrypi:~/Projects/flutter_gallery/build/flutter_assets $ /home/pi/flutter-pi/out/flutter-pi .
[flutter-pi] WARNING: display didn't provide valid physical dimensions.
             The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
[flutter-pi] Could not find a suitable DRM encoder.
```

Still error :(

ardera, 2020-10-09:
> ```
> pi@raspberrypi:~/flutter-pi $ git log
> commit 9b11f127fb46cb08e70b2a7cdfe8eaa8de977d5f (HEAD)
> Author: Renzo Olivares <rmolivares@renzo-olivares.dev>
> Date:   Wed Jul 29 11:55:00 2020 -0700 [...]

I think you messed something up here. flutter-pi has no commit `9b11f127fb46cb08e70b2a7cdfe8eaa8de977d5f`, but flutter_gallery has. Maybe you `cd`'d into the flutter_gallery directory first and then ran git log?

Also, I'm pretty sure the output of `git status` is outdated if you didn't run `git pull` or `git fetch` before.

Just to be on the safe side, could you delete the whole flutter-pi folder and then check it out again using `git clone -b develop https://github.com/ardera/flutter-pi.git`? (Just to be sure we're not actually debugging git problems)

Also, can you send me the complete output of the `raspinfo` command? (Install using `sudo apt install raspinfo`)

matgrz1993, 2020-10-09:
Here is output from `raspinfo`: [raspinfo.txt](https://drive.google.com/file/d/1Ku0BoV7FsZWNtLQJxdVlfa0Zhnc-qJAn/view?usp=sharing)

Regarding `git log` - sorry for that, it was my mistake.
Here is proper output:
```
pi@raspberrypi:~/flutter-pi $ git pull
Already up to date.
pi@raspberrypi:~/flutter-pi $ git branch
* develop
  master
pi@raspberrypi:~/flutter-pi $ git log | head -6
commit 449c73fe07261c69dca985e41413cd527cfbe80d
Author: Hannes Winkler <hanneswinkler2000@web.de>
Date:   Fri Oct 9 00:58:12 2020 +0200

    fix typo

```

ardera, 2020-10-09:
Okay, thanks

Can you try two more things?
- run `kmscube` (install using `sudo apt install kmscube`)
- run `sudo apt update && sudo apt upgrade`

matgrz1993, 2020-10-10:
I've run those commands and here is the output:
```
pi@raspberrypi:~ $ sudo apt update && sudo apt upgrade
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Fetched 15.0 kB in 1s (18.2 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
127 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  agnostics alacarte apache2 apache2-bin apache2-data apache2-utils base-files bind9-host bluetooth bluez bluez-firmware bluez-obexd dbus-user-session firmware-atheros firmware-brcm80211 firmware-libertas firmware-misc-nonfree
  firmware-realtek glib-networking glib-networking-common glib-networking-services libbind9-161 libbluetooth3 libdbi-perl libdns-export1104 libdns1104 libexif12 libfm-data libfm-extra4 libfm-gtk-data libfm-gtk4 libfm-modules libfm4
  libgnutls30 libgs9 libgs9-common libgssdp-1.0-3 libgupnp-1.0-4 libisc-export1100 libisc1100 libisccc161 libisccfg163 libjavascriptcoregtk-4.0-18 libjson-c3 liblirc-client0 liblwres161 libmariadb3 libopenmpt-modplug1 libopenmpt0
  libpq5 libpython3.7 libpython3.7-dev libpython3.7-minimal libpython3.7-stdlib libqt5concurrent5 libqt5core5a libqt5dbus5 libqt5gui5 libqt5network5 libqt5printsupport5 libqt5sql5 libqt5sql5-sqlite libqt5widgets5 libqt5xml5
  libraspberrypi-bin libraspberrypi-dev libraspberrypi-doc libraspberrypi0 libunwind8 libvlc-bin libvlc5 libvlccore9 libwebkit2gtk-4.0-37 libx11-6 libx11-data libx11-dev libx11-xcb1 libzmq5 lxpanel lxpanel-data lxplug-bluetooth
  lxplug-cputemp lxplug-ejecter lxplug-network lxplug-ptbatt mariadb-client-10.3 mariadb-client-core-10.3 mariadb-common mariadb-server mariadb-server-10.3 mariadb-server-core-10.3 nfs-common pcmanfm pi-bluetooth pipanel piwiz pprompt
  python-pil python3-pil python3.7 python3.7-dev python3.7-minimal python3.7-venv qt5-gtk-platformtheme raspberrypi-bootloader raspberrypi-kernel raspberrypi-sys-mods raspi-config rc-gui rp-prefapps rpd-plym-splash rpi-chromium-mods
  rpi-eeprom vlc vlc-bin vlc-data vlc-l10n vlc-plugin-base vlc-plugin-notify vlc-plugin-qt vlc-plugin-samba vlc-plugin-skins2 vlc-plugin-video-output vlc-plugin-video-splitter vlc-plugin-visualization xserver-common xserver-xorg-core
127 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 264 MB of archives.
After this operation, 356 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://archive.raspberrypi.org/debian buster/main armhf rpi-chromium-mods armhf 20200901 [9,413 kB]
.
.
.
.
.
pi@raspberrypi:~ $ kmscube
drmModeGetResources failed: Invalid argument
failed to initialize legacy DRM
pi@raspberrypi:~ $ cd ~/Projects/flutter_gallery/build/flutter_assets
pi@raspberrypi:~/Projects/flutter_gallery/build/flutter_assets $ /home/pi/flutter-pi/out/flutter-pi .
[flutter-pi] WARNING: display didn't provide valid physical dimensions.
             The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
[flutter-pi] Could not find a suitable DRM encoder.
```

matgrz1993, 2020-10-12:
kmscube with `/dev/dri/card0` works:
```
pi@raspberrypi:~ $ sudo /usr/bin/kmscube -D /dev/dri/card1
drmModeGetResources failed: Operation not supported
failed to initialize legacy DRM
pi@raspberrypi:~ $ sudo /usr/bin/kmscube -D /dev/dri/card0
Using display 0x14915d0 with EGL version 1.4
===================================
EGL information:
  version: "1.4"
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless EGL_EXT_platform_device"
  display extensions: "EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
OpenGL ES 2.x information:
  version: "OpenGL ES 3.1 Mesa 19.3.2"
  shading language version: "OpenGL ES GLSL ES 3.10"
  vendor: "Broadcom"
  renderer: "V3D 4.2"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_implicit_conversions GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_primitive_bounding_box GL_EXT_shader_io_blocks GL_EXT_texture_border_clamp GL_EXT_texture_norm16 GL_KHR_context_flush_control GL_NV_image_formats GL_OES_draw_elements_base_vertex GL_OES_primitive_bounding_box GL_OES_shader_io_blocks GL_OES_texture_border_clamp GL_OES_texture_stencil8 GL_OES_texture_storage_multisample_2d_array GL_EXT_buffer_storage GL_EXT_float_blend GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_OES_shader_image_atomic GL_MESA_shader_integer_functions GL_KHR_parallel_shader_compile GL_MESA_framebuffer_flip_y GL_EXT_texture_query_lod "
===================================
```

ardera, 2020-10-12:
Sorry, I thought I replied to this thread already.

I couldn't reproduce the problem in my testing, even when I used a HDMI monitor and changed my `/boot/cmdline.txt` to match yours exactly.

> kmscube with `/dev/dri/card0` works:

Yeah, it's always one of the cards in `/dev/dri` that works. For me `/dev/dri/card1` is the working one. Though this shouldn't really matter as flutter-pi ignores non-working cards and tries to find the first working one.

Can you try this for me?
1. `$ sudo bash`
2. `# echo 0x1ff > /sys/module/drm/parameters/debug`
3. `# exit`
4. run flutter-pi (you can also try running it as root)
5. `$ sudo bash`
6. `# echo 0x0 > /sys/module/drm/parameters/debug`
7. `# exit`
8. send me the complete contents of `/var/log/syslog`

matgrz1993, 2020-10-12:
syslog: [syslog](https://drive.google.com/file/d/172fjvRAJs6iMLaEFmPD0Qo7W_CVsrTj9/view?usp=sharing)
command line output: [cmdline](https://drive.google.com/file/d/11OZjY6mYWjroVbiBbni21w0HonbZcN1P/view?usp=sharing)

I don't know why but the DRM bug is gone (replaced by `Dart Error: Can't load Kernel binary`)
Could it be related to the execution of this command:
`sudo /usr/bin/kmscube -D /dev/dri/card0`
?

ardera, 2020-10-12:
> replaced by `Dart Error: Can't load Kernel binary`

Make sure your [engine binaries](https://github.com/ardera/flutter-pi/tree/engine-binaries) and flutter SDK are up to date.

> Could it be related to the execution of this command:
> `sudo /usr/bin/kmscube -D /dev/dri/card0`

No, that's not possible. Did you reboot between the `sudo apt upgrade` and the invocation of flutter-pi though? The changes to kernel & firmware only take effect after reboot.

matgrz1993, 2020-10-13:
DRM bug related to this issue is fixed probably by executing commands `sudo apt update && sudo apt upgrade` and rebooting RPi, so I created a new issue related to `Invalid SDK hash`:
[Invalid SDK hash](https://github.com/ardera/flutter-pi/issues/100)

closed 2020-10-13 by matgrz1993