ardera/flutter-pi #97: Any chance to support legacy modesetting as an option for some SBC like nanopi fire3

bing-mao, 2020-10-11:
HI ardera,
Thank you for helping solve the #86, I have some other SBCs want to use you flutter-pi. I remember in that CR, you mentioned it's possible to implement a fallback to legacy modesetting, if atomic modesetting is not supported.

My nanopi fire 3 is based on nexell cpu with ARM mali 400MP2. The distributions for it are all based on Kernel 4.4.
I have recently upgrade it to kernel 5.4 (only to enable the hardware rendering support for mali GPU), now everything works fine. I can run kmscube. But the kernel driver for DRM is using legacy rendering. I have you use your old version in May before atomic modesetting is merged. Is it possible you let your latest code to optionally run legacy modesetting for these SBCs,

Thanks!

bing-mao, 2020-10-12:
[nanopi3_flutter-pi_log.txt](https://github.com/ardera/flutter-pi/files/5362600/nanopi3_flutter-pi_log.txt)
[nanopi3_modetest_log.txt](https://github.com/ardera/flutter-pi/files/5362601/nanopi3_modetest_log.txt)

Here is the log when testing commit 3e3c53ed85c30331a6c0ba2af349223654caa5dd

ardera, 2020-10-13:
landed in master #101

closed 2020-10-13 by ardera

bing-mao, 2020-10-14:
Hi ardera,
I have resolved the display issue. Now gallery app runs perfectly on my nanopi fire3. It is using your latest change to run under non atomic mode.
The key is to use latest mesa. We have to use mesa 20.0.2 (originally I used 19.1.6). It's really important. lima driver kernel developer told me I have to use latest mesa.

BR,
Bing

ardera, 2020-10-14:
That's good to know, I have Mesa 19.3.2 installed on my Pi and I experienced some graphics bugs when using HDMI. Maybe that's related to the mesa version as well

bing-mao, 2020-10-15:
Hi ardera,
For my board it is using mali400, the DRI driver has been supported by mesa recently (support in 19.1.x, but function well from 20.0.2). Different type of HW will have different status. But at least he told me the mesa is most likely introduce such problem. It's better always use the latest version of it.

BR,
Bing