ardera/flutter-pi #104: Make error: console_keyboard.h not found

g-jessmuir, 2020-10-15:
During `make`, I get the output:
```
cc -c -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_TEST_PLUGIN -DBUILD_OMXPLAYER_VIDEO_PLAYER_PLUGIN -O0 -ggdb  src/plugins/raw_keyboard.c -o out/obj/plugins/raw_keyboard.o
src/plugins/raw_keyboard.c:9:10: fatal error: console_keyboard.h: No such file or directory
 #include <console_keyboard.h>
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:38: out/obj/plugins/raw_keyboard.o] Error 1
```
Googling for `console_keyboard.h` didn't bring anything up, any hints on where to find it?

ardera, 2020-10-15:
`console_keyboard.h` was removed in the latest commit, but I don't know why the compilation worked fine for me despite this include file missing.
Whatever, should work now

g-jessmuir, 2020-10-16:
Yep, works on the latest commit. Thanks for the quick response!

closed 2020-10-16 by g-jessmuir