ardera/flutter-pi #105: Unhandled Exception: MissingPluginException

matgrz1993, 2020-10-15:
Hello,

I was trying to run simple application with `url_launcher` plugin and it crashed with following exception:
```
[ERROR:engine/src/flutter/lib/ui/ui_dart_state.cc(177)] Unhandled Exception: MissingPluginException(No implementation found for method canLaunch on channel plugins.flutter.io/url_launcher)
#0      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:157:7)
<asynchronous suspension>
#1      MethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart:332:12)
#2      MethodChannelUrlLauncher.canLaunch (package:url_launcher_platform_interface/method_channel_url_launcher.dart:18:21)
```

In README I found:
```
You can now theoretically run every flutter app you want using flutter-pi, including apps using extensions & plugins, 
just that you'd have to build the platform side of the plugins you'd like to use yourself.
```

So how can I build this platform side of the plugin and then use it with `flutter-pi`?
I see that [url_launcher ](https://github.com/flutter/plugins/tree/master/packages/url_launcher/url_launcher_linux) have linux platform code, so I need to somehow port this code to RPi?

Thanks in advance!

ardera, 2020-10-15:
You can't open multiple windows on headless linux. That's the thing about flutter-pi: there's only one window, which is flutter-pi. There's no window manager running in the background (Wayland or X11) like in desktop linux. If you want multiple windows, you may have better luck with the official linux desktop embedding.

matgrz1993, 2020-10-16:
Ok, thank you for info!

closed 2020-10-16 by matgrz1993