ardera/flutter-pi #106: about "-Wincompatible-point-types"

zhuangpeng, 2020-10-17:
hi ardear:
   thanks for your cool project!
   when I fetch this repo source code to build, it throw errors as follows:
   ```
   pi@raspberrypi:~/install/flutter-pi $ make
cc -c -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_TEST_PLUGIN -DBUILD_OMXPLAYER_VIDEO_PLAYER_PLUGIN -O0 -ggdb  src/flutter-pi.c -o out/obj/flutter-pi.o
cc -c -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_TEST_PLUGIN -DBUILD_OMXPLAYER_VIDEO_PLAYER_PLUGIN -O0 -ggdb  src/platformchannel.c -o out/obj/platformchannel.o
cc -c -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_TEST_PLUGIN -DBUILD_OMXPLAYER_VIDEO_PLAYER_PLUGIN -O0 -ggdb  src/pluginregistry.c -o out/obj/pluginregistry.o
cc -c -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_TEST_PLUGIN -DBUILD_OMXPLAYER_VIDEO_PLAYER_PLUGIN -O0 -ggdb  src/texture_registry.c -o out/obj/texture_registry.o
cc -c -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_TEST_PLUGIN -DBUILD_OMXPLAYER_VIDEO_PLAYER_PLUGIN -O0 -ggdb  src/compositor.c -o out/obj/compositor.o
cc -c -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_TEST_PLUGIN -DBUILD_OMXPLAYER_VIDEO_PLAYER_PLUGIN -O0 -ggdb  src/modesetting.c -o out/obj/modesetting.o
src/modesetting.c: In function ‘drmdev_new_from_fd’:
src/modesetting.c:437:46: warning: passing argument 3 of ‘fetch_crtcs’ from incompatible pointer type [-Wincompatible-pointer-types]
     ok = fetch_crtcs(drmdev, &drmdev->crtcs, &drmdev->n_crtcs);
                                              ^~~~~~~~~~~~~~~~
src/modesetting.c:171:84: note: expected ‘size_t *’ {aka ‘long unsigned int *’} but argument is of type ‘int *’
 static int fetch_crtcs(struct drmdev *drmdev, struct drm_crtc **crtcs_out, size_t *n_crtcs_out) {
                                                                            ~~~~~~~~^~~~~~~~~~~
src/modesetting.c:442:48: warning: passing argument 3 of ‘fetch_planes’ from incompatible pointer type [-Wincompatible-pointer-types]
     ok = fetch_planes(drmdev, &drmdev->planes, &drmdev->n_planes);
                                                ^~~~~~~~~~~~~~~~~
src/modesetting.c:270:87: note: expected ‘size_t *’ {aka ‘long unsigned int *’} but argument is of type ‘int *’
 static int fetch_planes(struct drmdev *drmdev, struct drm_plane **planes_out, size_t *n_planes_out) {
                                                                               ~~~~~~~~^~~~~~~~~~~~
cc -c -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_TEST_PLUGIN -DBUILD_OMXPLAYER_VIDEO_PLAYER_PLUGIN -O0 -ggdb  src/collection.c -o out/obj/collection.o
cc -c -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_TEST_PLUGIN -DBUILD_OMXPLAYER_VIDEO_PLAYER_PLUGIN -O0 -ggdb  src/cursor.c -o out/obj/cursor.o
cc -c -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_TEST_PLUGIN -DBUILD_OMXPLAYER_VIDEO_PLAYER_PLUGIN -O0 -ggdb  src/keyboard.c -o out/obj/keyboard.o
cc -c -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_TEST_PLUGIN -DBUILD_OMXPLAYER_VIDEO_PLAYER_PLUGIN -O0 -ggdb  src/plugins/services.c -o out/obj/plugins/services.o
cc -c -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_TEST_PLUGIN -DBUILD_OMXPLAYER_VIDEO_PLAYER_PLUGIN -O0 -ggdb  src/plugins/testplugin.c -o out/obj/plugins/testplugin.o
cc -c -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_TEST_PLUGIN -DBUILD_OMXPLAYER_VIDEO_PLAYER_PLUGIN -O0 -ggdb  src/plugins/text_input.c -o out/obj/plugins/text_input.o
cc -c -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_TEST_PLUGIN -DBUILD_OMXPLAYER_VIDEO_PLAYER_PLUGIN -O0 -ggdb  src/plugins/raw_keyboard.c -o out/obj/plugins/raw_keyboard.o
cc -c -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_TEST_PLUGIN -DBUILD_OMXPLAYER_VIDEO_PLAYER_PLUGIN -O0 -ggdb  src/plugins/omxplayer_video_player.c -o out/obj/plugins/omxplayer_video_player.o
cc -I./include -I/usr/include/libdrm -DBUILD_TEXT_INPUT_PLUGIN -DBUILD_TEST_PLUGIN -DBUILD_OMXPLAYER_VIDEO_PLAYER_PLUGIN -O0 -ggdb  -lgbm -ldrm -lGLESv2 -lEGL -lsystemd -linput -ludev -lxkbcommon -lrt -lpthread -ldl -lm -rdynamic  out/obj/flutter-pi.o out/obj/platformchannel.o out/obj/pluginregistry.o out/obj/texture_registry.o out/obj/compositor.o out/obj/modesetting.o out/obj/collection.o out/obj/cursor.o out/obj/keyboard.o out/obj/plugins/services.o out/obj/plugins/testplugin.o out/obj/plugins/text_input.o out/obj/plugins/raw_keyboard.o out/obj/plugins/omxplayer_video_player.o -o out/flutter-pi
```
how can I resolve it?

    thanks!

ardera, 2020-10-17:
Can you try latest master? Should work now.
You're using 64bit Raspbian, right? On 32bit, the size of `size_t` and `int` is the same, so I don't get the warning.

zhuangpeng, 2020-10-17:
hi ardera,
    thanks for you reply.
    yes, I use the 64bit raspberry offical OS. can I igrone this wanring?
    I have a other question:
    when I flutter-pi run app, throw an error:

    ```

    $ ~/install/flutter-pi/out/flutter-pi -r 90 ./flutter_assets

```

   [flutter-pi] WARNING: display didn't provide valid physical dimensions.
             The device-pixel ratio will default to 1.0, which may not be the fitting device-pixel ratio for your display.
===================================
display mode:
  resolution: 1024 x 600
  refresh rate: 60Hz
  physical size: 0mm x 0mm
  flutter device pixel ratio: 1.000000
===================================
EGL information:
  version: 1.4
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_device_base EGL_EXT_device_enumeration EGL_EXT_device_query EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm EGL_MESA_platform_surfaceless EGL_EXT_platform_device"
  display extensions: "EGL_ANDROID_blob_cache EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_MESA_query_driver EGL_WL_bind_wayland_display "
===================================
OpenGL ES information:
  version: "OpenGL ES 3.1 Mesa 19.3.2"
  shading language version: "OpenGL ES GLSL ES 3.10"
  vendor: "Broadcom"
  renderer: "V3D 4.2"
  extensions: "GL_EXT_blend_minmax GL_EXT_multi_draw_arrays GL_EXT_texture_format_BGRA8888 GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_mapbuffer GL_OES_rgb8_rgba8 GL_OES_standard_derivatives GL_OES_stencil8 GL_OES_texture_3D GL_OES_texture_float GL_OES_texture_half_float GL_OES_texture_half_float_linear GL_OES_texture_npot GL_OES_vertex_half_float GL_EXT_texture_sRGB_decode GL_OES_EGL_image GL_OES_depth_texture GL_OES_packed_depth_stencil GL_EXT_texture_type_2_10_10_10_REV GL_OES_get_program_binary GL_APPLE_texture_max_level GL_EXT_discard_framebuffer GL_EXT_read_format_bgra GL_EXT_frag_depth GL_NV_fbo_color_attachments GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_vertex_array_object GL_EXT_occlusion_query_boolean GL_EXT_texture_rg GL_EXT_unpack_subimage GL_NV_draw_buffers GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_EXT_draw_buffers GL_EXT_map_buffer_range GL_KHR_debug GL_KHR_texture_compression_astc_ldr GL_OES_depth_texture_cube_map GL_OES_required_internalformat GL_OES_surfaceless_context GL_EXT_color_buffer_float GL_EXT_sRGB_write_control GL_EXT_separate_shader_objects GL_EXT_shader_implicit_conversions GL_EXT_shader_integer_mix GL_EXT_base_instance GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_draw_elements_base_vertex GL_EXT_primitive_bounding_box GL_EXT_shader_io_blocks GL_EXT_texture_border_clamp GL_EXT_texture_norm16 GL_KHR_context_flush_control GL_NV_image_formats GL_OES_draw_elements_base_vertex GL_OES_primitive_bounding_box GL_OES_shader_io_blocks GL_OES_texture_border_clamp GL_OES_texture_stencil8 GL_OES_texture_storage_multisample_2d_array GL_EXT_buffer_storage GL_EXT_float_blend GL_KHR_no_error GL_KHR_texture_compression_astc_sliced_3d GL_OES_EGL_image_external_essl3 GL_OES_shader_image_atomic GL_MESA_shader_integer_functions GL_KHR_parallel_shader_compile GL_MESA_framebuffer_flip_y GL_EXT_texture_query_lod "
===================================
Error finding variable in buffer: find_var_offset_in_string: Invalid parameter
Error finding variable in buffer: find_var_offset_in_string: Invalid parameter
Error finding variable in buffer: find_var_offset_in_string: Invalid parameter
Error finding variable in buffer: find_var_offset_in_string: Invalid parameter
[flutter-pi] Warning: Could not load libflutter_engine.so.debug. Trying to open libflutter_engine.so...
[flutter-pi] Could not load libflutter_engine.so. dlopen: sccuess
[flutter-pi] Could not find a fitting libflutter_engine.

    
```

I find a closed issue [here](https://github.com/ardera/flutter-pi/issues/84) mentioned,and you give answer. I follow the engine branch copy files:

```
$ git clone --depth 1 --branch engine-binaries https://github.com/ardera/flutter-pi.git engine-binaries
$ cd engine-binaries
$ sudo cp ./libflutter_engine.so.* ./icudtl.dat /usr/lib
$ sudo cp ./flutter_embedder.h /usr/include
```

but still can't resolve this problem. 

Sorry, I usually do web development, not familiar with C language, disturb you.

Thank you again

ardera, 2020-10-17:
No, I added a commit to flutter-pi that fixes it 😄 It was a mistake on my side. I didn't notice it since I'm developing on 32bit Raspbian and the warning isn't printed there. Of course, the changes I make in this Github repo aren't automatically applied to your local cloned version of flutter-pi. So you need to do that manually. (== check out latest master)

To do this, just do `git pull` inside your cloned flutter-pi directory & then rebuild flutter-pi using `make`. (If it still doesn't work after that, try `git checkout origin/master` or just delete the directory and clone it again)

zhuangpeng, 2020-10-18:
hi ardera,
    thanks for you improve flutter-pi support arm 64bit of raspberry. the master branch perfect resolve `make` command issue.
    but when I run flutter use follows command:

`pi@raspberrypi:~/install/flutter-pi/out $ ./flutter-pi -r 90  ~/flutter_gallery_assets/flutter_assets/`

still throw errors:
```
Error finding variable in buffer: find_var_offset_in_string: Invalid parameter
Error finding variable in buffer: find_var_offset_in_string: Invalid parameter
Error finding variable in buffer: find_var_offset_in_string: Invalid parameter
Error finding variable in buffer: find_var_offset_in_string: Invalid parameter
[flutter-pi] Warning: Could not load libflutter_engine.so.debug. Trying to open libflutter_engine.so...
[flutter-pi] Could not load libflutter_engine.so. dlopen: sccuess
[flutter-pi] Could not find a fitting libflutter_engine.
```
I think maybe copied files location error, so I find file by:

```
$ sudo find / -name libflutter_engine.so.*
/home/pi/install/engine-binaries/libflutter_engine.so.debug
/home/pi/install/engine-binaries/libflutter_engine.so.release
/usr/lib/libflutter_engine.so.debug
/usr/lib/libflutter_engine.so.release
```
```
$ sudo find / -name flutter_embedder.h
/home/pi/install/engine-binaries/flutter_embedder.h
/usr/include/flutter_embedder.h
```
I can't find out what's wrong there. Can you check it for me? thank you very much!

ardera, 2020-10-18:
Oh, I forgot, the engine binaries are compiled for 32bit. I'll have to compile the engine binaries for 64bit so they work on your 64bit system. I'll have to look into whether thats viable

zhuangpeng, 2020-10-18:
hi ardea:
   thanks！ hope viable. 
   If you need me to do a test, I'm happy to do it.
   My email address is: 139128688@qq.com

closed 2020-11-04 by zhuangpeng