ardera/flutter-pi #107: any guide to port this flutter-pi to other ARM SOC?

nanfang2000, 2020-10-17:
I'm interested in implement the embedder to other SOC (also have opengles GPU).
Where is flutter-pi code that specified to  Raspberry PI?

ardera, 2020-10-19:
There's not much about flutter-pi that's specific to Raspberry Pi. The things that are specific to Raspberry Pi:
- The `omxplayer_video_player` plugin (Since `omxplayer` is only really supported on Raspberry Pi)
- On Raspberry Pi, OpenGL reports some extensions as supported, even though they aren't. See my [`glGetString` hack](https://github.com/ardera/flutter-pi/blob/f95da4b574d5e1aa2b2763b3c5ff0f4f86ab0479/src/flutter-pi.c#L234).

That's basically it. On most platforms, flutter-pi should work out of the box. (See the [supported platforms](https://github.com/ardera/flutter-pi#supported-platforms) section in the readme)

Other people use flutter-pi on Dragon Board 410c, NanoPi Fire 3 & ODROID C4 without modification. (Though to be fair, they did require some fixes since flutter-pi was relying on Raspberry Pi-specific behaviour some times.)

nanfang2000, 2020-10-20:
@ardera Thank you, I will try to run on my SBC. any good new or modification I will share.

closed 2020-10-20 by nanfang2000